agherzan/meta-raspberrypi #381: Generic raspberry image

thknx, 2019-02-13:
How can I build generic-raspberry yocto image which be able to work all raspberry boards? Everytime I have to change MACHINE parameter on yocto but I want to build a generic image like raspbian which consists necessery drivers and firmwares?

Can you advice me please?

kraj, 2019-02-13:
There is no pre-cooked recipe for that, you can create a new image from `MACHINE="raspberrypi"` and then add all firmwares into image. You also might need to package all kernel configs and DTBO for all machines into this jumbo image.

thknx, 2019-02-13:
I thought may be Mr. @agherzan help me because I have not knowledge about tuning on arm&rpi.

agherzan, 2019-02-14:
As Khem said above, we don't have this implemented at this point. I recommend defining a new machine and start building up all the needed bits (dtbs, firmware etc). The only thing you need to pay extra care to is the kernel image filename because that is machine specific (actually arch specific). So you will have to provide all the needed ones - kernel7, kernel and kernel8.

commit 7059c374 (2019-04-24): psplash: Raise alternatives priority to 200

Alternatives priority of psplash in poky is now 100,
so we need something higher (fixes #381).
PR #496 (2019-09-26): psplash: Raise alternatives priority to 200
Alternatives priority of psplash in poky is now 100,
so we need something higher (fixes #381).

(cherry picked from commit 7059c374512f1c1c0df9ecab0703d11438bdf78b)
commit 4e5be97d (2019-09-26): psplash: Raise alternatives priority to 200

Alternatives priority of psplash in poky is now 100,
so we need something higher (fixes #381).
closed 2019-04-24 by agherzan