agherzan/meta-raspberrypi #382: Boot partition should be fat32

rosterloh, 2019-02-13:
**Description**
As per the Raspberry Pi documentation there is an issue with some compute modules when the boot partition is not FAT32. 

**Steps to reproduce the issue:**
1. Build image
2. Check partitions using gparted

**Describe the results you received:**
Some images don't boot on a compute module

**Describe the results you expected:**
Should boot

**Additional information you deem important (e.g. issue happens only occasionally):**
As I see it line [114 of lasses/sdcard_image-rpi.bbclass](https://github.com/agherzan/meta-raspberrypi/blob/master/classes/sdcard_image-rpi.bbclass#L114) should change from "mkfs.vfat -n "${BOOTDD_VOLUME_ID}" -S 512 -C ${WORKDIR}/boot.img $BOOT_BLOCKS" to "mkfs.vfat -F32 -n "${BOOTDD_VOLUME_ID}" -S 512 -C ${WORKDIR}/boot.img $BOOT_BLOCKS" but I'd like someone else to verify

**Additional details (revisions used, host distro, etc.):**

agherzan, 2019-02-14:
@rosterloh that is correct. Would you feel like pushing a PR to address this?

rosterloh, 2019-02-15:
@agherzan should be good now

agherzan, 2019-02-15:
Nice. Waiting for builds and will merge.

PR #384 (2019-02-19): sdcard_image-rpi: Format boot partition as FAT32
Closes: #382

**- What I did**
Added -F32 to the mkfs.vfat step
commit ba54962b (2019-02-19): sdcard_image-rpi: Format boot partition as FAT32

Closes: #382
commit 35331f2d (2019-02-19): sdcard_image-rpi: Format boot partition as FAT32

Closes: #382
commit 2b3693ea (2019-02-19): sdcard_image-rpi: Format boot partition as FAT32

Closes: #382
closed 2019-02-19 by agherzan