agherzan/meta-raspberrypi #391: file /usr/include/KHR/khrplatform.h is installed by both userland and mesa-gl, aborting

Romaxx, 2019-02-22:
**Description**

**Steps to reproduce the issue:**
1.
2.
3.

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

agherzan, 2019-02-24:
@Romaxx this should be fixed now in master. Can you confirm? cc @kraj

kraj, 2019-02-24:
it should be fixed now

Romaxx, 2019-02-25:
error always appear if MACHINE_FEATURES_append =" vc4graphics" is not provided,
for raspberrypi3-64 it's not throwing because it's set in .conf

agherzan, 2019-02-25:
@romaxx what master revision have you used?

Romaxx, 2019-02-25:
@adherzan oh well, I've made some modifications to userland_git.bb.. git didn't pull and no advertise me.. 
now have,
`meta-raspberrypi     = "master:60dbfd25443754690f834afe30e7a6dca46a40da" `
little issue with fetching 
`DEBUG: Fetcher failure: Unable to find revision e2d2941326922b63d722ebc46520c3a2287b675f in branch rpi-4.19.y even from upstream`
so I've replaced with 18b4bd6567bc78af187211c5cb8bb439e9369811
have error with 
`linux-raspberrypi-1_4.19.23+gitAUTOINC+18b4bd6567-r0 do_kernel_version_sanity_check: Package Version (4.19.23+gitAUTOINC+18b4bd6567) does not match of kernel being built (4.19.25)`
have replaced LINUX_VERSION ?= "4.19.25" in *4.19.bb 
now I have (without MACHINE_FEATURES_append =" vc4graphics")
` include/epoxy/egl_generated.h:11:10: fatal error: EGL/eglplatform.h: No such file or directory
|  #include "EGL/eglplatform.h"`

kraj, 2019-02-26:
I can reproduce this issue with userland graphics and latest OE-core/mesa, it needs a bit of patching and I have posted a pull request which should fix it finally.

https://github.com/agherzan/meta-raspberrypi/pull/392

This will make sure X11/eglfs images can be built with userland graphics, please note that
wayland may not work, thats a well known problem and limitation of userland graphics

agherzan, 2019-03-08:
Thanks, @kraj. @Romaxx can you confirm that current master works as expected?

kraj, 2019-04-17:
> same pb without MACHINE_FEATURES_append =" vc4graphics"
> 
> ```
> /home/ubuntu/yocto/master/rpi-build/tmp/work/cortexa7t2hf-neon-vfpv4-test-linux-gnueabi/xserver-xorg/2_1.20.4-r0/recipe-sysroot/usr/include/GL/glext.h:467:10: fatal error: KHR/khrplatform.h: No such file or directory
> |  #include <KHR/khrplatform.h>
> ```

this seems to be related to old khronos header APIs in userland package probably. Recently I have not tested userland option so much, but my guess is we need to patch userland headers here.

Romaxx, 2019-04-27:
@agherzan hi, yes I've tried with the last 7059c37, same error

agherzan, 2019-04-29:
That's a bummer. We will need to look into this and if by any chance you can and have the time, patches are welcome.

pierreinglebert, 2020-03-06:
Does someone have resolved this issue or found a workaround ?

kraj, 2020-03-06:
@pierreinglebert are you using `userland` for graphics ?

xyklex, 2020-09-15:
Hi @kraj, thanks for the fix

closed 2020-09-16 by agherzan