agherzan/meta-raspberrypi #393: BLE for RPI3B+

utsavjain0511, 2019-02-28:
**Description**

I am using the master branch of meta-raspberrypi. I am trying to configure bluetooth device on RPI3B+ 

My local.conf looks like this:

```
CONF_VERSION = "1"
ENABLE_UART = "1"
ENABLE_RPI3_SERIAL_CONSOLE = "1"
SERIAL_CONSOLES = "115200;ttyAMA0"
MACHINE = "raspberrypi3-64"

CORE_OS = " \
    bluez5 \
    openssh openssh-keygen openssh-sftp-server \
    tzdata \
"

WIFI_SUPPORT = " \
    crda \
    iw \
    wpa-supplicant \
"

DEV_SDK_INSTALL = " \
    python3 \
    pi-bluetooth \
    python3-modules \
    python3-pip  \
    python3-pygobject \
    python3-dbus \
    dbus-glib \
    gobject-introspection \
"

DEV_EXTRAS = " \
    ntp \
    ntp-tickadj \
"

EXTRA_TOOLS_INSTALL = " \
    bzip2 \
    ethtool \
    i2c-tools \
    iperf3 \
    iproute2 \
    iptables \
    less \
    netcat \
    procps \
    sysfsutils \
    tcpdump \
    unzip \
    util-linux \
    zip \
    dbus \
"

PREFERRED_VERSION_linux-raspberrypi = "4.%"
DISTRO_FEATURES_remove = "x11 wayland"

DISTRO_FEATURES_append = " bluez5 bluetooth wifi systemd"

IMAGE_INSTALL_append = " \
    ${CORE_OS} \
    ${DEV_SDK_INSTALL} \
    ${DEV_EXTRAS} \
    ${EXTRA_TOOLS_INSTALL} \
    ${WIFI_SUPPORT} \
"

VIRTUAL-RUNTIME_init_manager = "systemd"

PACKAGE_EXCLUDE_append = " packagegroup-core-ssh-dropbear"
```

When I start bluetoothctl
> devices
returns "No default controller available"
  
My understanding is that at the boot time the image is not able to attach the BLE device. 

**Additional details (revisions used, host distro, etc.):**
I am using the latest master versions of all the sources.

kraj, 2019-02-28:
Can you post the kernel boot logs ? does it discover bluetooth at all

utsavjain0511, 2019-02-28:
Here are the boot logs

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.98 (oe-user@oe-host) (gcc version 8.2.0 (GCC)) #1 SMP PREEMPT Thu Feb 28 01:00:52 UTC 2019
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:31
[    0.000000] percpu: Embedded 22 pages/cpu @ffffffd5fb377000 s50456 r8192 d31464 u90112
[    0.000000] pcpu-alloc: s50456 r8192 d31464 u90112 alloc=22*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: 8250.nr_uarts=1 cma=256M bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Memory: 675740K/970752K available (7292K kernel code, 900K rwdata, 2572K rodata, 2816K init, 676K bss, 32868K reserved, 262144K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff96fa080000 - 0xffffff96fa7a0000   (  7296 KB)
[    0.000000]     .rodata : 0xffffff96fa7a0000 - 0xffffff96faa30000   (  2624 KB)
[    0.000000]       .init : 0xffffff96faa30000 - 0xffffff96facf0000   (  2816 KB)
[    0.000000]       .data : 0xffffff96facf0000 - 0xffffff96fadd1200   (   901 KB)
[    0.000000]        .bss : 0xffffff96fadd1200 - 0xffffff96fae7a568   (   677 KB)
[    0.000000]     fixed   : 0xffffffbefe7fb000 - 0xffffffbefec00000   (  4116 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf57000000 - 0xffffffbf57ed0000   (    14 MB actual)
[    0.000000]     memory  : 0xffffffd5c0000000 - 0xffffffd5fb400000   (   948 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26487 entries in 104 pages
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000230] Console: colour dummy device 80x25
[    0.000263] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=19200)
[    0.000284] pid_max: default: 32768 minimum: 301
[    0.000613] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000643] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.001666] Disabling memory control group subsystem
[    0.006074] ASID allocator initialised with 32768 entries
[    0.008071] Hierarchical SRCU implementation.
[    0.010324] EFI services will not be available.
[    0.012137] smp: Bringing up secondary CPUs ...
[    0.019338] Detected VIPT I-cache on CPU1
[    0.019403] CPU1: Booted secondary processor [410fd034]
[    0.026439] Detected VIPT I-cache on CPU2
[    0.026482] CPU2: Booted secondary processor [410fd034]
[    0.033556] Detected VIPT I-cache on CPU3
[    0.033597] CPU3: Booted secondary processor [410fd034]
[    0.033733] smp: Brought up 1 node, 4 CPUs
[    0.033773] SMP: Total of 4 processors activated.
[    0.033789] CPU features: detected feature: 32-bit EL0 Support
[    0.033802] CPU features: detected feature: Kernel page table isolation (KPTI)
[    0.037207] CPU: All CPU(s) started at EL2
[    0.037242] alternatives: patching kernel code
[    0.038531] devtmpfs: initialized
[    0.050906] random: get_random_u32 called from bucket_table_alloc+0x108/0x270 with crng_init=0
[... 222 lines cut ...]
[    6.189706] Bluetooth: Core ver 2.22
[    6.193590] NET: Registered protocol family 31
[    6.198338] Bluetooth: HCI device and connection manager initialized
[    6.204919] Bluetooth: HCI socket layer initialized
[    6.210003] Bluetooth: L2CAP socket layer initialized
[    6.215339] Bluetooth: SCO socket layer initialized
[   42.262709] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   42.268181] Bluetooth: BNEP filters: protocol multicast
[   42.273806] Bluetooth: BNEP socket layer initialized
[  153.253477] usb 1-1.2: new full-speed USB device number 7 using dwc_otg
[  153.354847] usb 1-1.2: New USB device found, idVendor=062a, idProduct=4102
[... cut ...]
[ 2363.805144] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2363.811493] brcmfmac: power management disabled
[ 2366.781054] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 2379.193284] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[ 2379.216599] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[ 3001.069413] usb 1-1.2: USB disconnect, device number 7
[ 3932.095928] ip_tables: (C) 2000-2006 Netfilter Core Team
[ 4077.145263] nf_conntrack version 0.5.0 (7680 buckets, 30720 max)
```

kraj, 2019-02-28:
OK so it seems kernel has initialized it fine, so it seems the userspace part is not initialized.

can you try with this patch applied 

```patch
diff --git a/recipes-connectivity/bluez5/bluez5_%.bbappend b/recipes-connectivity/bluez5/bluez5_%.bbappend
index 6651fb8..e71d634 100644
--- a/recipes-connectivity/bluez5/bluez5_%.bbappend
+++ b/recipes-connectivity/bluez5/bluez5_%.bbappend
@@ -11,6 +11,8 @@ BCM_BT_RDEPENDS = "pi-bluetooth"

 SRC_URI_append_raspberrypi0-wifi = " ${BCM_BT_SOURCES}"
 SRC_URI_append_raspberrypi3 = " ${BCM_BT_SOURCES}"
+SRC_URI_append_raspberrypi3-64 = " ${BCM_BT_SOURCES}"

 RDEPENDS_${PN}_append_raspberrypi0-wifi = " ${BCM_BT_RDEPENDS}"
 RDEPENDS_${PN}_append_raspberrypi3 = " ${BCM_BT_RDEPENDS}"
+RDEPENDS_${PN}_append_raspberrypi3-64 = " ${BCM_BT_RDEPENDS}"
```

agherzan, 2019-02-28:
Looks like it.

utsavjain0511, 2019-02-28:
I did apply the patch. There is a failure in building the image now due to the following logs

```
NOTE: Applying patch '0001-bcm43xx-Add-bcm43xx-3wire-variant.patch' (../sources/meta-raspberrypi/recipes-connectivity/bluez5/bluez5/0001-bcm43xx-Add-bcm43xx-3wire-variant.patch)
ERROR: Command Error: 'quilt --quiltrc /home/ubuntu/afr-ble-yocto/build/tmp/work/aarch64-poky-linux/bluez5/5.50-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Patch 0001-bcm43xx-Add-bcm43xx-3wire-variant.patch is already applied; check your series file
```

I am guessing it is because of the following line in the raspberrypi3-64.conf machine config

`MACHINEOVERRIDES = "raspberrypi3:${MACHINE}"`

This means its applying the patches required.

kraj, 2019-02-28:
Oh I see then the patch I suggested is wrong, its already effective

utsavjain0511, 2019-03-01:
Is there something I am missing here? I checked the patches are applied correctly. 

When I run /usr/bin/btuart command 

I get following error
`bcm43xx_init`
`Initialization timed out.`

kraj, 2019-03-01:
Can you try using kernel 3.19

agherzan, 2019-03-01:
@utsavjain0511 4.19 is default now in master. Give that a spin.

utsavjain0511, 2019-03-04:
Thanks! I was able to fix it by removing the line 

`SERIAL_CONSOLES = "115200;ttyAMA0"` 

from the local.conf . I am guessing it is because the bluetooth initializing script is trying to attach the interface to /dev/ttyAMA0 . I will find the exact reason and post it here. Closing the issue.

closed 2019-03-04 by utsavjain0511