agherzan/meta-raspberrypi #394: RPI3B+ bitbake rpi-test-image build issue

infojg9, 2019-03-03:
Hi All,

How to solve this RPI3B+ bitbake rpi-test-image build issue. Any hints would be highly appreciated. 

Configuration(config/local.conf, git branches, system) and Log as follows：

**Description**

Followed the steps from https://meta-raspberrypi.readthedocs.io/en/latest/readme.html 

**Steps to reproduce the issue:**

1. checkout thud branch (ans later also tried master branch with clean cache) for meta-openembedded,  meta-raspberrypi,  poky-thud official repositories.
2. Then sourced  `source poky-thud/oe-init-build-env ./build` and modified local.conf as described in the above link.
3. Then started the bitbake build which is failed at sanity check as inline.

`$ bitbake rpi-test-image
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    MACHINE=raspberrypi3-64 is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.
`

** Configurations **

`
$ ls ../
build  meta-openembedded  meta-raspberrypi  poky-thud
$ uname -a
Linux u1804lts 4.15.0-45-generic #48-Ubuntu SMP Tue Jan 29 16:28:13 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
$ lsb_release -a
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

$ cd ../poky-thud/
$ git checkout thud
Switched to branch 'thud'
Your branch is up to date with 'origin/thud'.
 $ git pull --rebase origin thud
From git://git.yoctoproject.org/poky
 * branch                  thud       -> FETCH_HEAD
Already up to date.
Current branch thud is up to date.
$ cd ../meta-openembedded/
$ git checkout thud
Switched to branch 'thud'
Your branch is up to date with 'origin/thud'.
$ git pull --rebase origin thud
From github.com:openembedded/meta-openembedded
 * branch                thud       -> FETCH_HEAD
Already up to date.
Current branch thud is up to date.
$ cd ../meta-raspberrypi/
$ git checkout thud
$ git pull --rebase origin thud
From github.com:agherzan/meta-raspberrypi
 * branch            thud       -> FETCH_HEAD
Already up to date.
Current branch thud is up to date.
`

**Describe the results you received:**

`$ bitbake rpi-test-image
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    MACHINE=raspberrypi3-64 is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.
`

**Describe the results you expected:**

Successful raspberrypi3-64 build image

**Additional information you deem important (e.g. issue happens only occasionally):**

`
$ cat poky-thud/build/conf/local.conf

# Machine Selection
MACHINE ?= "raspberrypi3-64"
#MACHINE ?= "raspberrypi3"

# deleting this lets us create a rpi-sdimg, default line commented
#IMAGE_FSTYPES = "tar.xz"
IMAGE_FSTYPES = "tar.bz2 ext3.xz"

#  Overwrite SDIMG_ROOTFS_TYPE in local.conf
SDIMG_ROOTFS_TYPE = "ext3.xz"

DISABLE_OVERSCAN = "1"
DISABLE_SPLASH = "1"

# To add the kdbg over console (kgdboc) parameter to the kernel command line, set this variable in local.conf:
ENABLE_KGDB = "1"

# To build an initramfs image:
INITRAMFS_IMAGE = "Joy_Yocto_Thud_Custom"
INITRAMFS_IMAGE_BUNDLE = "1"
BOOT_SPACE = "1073741"
INITRAMFS_MAXSIZE = "315400"
IMAGE_FSTYPES_pn-${INITRAMFS_IMAGE} = "${INITRAMFS_FSTYPES}"

#DL_DIR ?= "${TOPDIR}/downloads"

#SSTATE_DIR ?= "${TOPDIR}/sstate-cache"

# DISTRO ?= "poky-bleeding"

# E.g.: PACKAGE_CLASSES ?= "package_rpm package_deb package_ipk"
# We default to rpm:
PACKAGE_CLASSES ?= "package_rpm"

# Supported values are i686 and x86_64
#SDKMACHINE ?= "x86_64"

EXTRA_IMAGE_FEATURES ?= "debug-tweaks"

USER_CLASSES ?= "buildstats image-mklibs image-prelink"

PATCHRESOLVE = "noop"

BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
CONF_VERSION = "1"
`

**Additional details (revisions used, host distro, etc.):**

all other config/* files untouched.

agherzan, 2019-03-07:
What is your `bblayers.conf` file in the build directory?

agherzan, 2019-03-13:
Most probably you don't have the BSP layer in the `bblayers.conf` file. I'll close this issue for now but feel free to add any feedback if the problem persists.

closed 2019-03-13 by agherzan