agherzan/meta-raspberrypi #395: linux-raspberrypi 4.19 cannot be fetched on master

lbonn, 2019-03-04:
**Description**

The linux-raspberrypi recipe references a commit that is outside of the specified `rpi-4.19.y` branch: https://github.com/raspberrypi/linux/commit/e2d2941326922b63d722ebc46520c3a2287b675f. Bitbaking an image on up-to-date repos will fail at the fetch stage of the recipe.

Current workaround is to define `PREFERRED_VERSION_linux-raspberrypi = "4.14.%"` in build configuration (should at least fix the fetching).

**Steps to reproduce the issue:**

Build an image with the meta-raspberrypi layer on master, using the default preferred version (4.19.x since 6a3dc071cde4f6bec7dd807165db056a3d7115eb)

**Describe the results you received:**

```
WARNING: linux-raspberrypi-1_4.19.23+gitAUTOINC+e2d2941326-r0 do_fetch: Failed to fetch URL git://github.com/raspberrypi/linux.git;protocol=git;branch=rpi-4.19.y, attempting MIRRORS if available
ERROR: linux-raspberrypi-1_4.19.23+gitAUTOINC+e2d2941326-r0 do_fetch: Fetcher failure: Unable to find revision e2d2941326922b63d722ebc46520c3a2287b675f in branch rpi-4.19.y even from upstream
ERROR: linux-raspberrypi-1_4.19.23+gitAUTOINC+e2d2941326-r0 do_fetch: Fetcher failure for URL: 'git://github.com/raspberrypi/linux.git;protocol=git;branch=rpi-4.19.y'. Unable to fetch URL from any source.
ERROR: linux-raspberrypi-1_4.19.23+gitAUTOINC+e2d2941326-r0 do_fetch:
ERROR: linux-raspberrypi-1_4.19.23+gitAUTOINC+e2d2941326-r0 do_fetch: Function failed: base_do_fetch
```

**Additional details (revisions used, host distro, etc.):**

meta-raspberrypi: 60dbfd25443754690f834afe30e7a6dca46a40da
poky: 4d275d97b6c572fe11668ac16d2c77c018340c7c

kraj, 2019-03-04:
its fixed via https://github.com/agherzan/meta-raspberrypi/pull/392
please try it out

agherzan, 2019-03-05:
Was also merged in master. Try it out.

closed 2019-03-05 by agherzan

lbonn, 2019-03-06:
Thanks! Checked that it indeed built and booted.