agherzan/meta-raspberrypi #396: Missing extensions modules when cross compiling python 3.5.5(sumo)/3.5.6(thud) for arm on Linux

miikeat, 2019-03-06:
I'm trying to host a rest api on my raspberrypi3. I wrote everything in python and used falcon, so i would like to host it using gunicorn which I installed using python3-pip. This works fine on my desktop, but not on my raspberrypi. 

I am running a basic Yocto image where i just added ```IMAGE_INSTALL_append = " python3 python3-pip"``` to ```local.conf```. Everything else is as described in the README.

The Errormessage states that the module 'fcntl' is missing which should be a builtin module. I also tried to figure out if this is a problem with the python recipe in general, but if i build a qemux86 image and run in qemu gunicorn works and python has the module 'fcntl' included. 

I later stumbled across this issue: https://stackoverflow.com/questions/40047363/how-to-compile-missing-extensions-modules-when-cross-compiling-python-3-5-2-for which led me to this bugreport https://bugs.python.org/issue28444 which fits this issue pretty good but should be fixed since 3.5.2. Maybe for some reason that problem is back again...

**Steps to reproduce the issue:**
1. Build a basic core-image-base image based on sumo or thud with python3 and python3-pip added to the image. (Add ```IMAGE_INSTALL_append = " python3 python3-pip"``` to ```local.conf```)
2. Burn and boot image and run ```pip3 install gunicorn```
3. Run ```gunicorn```

**Describe the results you received:**
```
login as: root
root@raspberrypi3:~# pip3 install gunicorn
Collecting gunicorn
  Downloading https://files.pythonhosted.org/packages/8c/da/b8dd8deb741bff556db53902d4706774c8e1e67265f69528c14c003644e6/gunicorn-19.9.0-py2.py3-none-any.whl (112kB)
    100% |################################| 122kB 1.8MB/s
Installing collected packages: gunicorn
Successfully installed gunicorn-19.9.0
root@raspberrypi3:~# gunicorn
Traceback (most recent call last):
  File "/usr/bin/gunicorn", line 7, in <module>
    from gunicorn.app.wsgiapp import run
  File "/usr/lib/python3.5/site-packages/gunicorn/app/wsgiapp.py", line 9, in <module>
    from gunicorn.app.base import Application
  File "/usr/lib/python3.5/site-packages/gunicorn/app/base.py", line 12, in <module>
    from gunicorn import util
  File "/usr/lib/python3.5/site-packages/gunicorn/util.py", line 9, in <module>
    import fcntl
ImportError: No module named 'fcntl'
root@raspberrypi3:~#
```

**Describe the results you expected:**
Gunicorn should print the usage message.
```
usage: gunicorn [OPTIONS] [APP_MODULE]
gunicorn: error: No application module specified.
```

**Additional information you deem important (e.g. issue happens only occasionally):**
According to a comment in the bugreport ```sys.builtin_module_names``` should contain fcntl but on neither system that was the case.
Builtin Modules in my raspberrypi image:
```
root@raspberrypi3:~# python3
Python 3.5.6 (default, Mar  6 2019, 15:43:23)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.builtin_module_names
('_ast', '_codecs', '_collections', '_functools', '_imp', '_io', '_locale', '_operator', '_signal', '_sre', '_stat', '_string', '_symtable', '_thread', '_tracemalloc', '_warnings', '_weakref', 'atexit', 'builtins', 'errno', 'faulthandler', 'gc', 'itertools', 'marshal', 'posix', 'pwd', 'sys', 'time', 'xxsubtype', 'zipimport')
>>>
```

I then found out ```help('modules')``` lists all importable modules, which also works and contains 'fcntl' on the qemux86 machine. But on my raspberrypi i just get another error:
```
>>> help()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.5/_sitebuiltins.py", line 102, in __call__
    import pydoc
ImportError: No module named 'pydoc'
>>>

**Additional details (revisions used, host distro, etc.):**

```
Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.6.1"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "thud:f5a57e939e626a5b7c6de5b51799ca602ed355ed"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "thud:6ef9657068492d4644079c88f2adee9c3cac9543"
meta-raspberrypi     = "thud:c71d79efc5a06a0c896c278c94f5b14413bb4d69"
```

Additions to default ```local.conf```:

```
MACHINE = "raspberrypi3"

IMAGE_FEATURES_append = " ssh-server-dropbear"
IMAGE_INSTALL_append = " python3 python3-pip"

```

kraj, 2019-03-07:
you need to add 
```
IMAGE_INSTALL_append = " python3-fcntl"
```

miikeat, 2019-03-07:
Wow ... now I'm impressed...
I seriously have no idea how that works!

I also saw that hint somewhere earlier but as there is no recipe and ```bitbake python3-fcntl``` throws out an error:
```
ERROR: Nothing PROVIDES 'python3-fcntl'. Close matches:
  python3-prctl                                                                                                                                                                   
  python3-can                                                                                                                                                                     
  python3-ntplib                                                                                                                                                                  
  python3 RPROVIDES python3-fcntl          
```
I thought someone didn't care to check if that recipe even exists and just wrote that...
But now i noticed the last line of the error...
Where is this ```RPROVIDES``` coming from in the python3 recipe?

Also, shouldn't that module be included by default, same as pydoc and probably some other default modules? In Python 2 they are included and on x86 machines as well! Whats the reason that they are missing on ARM?

kraj, 2019-03-07:
a recipe can provide many packages, and when you install into image you install packages not recipes. Why many packages, because in some cases only portions are needed so we can keep the size in check. bitbake <recipe>  can generate multiple rpms/ipks check the deploy directory for rpms in the build tree of a recipe

closed 2019-03-07 by agherzan

miikeat, 2019-03-07:
yeah .. i kind of figured that out in the meantime, that via the manifest.json these packages are generated dynamically...

But shouldn't the behaviour of the python3 recipe be the same on x86 and arm? why do i need to manually pick those packages on arm but on x86 they are included when only installing python3!

agherzan, 2019-03-07:
Because you are using a RPi specific image which packages a set for packages. What image are you comparing with X86?

miikeat, 2019-03-07:
I hope if it's ok if i continue asking here out of pure curiosity:

I have one qemux86 image with just poky and i build core-image-base there.

And one raspberrypi3 image with poky, meta-openembedded and meta-raspberrypi where i build as well core-image-base.
In both images i add python3 and only in the qemux86 image i have all modules. 

Neither meta-raspberrypi nor meta-openembedded have a bbappend for python3 so i just wonder how it comes, that for raspberrypi i'm missing those modules?

I also looked through the machine configurations and could not find any clue what is causing these packages to be missing on raspberrypi.

Whats also kind of irritating, python 2.x contains all those modules by default! 

So I don't really see where this behaviour comes from and if that is really intended that way...

agherzan, 2019-03-07:
Can you try using the same build configuration to build core-image-base for both machines? Basically, run twice `bitbake core-image-base` with a different machine in `local.conf`

miikeat, 2019-03-07:
I had the same idea and I'm already building it... =) 

In a couple of hours I should know more!

miikeat, 2019-03-07:
The results:
```
Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "neon-18.04"
TARGET_SYS           = "i586-poky-linux"
MACHINE              = "qemux86"
DISTRO               = "poky"
DISTRO_VERSION       = "2.6.1"
TUNE_FEATURES        = "m32 i586"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "thud:f5a57e939e626a5b7c6de5b51799ca602ed355ed"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "thud:6ef9657068492d4644079c88f2adee9c3cac9543"
meta-raspberrypi     = "thud:c71d79efc5a06a0c896c278c94f5b14413bb4d69"

```

```
root@qemux86:~# pip3 install gunicorn
Collecting gunicorn
  Downloading https://files.pythonhosted.org/packages/8c/da/b8dd8deb741bff556db53902d4706774c8e1e67265f69528c14c003644e6/gunicorn-19.9.0-py2.py3-none-any.whl (112kB)
    100% |################################| 122kB 557kB/s 
Installing collected packages: gunicorn
Successfully installed gunicorn-19.9.0
root@qemux86:~# gunicorn
usage: gunicorn [OPTIONS] [APP_MODULE]
gunicorn: error: No application module specified.
root@qemux86:~# 
```

miikeat, 2019-03-08:
As ```help()``` -> ```modules``` in python console lists the importable modules here a comparison between the two images:

raspberrypi3:
```
root@raspberrypi3:~# python3
Python 3.5.6 (default, Mar  6 2019, 15:43:23)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> help()

Welcome to Python 3.5's help utility!

If this is your first time using Python, you should definitely check out
the tutorial on the Internet at http://docs.python.org/3.5/tutorial/.

Enter the name of any module, keyword, or topic to get help on writing
Python programs and using Python modules.  To quit this help utility and
return to the interpreter, just type "quit".

To get a list of available modules, keywords, symbols, or topics, type
"modules", "keywords", "symbols", or "topics".  Each module also comes
with a one-line summary of what it does; to list the modules whose name
or summary contain a given string such as "spam", type "modules spam".

help> modules

Please wait a moment while I gather a list of all available modules...

__future__          _weakrefset         gunicorn            reprlib
_ast                abc                 gzip                rlcompleter
_bisect             argparse            hashlib             select
_bootlocale         array               heapq               selectors
_bz2                ast                 hmac                setuptools
_codecs             asyncio             html                shelve
_collections        atexit              http                shlex
_collections_abc    base64              imaplib             shutil
_compat_pickle      binascii            imp                 signal
_compression        bisect              importlib           site
_crypt              builtins            inspect             sitecustomize
_csv                bz2                 io                  smtplib
_ctypes             calendar            ipaddress           socket
_ctypes_test        cgi                 itertools           socketserver
_datetime           cmath               json                sre_compile
_decimal            cmd                 keyword             sre_constants
_dummy_thread       code                linecache           sre_parse
_elementtree        codecs              locale              ssl
_functools          codeop              logging             stat
_hashlib            collections         lzma                string
_heapq              compileall          marshal             stringprep
_imp                concurrent          math                struct
_io                 configparser        mimetypes           subprocess
_json               contextlib          multiprocessing     symbol
_locale             copy                nntplib             sys
_lzma               copyreg             numbers             sysconfig
_markupbase         crypt               opcode              tarfile
_multiprocessing    csv                 operator            telnetlib
_opcode             ctypes              optparse            tempfile
_operator           datetime            os                  termios
_pickle             decimal             parser              textwrap
_posixsubprocess    difflib             pickle              threading
_pydecimal          dis                 pickletools         time
_pyio               distutils           pip                 token
_random             dummy_threading     pipes               tokenize
_sha256             easy_install        pkg_resources       traceback
_sha512             email               pkgutil             types
_signal             encodings           platform            unicodedata
_sitebuiltins       enum                plistlib            unittest
_socket             errno               poplib              urllib
_sre                faulthandler        posix               uu
_ssl                fnmatch             posixpath           uuid
[... 100 lines cut ...]
_sitebuiltins       faulthandler        pstats              uuid
_socket             fcntl               pty                 venv
_sqlite3            filecmp             pwd                 warnings
_sre                fileinput           py_compile          wave
_ssl                fnmatch             pyclbr              weakref
_stat               formatter           pydoc               webbrowser
_string             fractions           pydoc_data          wsgiref
_strptime           ftplib              pyexpat             xdrlib
_struct             functools           queue               xml
_symtable           gc                  quopri              xmlrpc
_sysconfigdata      genericpath         random              xxlimited
_testbuffer         getopt              re                  xxsubtype
_testcapi           getpass             readline            zipapp
_testimportmultiple gettext             reprlib             zipfile
_testmultiphase     glob                resource            zipimport
_thread             grp                 rlcompleter         zlib
_threading_local    gunicorn            runpy               
_tracemalloc        gzip                sched               
_warnings           hashlib             select              

Enter any module name to get more help.  Or, type "modules spam" to search
for modules whose name or summary contain the string "spam".

help> 
```

kraj, 2019-03-08:
This is a OE-Core bug please open a bugzilla entry https://bugzilla.yoctoproject.org/