agherzan/meta-raspberrypi #397: Kernel compilation failure

stefano-vallarino, 2019-03-07:
Hi all,

I tried to build the image at the current HEAD "7cfb0e892822f339e8a0c1f0207426cc388daa23" in the branch master, but I get a compilation error in the kernel recipe because it is looking for the commit "7f26b4456f70f9909c19936d550cf7c5dc47e1a5" which is not present in the kernel repo.

Patching the recipe to work with the current Linux version "4.19.27" it works fine.

Follows the patch, can you please verify?

diff --git a/recipes-kernel/linux/linux-raspberrypi_4.19.bb b/recipes-kernel/linux/linux-raspberrypi_4.19.bb
index 28aed98..6ebf838 100644
--- a/recipes-kernel/linux/linux-raspberrypi_4.19.bb
+++ b/recipes-kernel/linux/linux-raspberrypi_4.19.bb
@@ -1,9 +1,9 @@
 FILESEXTRAPATHS_prepend := "${THISDIR}/linux-raspberrypi:"
 
-LINUX_VERSION ?= "4.19.25"
+LINUX_VERSION ?= "4.19.27"
 LINUX_RPI_BRANCH ?= "rpi-4.19.y"
 
-SRCREV = "7f26b4456f70f9909c19936d550cf7c5dc47e1a5"
+SRCREV = "c0e09b3420442df016ee7906985535d644481e4b"
 SRC_URI = " \
     git://github.com/raspberrypi/linux.git;protocol=git;branch=${LINUX_RPI_BRANCH} \
     "

agherzan, 2019-03-07:
@stefano-vallarino looks good. Push a PR. Mind the commit log (check git log for reference https://github.com/agherzan/meta-raspberrypi/commit/7cfb0e892822f339e8a0c1f0207426cc388daa23).

agherzan, 2019-03-08:
This should be fixed now in master. Thanks for reporting it.

closed 2019-03-08 by agherzan