agherzan/meta-raspberrypi #400: Can't enable UART1 on CM3

DonMorr, 2019-03-11:
**Description**
If I try to enable uart1 in the config.txt on my Compute Module 3, the device is not enabled and the  kernel log reports an error. uart0 works as expected on pins GPIO_14 and GPIO_15.
I've reproduced this issue on the heads of sumo, thud and master branches.
This is not an issue on the official Raspian Image, when using this, both uarts can be enabled simultaneously on different sets of GPIO pins..

**Steps to reproduce the issue:**
1. Modify the local.conf of a fresh checkout of the required layers, to include the following line: CMDLINE_append += " console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 "
Also set the MACHINE to "raspberrypi-cm3".
2. Run the bitbake build: **bitbake core-image-minimal**
3. Copy the generated rpi-sdimg file to the target.
4. Mount the file targets' file systems, and add the following to /boot/config.txt:
```
core_freq=250                                                                   
dtoverlay=pi3-disable-bt
dtparam=uart1=on                                                                
dtoverlay=uart1,txd1_pin=40,rxd1_pin=41
enable_uart=1  
```
This should enable uart1 to operate on pins 40 and 41.
5. Boot the target.

**Describe the results you received:**
When the target boots, the following is currently observed:
- The device file for uart1, */dev/ttyS0*,  is not created.
- The following errors are reported by dmesg:
[    0.721626] uart-pl011 3f201000.serial: cts_event_workaround enabled
[    0.721766] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 81, base_baud = 0) is a PL011 rev2
[    1.703800] bcm2835-aux-uart 3f215040.serial: unable to register 8250 port - -28
[    1.711224] bcm2835-aux-uart: probe of 3f215040.serial failed with error -28
[    1.783593] uart-pl011 3f201000.serial: no DMA platform data

**Describe the results you expected:**
- The device file for uart1, */dev/ttyS0*,  to be created.
- No errors reported in dmesg output.

**Additional information you deem important (e.g. issue happens only occasionally):**
- This issue occurs across the heads of sumo, thud and master branches.
- Does not occur on the latest official Raspian image.
- uboot is not enabled.

```
#
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
# to the system might want to change but pretty much any configuration option can
# be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file
# but new users likely won't need any of them initially.
#
# Lines starting with the '#' character are commented out and in some cases the
# default values are provided as comments to show people example syntax. Enabling
# the option is a question of removing the # character and making any change to the
# variable as required.

#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# There are also the following hardware board target machines included for 
# demonstration purposes:
[... 210 lines cut ...]
#
# By default qemu will build with a builtin VNC server where graphical output can be
# seen. The two lines below enable the SDL backend too. By default libsdl2-native will
# be built, if you want to use your host's libSDL instead of the minimal libsdl built
# by libsdl2-native then uncomment the ASSUME_PROVIDED line below.
PACKAGECONFIG_append_pn-qemu-system-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
#ASSUME_PROVIDED += "libsdl2-native"

# CONF_VERSION is increased each time build/conf/ changes incompatibly and is used to
# track the version of this file when it was generated. This can safely be ignored if
# this doesn't mean anything to you.
CONF_VERSION = "1"

#
# Parallel Build - make the most of your CPUs
#
PARALLEL_MAKE = " \
-j 10 \
-l ${@int(os.sysconf(os.sysconf_names['SC_NPROCESSORS_ONLN'])) * 150/100} \
"

CMDLINE_append += " console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 "

```

kraj, 2019-03-11:
Add `ENABLE_UART = "1"` to `local.conf`

DonMorr, 2019-03-12:
Hi @kraj, thank you for the suggestion. I tested this on head of master, but no luck, I'm still getting the same errors.
Any other suggestions? Is there anyone free to test this with their Compute Module 3?
Cheers,
Donal

agherzan, 2019-03-13:
I've never tested uart1 on cm3. Can you try without `dtoverlay=pi3-disable-bt`?

DonMorr, 2019-03-15:
@agherzan , just tried that again without `dtoverlay=pi3-disable-bt`, but the same result for uart1:
```
[    1.706084] bcm2835-aux-uart 3f215040.serial: unable to register 8250 port - -28                                                                    
[    1.713529] bcm2835-aux-uart: probe of 3f215040.serial failed with error -28   
```
I have also tried uart1 on pins 32 and 33, but the same issue.

Any further suggestions?

My additions to config.txt now looks like the following:
```
# Enable UART                                            
enable_uart=1
core_freq=250                                  
dtparam=uart1=on
dtoverlay=uart1,txd1_pin=40,rxd1_pin=41
```

DonMorr, 2019-03-15:
My current suspicion is that this is a kernel configuration issue. Tracing the issue through `8250_core.c`, it looks to be caused by the `CONFIG_SERIAL_8250_RUNTIME_UARTS` kernel configuration to being set to 0.
If I manually override this in the kernel config, I can get /dev/ttyS0 to be created with no errors when the device is being registered.
However, I can't appear get data to be transmitted successfully through /dev/ttyS0. Will continue to do some more testing.

agherzan, 2019-03-15:
That is interesting @DonMorr . Keep us updated.

DonMorr, 2019-03-22:
@agherzan What's the best way to test a kernel configuration? I'd like to try to narrow the issue down to either the kernel configuration or something in userland. But can't seem to get the yocto build to use the kernel config from the Raspian image (which uart1 works correctly on).
I've tried adding a `do_configure_append` to my `linux-raspberrypi_%.bbappend`, which gets called, but the config (~/yocto-master-rpi/build/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/linux-raspberrypi/1_4.19.27+gitAUTOINC+c0e09b3420-r0/linux-raspberrypi_cm3-standard-build/.config) gets overridden later in the build.
```
do_configure_append() {
    bbwarn "Using custom kernel configuration"

    if [ "${S}" != "${B}" ] && [ -f "${S}/.config" ] && [ ! -f "${B}/.config" ]; then
        mv "${S}/.config" "${B}/.config"
    fi

    cp "${WORKDIR}/custom_config" "${B}/.config"
}
```

agherzan, 2019-03-22:
If it's just temporary I would probably use a devshell or copy something over just before compile.

agherzan, 2019-04-17:
@DonMorr can you try without `enable_uart=1`?

DonMorr, 2019-04-17:
@agherzan Thanks for your suggestion. Just tried without that, but getting the same result. My config.txt now only has the following related to uart1:

```
core_freq=250
dtparam=uart1=on
dtoverlay=uart1,txd1_pin=40,rxd1_pin=41
```
Getting the same error:
```
root@raspberrypi-cm3:~# dmesg | grep uart
[    0.721599] uart-pl011 3f201000.serial: cts_event_workaround enabled
[    1.704379] bcm2835-aux-uart 3f215040.serial: unable to register 8250 port - -28
[    1.711814] bcm2835-aux-uart: probe of 3f215040.serial failed with error -28
[    1.784219] uart-pl011 3f201000.serial: no DMA platform data
```

agherzan, 2019-04-18:
@DonMorr What I actually meant but didn't come out right was to use `enable_uart=0`. Sorry for the confusion.

s-maynard, 2019-04-25:
I use both UARTs in my yocto build for the CM3.  I had to use the following overlay:

/*
 * Device tree overlay for D600 UARTs
 *
 * Compile:
 *         dtc -@ -I dts -O dtb -o d600-uarts-overlay.dtb d600-uarts-overlay.dts
 */

/dts-v1/;
/plugin/;

/{
    compatible = "brcm,bcm2708";

    fragment@0 {
        target = <&uart0>;
        __overlay__ {
            pinctrl-names = "default";
            pinctrl-0 = <&uart0_pins>;
            status = "okay";
        };
    };

    fragment@1 {
        target = <&uart1>;
        __overlay__ {
            pinctrl-names = "default";
            pinctrl-0 = <&uart1_pins>;
            status = "okay";
        };
    };

    fragment@2 {
        target = <&gpio>;
        __overlay__ {
            uart0_pins: uart0_pins {
                brcm,pins = <36 37 38 39>; /* UART0: TXD0 RXD0 RTS0 CTS0 */
                brcm,function = <6>; /* alt2 */
                brcm,pull = <0 2 0 2>;
            };
        };
    };

    fragment@3 {
        target = <&gpio>;
        __overlay__ {
            uart1_pins: uart1_pins {
                brcm,pins = <14 15>; /* UART1: TX1 RX1 */
                brcm,function = <2>; /* alt5 */
                brcm,pull = <0 2>;
            };
        };
    };

    fragment@4 {
        target-path = "/chosen";
        __overlay__ {
            bootargs = "8250.nr_uarts=1";
        };
    };

    __overrides__ {
        txd0_pin = <&uart0_pins>,"brcm,pins:0";
        rxd0_pin = <&uart0_pins>,"brcm,pins:4";
        rts0_pin = <&uart0_pins>,"brcm,pins:0";
        cts0_pin = <&uart0_pins>,"brcm,pins:4";
        txd1_pin = <&uart1_pins>,"brcm,pins:0";
        rxd1_pin = <&uart1_pins>,"brcm,pins:4";
    };
};

I'm obviously on different pins, but you can modify that and recreate your dtb then load like this (in your config.txt):

# Enable UARTs
enable_uart=1
dtoverlay=pi3-disable-wifi
dtoverlay=pi3-disable-bt
dtoverlay=d600-uarts

DonMorr, 2019-04-29:
@agherzan @s-maynard Cheers guys, will give both of those a go.

DonMorr, 2019-04-30:
@agherzan Setting `enable_uart=0` disables all uart functionality for me (inc UART0).

@s-maynard That worked a treat, thank you very much for sharing.

I had missed that there is an overlay for uart1 in the repo here:
`arch/arm/boot/dts/overlays/uart1-overlay.dts`, which could be patched to use different pins as required.
@agherzan Is it worth putting details of this in the documentation [here](https://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#enable-uart)?

Thanks to everyone for their help, very much appreciated.

agherzan, 2019-04-30:
@DonMorr That is definitely something to push in the docs. Do you want to have a PR for it?

DonMorr, 2019-04-30:
@agherzan yes, happy to do this over the next few days.

agherzan, 2019-04-30:
Appreciated.

agherzan, 2019-07-25:
@DonMorr Any updates on this?

stevep-korg, 2019-09-03:
My organization has what seems to be the same problem using uart1 on a CM3/CM3+.  We only have observed the problem since we switched over to testing 4.19.66.  The 4.14.112 code we were previously using does not exhibit the problem.

@s-maynard Thanks for posting what may be a possible workaround.  What is the meaning of the d600 in the name of your overlay file; does it hav some specific meaning to your work or your environment, or is this something related to the Raspberry Pi with which I am unfamiliar?

stevep-korg, 2019-09-03:
By the way, the following entried in config.txt were sufficient in 4.14.112 kernels running on CM3/CM3+ for allowing use of both uart0 and uart1:

#  Tell Linux to not drive serial Console on GPIO pins 14 & 15
enable_uart=1
#  Disable Bluetooth (frees up UART0 and connects UART0 TXD0 and RXD0 pins to GPIO pins 14 & 15, respectively)
dtoverlay=pi3-disable-bt
# CONFIGURE UART1 FOR LEGACY MIDI USE, connecting UART1 TXD1 and RXD1 pins to GPIO pins 32 & 33, respectively
dtoverlay=uart1,txd1_pin=32,rxd1_pin=33
# midi-uart1: fake out tty driver so that baudrate 38400 yields 31250 for UART1
dtoverlay=midi-uart1

This is evidently not sufficient in 4.19 flavors.  It sure would be good to understand why.

s-maynard, 2019-09-03:
> My organization has what seems to be the same problem using uart1 on a CM3/CM3+. We only have observed the problem since we switched over to testing 4.19.66. The 4.14.112 code we were previously using does not exhibit the problem.
> 
> @s-maynard Thanks for posting what may be a possible workaround. What is the meaning of the d600 in the name of your overlay file? - Does it have some specific meaning to your work or your environment, or is this something related to the Raspberry Pi with which I am unfamiliar?

apologies - the d600 is a product name for my company; I should have generalized the comment for all to consume...

s-maynard, 2019-09-03:
PS. I am still on 4.14.112

stevep-korg, 2019-09-03:
@s-maynard Thanks much for the clarification about d600 potential meaning, and also for reporting that you are using 4.14.112.  It seems that my issue of losing the ability to use uart1 in 4.19.66 as compared to 4.14.112 is a different problem than this issue 400, which is just related to having combined acceptable settings for uart1 in the overlays and config.txt, which does seem to be more of a doc issue.  I may have to consider opening a separate issue.  Thanks much again.

stevep-korg, 2019-09-03:
@DonMorr Were you able to get uart0 and uart1 this to work in 4.19?  (I noticed above your commentscontained "linux-raspberrypi/1_4.19.27+gitAUTOINC+c0e09b3420-r0/").

Can anyone get it to work on 4.19.66? We can not, yet both work beautifully on 4.14.112.

pbrkr, 2019-11-14:
The original issue seems to be resolved and there's a new issue open for the problem with the more recent kernel. Closing this.

closed 2019-11-14 by pbrkr