agherzan/meta-raspberrypi #401: WiringPi requires update to support CM3+

DonMorr, 2019-03-15:
The WiringPi recipe needs to be updated to support the CM3+.
When the `gpio` command is used on the CM3+, an error is reported on the board type.
The WiringPi repos has been updated on the 13th/14th of march to support the CM3+

**Steps to reproduce the issue:**
1. Add wiringpi to the IMAGE_INSTALL
2. Build the image
3. Execute the gpio command

**Describe the results you received:**
```
root@raspberrypi-cm3:~# gpio readall
Oops - unable to determine board type... model: 16
```

**Describe the results you expected:**
The gpio command reads the pin modes/values without reporting any error.

**Additional information you deem important (e.g. issue happens only occasionally):**
I tried a quick fix by updating the 'SRCREV' in 'wiringpi_git.bb', but the build fails when the '0001-Add-initial-cross-compile-support.patch' patch is being applied. 

**Additional details (revisions used, host distro, etc.):**
Tested on master: 1b990fb8e2c21fdfac97c72d471e658087ce7026 (7th March).

agherzan, 2019-03-19:
We will need to migrate that patch.

DonMorr, 2019-06-13:
@agherzan, I have a fix ready for this. When I'm creating the pull request, which branch(s) should I create the pull request to?
Also, what's the policy on keeping the older branches up to date?

agherzan, 2019-06-13:
@DonMorr That is great news. The first PR needs to go against master. For backports, new PRs can be raised as soon as the initial one gets merged against the required branch support.

agherzan, 2019-07-25:
@DonMorr A friendly reminder for this.

pbrkr, 2019-11-14:
WiringPi is dead after e050601d868df6f7b38812ca930e30a6a4cff135, closing this.

closed 2019-11-14 by pbrkr