agherzan/meta-raspberrypi #402: FFMpeg override

TomMettam, 2019-03-20:
On Raspberry Pi, FFMPEG can provide hardware acceleration through the use of --enable-omx and --enable-omx-rpi compile-time switches.

It would make sense for meta-raspberrypi to override the recipe in oe-core to add these compile-time switches to FFMPEG.

kraj, 2019-03-20:
It’s better to write bbappends to express these
In this layer itself and apply the package config using ‘rpi’ overrides

agherzan, 2019-03-21:
I subscribe to what @kraj said. I'm also OK with adding them in the BSP layer as long as you feel like pushing and testing a PR.

TomMettam, 2019-04-07:
I have a working bbappend for this, however it breaks because for some reason the openmax libraries are in /usr/lib/ rather than the expected /opt/vc/lib/ - any clue why this might be?

kraj, 2019-04-08:
thats normal for this libs to be in /usr/lib so make it look into standard paths.

TomMettam, 2019-04-08:
But it's not normal for raspberry pi - userland libraries are expected to be under /opt/vc/lib, and I believe it's hardcoded in ffmpeg!

TomMettam, 2019-04-08:
Yep... https://gyazo.com/74b7141b4d088c6ee0b4c136d3f1fc27

kraj, 2019-04-08:
Right you need to patch ffmpeg

TomMettam, 2019-04-08:
But every single other OS which implements the raspberry pi userland puts these libraries in /opt/vc/lib. And actually,  having these libraries in /usr/lib breaks the omx samples too.

agherzan, 2019-07-25:
@TomMettam Any updates on this?

pbrkr, 2019-11-14:
@agherzan I recommend we close this due to lack of follow-up.

closed 2019-11-15 by agherzan