agherzan/meta-raspberrypi #403: Question: Ethernet stability issues on raspberrypi3 B+

manszigher, 2019-03-21:
**Description**

I am trying to build an image for the raspberrypi3 B+ I am currently using my own distro because I need systemd. I am using my own image. I have a python application exposing a cli over the USB to my host.

**Describe the results you received:**
I have some stability issues with my image I am continuously seeing

<pre>
Mar 15 05:39:07 raspberrypi3 kernel: lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
Mar 15 05:39:07 raspberrypi3 kernel: lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
Mar 15 05:39:07 raspberrypi3 kernel: lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000098. ret = -19
Mar 15 05:39:07 raspberrypi3 kernel: lan78xx 1-1.1.1:1.0 eth0: Failed to write register index 0x00000098. ret = -19
Mar 15 05:39:07 raspberrypi3 kernel: lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000098. ret = -19
Mar 15 05:39:07 raspberrypi3 kernel: lan78xx 1-1.1.1:1.0 eth0: Failed to write register index 0x00000098. ret = -19
Mar 15 05:39:07 raspberrypi3 kernel: lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
Mar 15 05:39:07 raspberrypi3 kernel: lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
</pre>

Sometime I am getting a kernel panic all of it looks like it is related to the ethernet driver.

**Question**
Since I am using my own distro even though I am basing it on poky and I am using my own image what is the recommended way to build an image for raspberrypi3 that is stable. I am asking because this is not an issue that I am seeing when using raspbian so I suspect it might have something to do with my build even tough I don't understand how that can result in kernel panic unless there is a bug in the kernel.

agherzan, 2019-03-21:
All you need to do is set:

> DISTRO_FEATURES_append = " systemd"      
> VIRTUAL-RUNTIME_init_manager = "systemd"                        
> DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"

kraj, 2019-03-21:
Use Yoe distro 

https://github.com/YoeDistro/yoe-distro

set DISTRO = "yoe-glibc-systemd-wayland"

closed 2019-03-22 by manszigher