agherzan/meta-raspberrypi #408: [Question] experiencing kernel stability issues raspberrypi3 B+ 32 bit build

manszigher, 2019-04-08:
Hi,

I am experiencing stability issues on my build I have distro based on poky and an image based on core-image-base.bb and I am currently on the latest sumo. I am seeing issues related to the ethernet as I have earlier posted a question about it here https://github.com/agherzan/meta-raspberrypi/issues/403. I suspect that the meta-raspberrypi is tested and used so if this was an actual issue that is known it should been reported. Sometimes when leaving the device alone I get a core dump and it is unresponsive. Since what I am experiencing is related to the kernel is there any difference to the kernel that meta-raspberrypi are using compared to raspbian? Or do you have any other pointer to what I am doing wrong  that can explain my stability issues? Any help to get me in the right direction is much appreciated.

BR

kraj, 2019-04-08:
Can you add more info e.g. stack traces etc.

manszigher, 2019-04-09:
I will try and collect any information when it occurs and post it here. I need to tweak my build so I can get more information out of it. The issues is not occurring that often. Did not want to file an issue because no one else have reported it. If you think it makes sense I can collect the information and create an issue instead of having it as question. I just wanted to know if I should investigate my yocto build to tweak it in some way that could potentially explain it some driver that I am missing or some boot configuration or some firmware?

agherzan, 2019-04-16:
I see you also compare outcome with raspbian. Do you know the exact kernel version you tested it on Raspbian?

manszigher, 2019-04-17:
Hi, I will supply some more data but here is some interesting output.

<pre>
[14844.668909] hid-generic 0003:0525:B4A4.0006: hiddev96,hidraw2: USB HID v1.10 Device [FSL USB download gadget] on usb-3f980000.usb-1.3/input0
[14847.795346] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[14847.806809] usb 1-1: USB disconnect, device number 2
[14847.817663] usb 1-1.1: USB disconnect, device number 3
[14847.828701] usb 1-1.1.1: USB disconnect, device number 7
[14847.839991] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
[14847.850443] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
[14847.860718] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000098. ret = -19
[14847.870900] lan78xx 1-1.1.1:1.0 eth0: Failed to write register index 0x00000098. ret = -19
[14847.880521] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
[14847.890251] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000098. ret = -19
[14847.899297] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
[14847.908389] lan78xx 1-1.1.1:1.0 eth0: Failed to write register index 0x00000098. ret = -19
[14847.916950] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
[14847.925588] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
[14847.934810] failed to kill vid 0081/0 for device eth0
[14847.968529] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[14848.016478] usb 1-1.1.3: USB disconnect, device number 5
[14848.606878] usb 1-1.2: USB disconnect, device number 4
[14848.622423] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[14848.630203] cp210x 1-1.2:1.0: device disconnected
[14848.642206] usb 1-1.3: USB disconnect, device number 12
[14848.855316] Indeed it is in host mode hprt0 = 00001501
[14849.065268] usb 1-1: new high-speed USB device number 13 using dwc_otg
[14849.073141] Indeed it is in host mode hprt0 = 00001101
[14849.315513] usb 1-1: New USB device found, idVendor=0424, idProduct=2514
[14849.323489] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[14849.334892] hub 1-1:1.0: USB hub found
[14849.343496] hub 1-1:1.0: 4 ports detected
[14849.665272] usb 1-1.1: new high-speed USB device number 14 using dwc_otg
[14849.795518] usb 1-1.1: New USB device found, idVendor=0424, idProduct=2514
[14849.803857] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[14849.816332] hub 1-1.1:1.0: USB hub found
[14849.826558] hub 1-1.1:1.0: 3 ports detected
[14849.935276] usb 1-1.2: new full-speed USB device number 15 using dwc_otg
[14850.103002] usb 1-1.2: New USB device found, idVendor=10c4, idProduct=ea60
[14850.111116] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[14850.119577] usb 1-1.2: Product: CP2102 USB to UART Bridge Controller
[14850.127746] usb 1-1.2: Manufacturer: Silicon Labs
[14850.136171] usb 1-1.2: SerialNumber: 0001
[14850.154674] cp210x 1-1.2:1.0: cp210x converter detected
[14850.166683] usb 1-1.1.3: new low-speed USB device number 16 using dwc_otg
[14850.180235] usb 1-1.2: cp210x converter now attached to ttyUSB0
[14850.425274] usb 1-1.3: new high-speed USB device number 17 using dwc_otg
[14850.468876] usb 1-1.1.3: New USB device found, idVendor=03f0, idProduct=034a
[14850.477162] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[14850.485744] usb 1-1.1.3: Product: HP Elite USB Keyboard
[14850.494019] usb 1-1.1.3: Manufacturer: Chicony
[14850.555841] usb 1-1.3: New USB device found, idVendor=0525, idProduct=a4a5
[14850.564279] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[14850.573005] usb 1-1.3: Product: USB download gadget
[14850.581315] usb 1-1.3: Manufacturer: FSL
[14850.589868] usb 1-1.3: SerialNumber: 17072a0a5c85d6ee
[14850.600022] input: Chicony HP Elite USB Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/0003:03F0:034A.0007/input/input2
[14850.613217] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[14850.622841] usb-storage 1-1.3:1.0: Quirks match for vid 0525 pid a4a5: 10000
[14850.676244] hid-generic 0003:03F0:034A.0007: input,hidraw0: USB HID v1.10 Keyboard [Chicony HP Elite USB Keyboard] on usb-3f980000.usb-1.1.3/input0
[14850.734214] input: Chicony HP Elite USB Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.1/0003:03F0:034A.0008/input/input3
[14850.805723] hid-generic 0003:03F0:034A.0008: input,hidraw1: USB HID v1.10 Device [Chicony HP Elite USB Keyboard] on usb-3f980000.usb-1.1.3/input1
[14851.115340] usb 1-1.1.1: new high-speed USB device number 18 using dwc_otg
[14851.245667] usb 1-1.1.1: New USB device found, idVendor=0424, idProduct=7800
[14851.254525] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[14851.543239] libphy: lan78xx-mdiobus: probed
[14851.549520] 8021q: adding VLAN 0 to HW filter on device eth0
[14851.666661] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
</pre>

After this I have lost my network and I cannot get it up again. I have a device connected to it over USB and it looks like my device affects the USB hub in some way.

manszigher, 2019-04-17:
On the raspbian I am running 4.14.79

manszigher, 2019-04-17:
I will test it out some more on the raspbian but I am not seeing this issue there so even though I do believe the device connected to raspbian might not behave nice I am pretty sure the result is not the same when running raspbian nor when running connecting the device to other devices.

agherzan, 2019-04-17:
Use serial connection and reproduce the issue. That will not depend on network.

manszigher, 2019-04-23:
@agherzan can you clarify? Not sure I understand

manszigher, 2019-04-23:
Maybe I should clarify I am using an fdti null modem cable

https://www.ftdichip.com/Support/Documents/DataSheets/Cables/DS_USBNMC.pdf

This is connected to the rpi USB and to my host. So I am not using the serial port pins.

agherzan, 2019-04-23:
What I meant is to attach a serial console and login over it to see details logs of what happens when the device breaks.

manszigher, 2019-04-26:
So I am trying to get the serial port working on the GPIO unfortunately I am struggling a bit. I have also encountered kernel panics when rebooting. My build is based on sumo, my distro is a custom based on poky

<pre>
require conf/distro/poky.conf

DISTRO = "strix-production"
DISTRO_NAME = "Strix Production Programmer"
DISTRO_VERSION = "${STRIX_OS_RELEASE}"
MOZARTOS_VERSION = "${STRIX_VERSION}"
SDK_VENDOR = "-strixsdk"
MAINTAINER = "Strixt Team"
TARGET_VENDOR = "-strix"
DISTROOVERRIDES = "strix"

# have build information available on the target under /etc/build
INHERIT += "image-buildinfo"

# mozart programmer defaults to ipk packages
PACKAGE_CLASSES ?= "package_ipk"

DISTRO_FEATURES_append = " \
    ${@bb.utils.contains('DEVELOPMENT_IMAGE','1','development-image','',d)} \
    systemd \
    usbhost \
"

SDK_VERSION := "${@'${DISTRO_VERSION}'.replace('snapshot-${DATE}','snapshot')}"

VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
</pre>

I am basing my image on core-image-base

<pre>
SUMMARY = "Programmer image used in production to flash products using uuu"
LICENSE = "CLOSED"

include recipes-core/images/core-image-base.bb

IMAGE_INSTALL += " \
    packagegroup-strix-programmer \
"

IMAGE_FEATURES += " \
    ${@bb.utils.contains('DISTRO_FEATURES', 'development-image', 'debug-tweaks', '', d)} \
"
</pre>

My local.conf file

<pre>
# Supported machines
MACHINE ?= "raspberrypi3"

# Set this to 1 if development image is desired
DEVELOPMENT_IMAGE = "1"

# Parallelism Options
BB_NUMBER_THREADS ?= "${@oe.utils.cpu_count()}"
PARALLEL_MAKE ?= "-j ${@oe.utils.cpu_count()}"

# Strix specific distros
DISTRO ?= "strix-production"

# Making sure we have 2GB extra space 
IMAGE_ROOTFS_EXTRA_SPACE = "2097152"

# Custom downloads directory
DL_DIR ?= "/home/mans/Workspace/strix-workspace/scripts/../.cache/downloads"

# Custom sstate directory
SSTATE_DIR ?= "/home/mans/Workspace/strix-workspace/scripts/../.cache/raspberrypi/sstate-cache"

# Remove the old image before the new one is generated to save disk space when RM_OLD_IMAGE is set to 1, this is an easy way to keep the DEPLOY_DIR_IMAGE clean.
RM_OLD_IMAGE = "1"

# Additional image features
USER_CLASSES ?= "buildstats image-mklibs image-prelink"

# By default disable interactive patch resolution (tasks will just fail instead):
PATCHRESOLVE = "noop"

# Disk Space Monitoring during the build
BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K"

CONF_VERSION = "1"

LICENSE_FLAGS_WHITELIST="commercial"
ACCEPT_FSL_EULA = "1"

VOLATILE_LOG_DIR = "no"
ENABLE_UART = "1"
DISABLE_RPI_BOOT_LOGO = "1"
CMDLINE_DEBUG = "1"
</pre>

my bblayer file

<pre>
# LAYER_CONF_VERSION is increased each time build/conf/bblayers.conf  
# changes incompatibly                                                
LCONF_VERSION = "7"                                                   

BBPATH = "${TOPDIR}"                                                  
MOZARTWORKDIR := "${@os.path.abspath(os.path.dirname(d.getVar('FILE', True)) + '/../../..')}"                                                

BBLAYERS ?= " \                                                       
    ${MOZARTWORKDIR}/layers/meta-strixt/meta-strix-sumo \            
    ${MOZARTWORKDIR}/layers/meta-strix/meta-strix-programmer \      
    \                                                                 
    ${MOZARTWORKDIR}/layers/poky/meta \                               
    ${MOZARTWORKDIR}/layers/poky/meta-poky \                          
    \                                                                 
    ${MOZARTWORKDIR}/layers/meta-openembedded/meta-oe \               
    ${MOZARTWORKDIR}/layers/meta-openembedded/meta-multimedia \       
    ${MOZARTWORKDIR}/layers/meta-openembedded/meta-networking \       
    ${MOZARTWORKDIR}/layers/meta-openembedded/meta-python \           
    ${MOZARTWORKDIR}/layers/meta-openembedded/meta-filesystems \      
    ${MOZARTWORKDIR}/layers/meta-openembedded/meta-webserver \        
    \                                                                 
    ${MOZARTWORKDIR}/layers/meta-raspberrypi \                        
" 
</pre>

Not much is going on in meta-strix/meta-strix-programmer and meta-strix-sumo. In meta-strix-programmer I have a bbappend on rpi-config_git

<pre>
do_deploy_append() {

    # Enable i2c by default
    echo "dtparam=i2c_arm=on" >>${DEPLOYDIR}/bcm2835-bootfiles/config.txt

    # Enable SPI by default
    echo "dtparam=spi=on" >>${DEPLOYDIR}/bcm2835-bootfiles/config.txt

    # Disable firmware splash by default
    echo "disable_splash=1" >>${DEPLOYDIR}/bcm2835-bootfiles/config.txt

    # Disable firmware warnings showing in non-debug images
    if ! ${@bb.utils.contains('DISTRO_FEATURES','development-image','true','false',d)}; then
        echo "avoid_warnings=1" >>${DEPLOYDIR}/bcm2835-bootfiles/config.txt
    fi

    # Enable audio (loads snd_bcm2835)
    echo "dtparam=audio=on" >> ${DEPLOYDIR}/bcm2835-bootfiles/config.txt

    # Setting overlay to disable BT and WIFI"
    #echo "dtoverlay=pi3-disable-bt" >> ${DEPLOYDIR}/bcm2835-bootfiles/config.txt
    #echo "dtoverlay=pi3-disable-wifi" >> ${DEPLOYDIR}/bcm2835-bootfiles/config.txt
}
</pre>

I am not seeing this issue on raspbian so it must be something in my build.

manszigher, 2019-04-26:
So the my distro was actually including more then what I posted above it was including a file that was a legacy from an old project I thought it would not mater since it should be empty but of course that was the case. By removing this include in my distro it appears as if my issue is not possible to reproduce I just need to enable every option and see what caused my issue.

closed 2019-04-26 by manszigher