agherzan/meta-raspberrypi #410: linux-raspberrypi 4.19.y is not considered stable

paul-betafive, 2019-04-11:
Looking upstream today I've seen the following issue reported by @schnitzeltony: https://github.com/raspberrypi/linux/issues/2931

On this issue it's been made clear that the rpi-4.19.y kernel branch is not considered stable and rebasing of this branch will likely continue. The rebasing is causing regular fetch errors for myself and other users.

I recommend we switch the default kernel version back to 4.14 in this layer.

@kraj, @agherzan: Any thoughts on this? It's a shame to move backwards but I don't think we can leave the default kernel on a branch that's being regularly rebased.

kraj, 2019-04-11:
We should add ```rebaseable=1``` to SRC_URI until upstream gets their act together and starts respecting default public branches.

agherzan, 2019-04-16:
What is rebaseable doing?

kraj, 2019-04-16:
it keeps the old SHA along even on refresh it will find the old SHA

paul-betafive, 2019-04-17:
Setting `rebaseable=1` doesn't fix things for us as we're not maintaining a downloads mirror.

PR #414 (2019-04-24): Update 4.14.y kernel & make default, drop 4.9.y kernel
The linux-raspberrypi 4.14 recipe is updated to include my recent merge of the 4.14.112 stable kernel release. This version is then made the default as the upstream rpi-4.19.y branch is still considered unstable.

The linux-raspberrypi 4.9 recipe is dropped as it's now rather obsolete.

Fixes #410
closed 2019-04-24 by agherzan