agherzan/meta-raspberrypi #415: [Question] ENABLE_UART="1" in local.conf will not enable cmdline console arg

manszigher, 2019-04-26:
Hi,

I am trying to get the serial console to work on the GPIO pins on raspberrypi3 B+ without any success. I am sure I am doing something wrong and I have gone through all the issues here but without any luck. One of the things that I don't understand is that according to the documentation I should add ENABLE_UART="1" to the local.conf which I have I have verified that enable_uart=1 in config.txt is there but the cmdline is not updated. 

https://github.com/agherzan/meta-raspberrypi/blob/7059c374512f1c1c0df9ecab0703d11438bdf78b/recipes-kernel/linux/linux-raspberrypi.inc#L24

<pre>
SERIAL = "${@oe.utils.conditional("ENABLE_UART", "1", "console=serial0,115200", "", d)}"
CMDLINE ?= "dwc_otg.lpm_enable=0 ${SERIAL} root=/dev/mmcblk0p2 rootfstype=ext4 rootwait"
</pre>

https://github.com/openembedded/openembedded-core/blob/ff578f4451a0a199202e576b647840910b4d3f59/meta/lib/oe/utils.py#L22

So why is oe.utils.conditional not finding ENABLE_UART? Am I doing something wrong?

manszigher, 2019-04-26:
So i dumped the env for linux-raspberrypi and noticed that ENABLE_UART actually was enabled after modifying the SERIAL to always set "console=serial0,115200" so I think the issue was that I did not do a clean of linux-raspberrypi after adding ENABLE_UART in local.conf.

manszigher, 2019-04-26:
So this is what I get when running bitbake -e linux-raspberrypi

<pre>
CMDLINE="dwc_otg.lpm_enable=0 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait    "
</pre>

But when booting up the console is missing in the cmdline.txt

manszigher, 2019-04-26:
The file under ./tmp/deploy/images/raspberrypi3/bcm2835-bootfiles/cmdline.txt also includes the console arg.

agherzan, 2019-04-26:
Can you mount the final image and check the cmdline.txt? Have you regenerated the image and burn the right one?

manszigher, 2019-04-26:
Well let me just say that I have tried to verify it accordingly but since it was still not working I must be messing something up. I will post a comment when I figures this out. Thanks anyway.

agherzan, 2019-06-02:
@manszigher Any news on this?

manszigher, 2019-06-03:
Sorry yes I just had to make sure that after enable it that everything was built as expected I think there was a cache interfering resulting it after making sure everything was built as expected I got it up and working.

closed 2019-06-03 by manszigher