agherzan/meta-raspberrypi #422: Raspberrypi3-64 fails on model 3A+

bvarner, 2019-05-29:
**rpi 3A+ boots properly for raspberrypi3, but not raspberrypi3-64**

I'm building an image that boots with my 3A+ when `MACHINE=raspberrypi3`.
The exact same config, but with `MACHINE=raspberrypi3-64` makes it through the uboot phase, then just sits there after uboot says it's loading the kernel.

The 3A+ is the only 'new' pi I have. Same image works with my other devices, which are all `raspberrypi` or `raspberypi0-wifi`.

**Steps to reproduce the issue:**
Build an image for `raspberrypi3-64`, try to boot it on a 3A+.

**Describe the results you received:**
An unexpected hang - kernel fails to load weather using uBoot or not.

**Describe the results you expected:**
I expected it to boot, like it does when I use `raspberrypi3`... :-)

**Additional details (revisions used, host distro, etc.):**
I'm using the 'warrior' branches, and I've tried with both kernel 4.14 and 4.19. No luck in any case.

agherzan, 2019-05-29:
Can you try current master for meta-raspberrypi with the warrior branches on the other layers? We've updated the kernel in the meanwhile.

bvarner, 2019-06-26:
Current build config and layers:
```
Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7"
TUNE_FEATURES        = "aarch64 cortexa53 crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:f2b442c5c8e3925a4d3fc9693f1e47863dade2a9"
meta-oe              
meta-initramfs       
meta-python          
meta-webserver       = "warrior:d7338c2b3a96ef7d8bf9df59dcde11f8c131f9c0"
meta-raspberrypi     = "master:8636b6375251ea9553639dcb8bdd985e54c4bc36"
```
Without uboot: I get the rainbow screen, some green activity light flashing, and nothing out the HDMI.

With uboot: I get a uboot prompt on HDMI and can use USB devices for the console input. After that, same when I boot without uboot.

I was attempting to use:
`LINUX_KERNEL_TYPE = "preempt-rt"`

So maybe that's the issue. I've removed that, and will see what happens now.

(preempt-rt worked under sumo on a `raspberrypi0-wifi`)

bvarner, 2019-06-26:
Removed preempt-rt preference, still nothing.

bvarner, 2019-06-26:
Removing preempt-rt didn't work.
Still unable to build a bootable raspberrypi3-64 image that boots on an A+ with yocto warrior.

.

agherzan, 2019-06-26:
@bvarner can you have a bare minimum setup (poky + meta-rpi), core-image-minimal, and test? Also, what kernel version did you end up using when you tested the above?

bvarner, 2019-07-06:
Previous test would have been kernel Version 4.19.. not sure beyond that.
Waiting on the requested core-image-minimal with restricted layer config.

bvarner, 2019-07-10:
Current build:

```
$ bitbake --setscene-only core-image-minimal
Loading cache: 100% |###############################################################################################################################################################################| Time: 0:00:00
Loaded 3079 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.43.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7+snapshot-20190710"
TUNE_FEATURES        = "aarch64 cortexa53 crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:67266331b0f557c01cde1cc1b1a1de7197443a6c"
meta-oe              
meta-python          = "master:64974b8779291419486338f75f229a732e450d78"
meta-raspberrypi     = "master:8d5fc54f41dab42d765644ea953bf451ced78f74"

Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 0 Found 0 Missed 0 Current 544 (0% match, 100% complete)
NOTE: Executing SetScene Tasks
NOTE: Tasks Summary: Attempted 546 tasks of which 546 didn't need to be rerun and all succeeded.
```
Current kernel image:
`Image-1-4.19.56+git0+fba6ea6a8f-r0-raspberrypi3-64-20190706234746.bin`

The pi 3A+ is booting. I can plug in a usb mouse and keyboard, and use ctrl-alt-delete to reboot the machine. The color-test shows up on HDMI output, as does uboot if I'm using it. But once the kernel is loading starts, I don't see anything via the HDMI output.

This is consistent with what I saw before, but I had assumed it wasn't booting at all. It is, it's just not got any output on the HDMI. Does the core-image-minimal still setup serial consoles on the UART pins (assuming the proper boot settings are applied)? If so, I could set up a serial debugger...

kraj, 2019-07-11:
Please try
https://github.com/agherzan/meta-raspberrypi/pull/436

agherzan, 2019-07-11:
The above mentioned PR was merged to master. Give that a go please.

bvarner, 2019-07-11:
I was really hoping this would fix it, but the same thing is happening.
Works fine for 32bit raspberrypi3, but raspberrypi3-64 doesn't output video on the HDMI.

Updated to:
`meta-raspberrypi     = "master:89401374988a507390546757d135ed31bcdb9d5b"`

Current Kernel Image:
Image-1-4.19.57+git0+55b89745a4-r0-raspberrypi3-64-20190711135303.bin

How can I help, aside from build / test / report? What other info or diagnostics do we need and how can I go about getting them? ;-)

kraj, 2019-07-12:
hmm we could use `fkms` set 
```
VC4DTBO = "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "vc4-fkms-v3d", "vc4-kms-v3d", d)}"
```

and see if that works

bvarner, 2019-07-12:
Interesting... This makes me wonder if it's my RAM split config.
Since I was running this image headless (and I've been building it for pi zero w's before getting a 3A+) I wasn't enabling the vc driver, and was setting the memory split to 16MB for the GPU.

I see now that in raspberrypi3-64.conf there's an
`MACHINE_FEATURES_append = " vc4graphics"`
and that my boot config has that dtbo loading, but the `gpu_mem` set to 16.

I'll increase my gpu mem and see if that fixes things.

bvarner, 2019-07-12:
I've tried adjusting the gpu_mem setting, but it seemed irrelevant.
I removed the MACHINE_FEATURES_append = " vc4graphics" from my local raspberrypi3-64.conf, and how it's booting to a console like I'd expect, and like it did with the raspberrypi3.conf.

So, it looks like the vc4graphics isn't working on the 3A+.

agherzan, 2019-07-25:
@bvarner can you retest with the current warrior HEAD?

bvarner, 2019-07-30:
Current bitbake setup.

Results are the same. Sorry for the delay in retesting.

```
Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7.1"
TUNE_FEATURES        = "aarch64 cortexa53 crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:51223035e66c5b27799b15a0f8ebe75d24d3971b"
meta-oe              
meta-initramfs       
meta-python          
meta-webserver       = "warrior:8d5dcd6522e9d15e68637b6d7dda0401f9bb91d0"
meta-raspberrypi     = "warrior:e9e3ad1920e1a6f4a05210c2ec4b77530871a886"
```

kraj, 2019-07-30:
@bvarner try `master` for all to see if we have a fix on master. Warrior is last release and there could be fixes to backport.

bvarner, 2019-08-07:
Tried again, on `master`.
```
Build Configuration:
BB_VERSION           = "1.43.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7+snapshot-20190807"
TUNE_FEATURES        = "aarch64 cortexa53 crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:841a1029c31192608c7222db4e05713053b266d0"
meta-oe              
meta-initramfs       
meta-python          
meta-webserver       = "master:80e0113f7a31447191204e1a1bda9c12e4ac693b"
meta-raspberrypi     = "master:9d0935c9bb1309431e62b8f8341eb503653e5ff5"
```

Same results.

Adding to local.conf:
```
MACHINE_FEATURES_remove = "vc4graphics"
```
Results in the console displaying appropriately.
Seems there must be some difference between the 3B+ and the 3A+ with the graphics driver?

kraj, 2019-08-07:
I would think there should be no differences in userspace at all.

bvarner, 2019-08-08:
I don't know -- I don't have a 3B+ to test with. I'm assuming you (or someone else here) does and that the 64 bit build works there, too.

It's only on the `-64` machine that this issue shows up, and only when I enable `vc4graphics`.

bvarner, 2019-11-14:
I saw the recent change to DISABLE_VC4GRAPHICS, so figured I'd retest.

For my purpose with the 3A+, I don't need vc4, but it would be nice to have working properly.
I've re-tried the following permutations this afternoon:
```
GPU_MEM = "32"
```
Machine boots. Nothing outputs on the HDMI port after the kernel starts loading. (No Kernel Console)

```
GPU_MEM = "32"
VC4DTBO = "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "vc4-fkms-v3d", "vc4-kms-v3d", d)}"
```
Same Result.

```
GPU_MEM = "64"
```
I was hoping the increase to 64 would help. It did not. Same Result.
```
GPU_MEM = "64"
VC4DTBO = "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "vc4-fkms-v3d", "vc4-kms-v3d", d)}"
```
Tried again with the fkms driver. Same result. Machine boots, no HDMI output.

Rebuilding with `DISABLE_VC4GRAPHICS="1"` inside local.conf results in the expected console output. I tested this with GPU_MEM as low as 16.

pbrkr, 2019-11-14:
Sorry for the dumb question but have you tried Raspbian on this board? Or tried the image on a different board? I'd like to rule out hardware damage here.

bvarner, 2019-11-14:
@pbrkr Not at all a stupid question!

Yes! In fact, here's a copy of glxinfo with the vc4 drivers enabled on my pi3a+.
The raspbian image here is one I use on my lab of pi4s. I literally took the card out of a pi4 and booted the 3a+. glxgears runs smooth as silk.

[glxinfo.txt](https://github.com/agherzan/meta-raspberrypi/files/3847364/glxinfo.txt)

While trying to figure out what to do next, I stumbled across [issue 1082](https://github.com/raspberrypi/firmware/issues/1082) from the raspberrypi firmware git repo, which was closed.

The [final comment](https://github.com/raspberrypi/firmware/issues/1082#issuecomment-553152650) from pelwell on there seems again to point the finger at something in our yotco board support.

bvarner, 2019-11-14:
I Found it!

And I feel like an idiot.

`core-image-minimal` does not include `kernel-modules`. When you use an image that does, it works.

Sigh. So the problem wasn't with yocto, or the meta-raspberrypi layer, but with the image not including kernel-modules when the vc4graphics are not enabled.

I'm going to mark this as closed, because we ought to all be pulling in kernel-modules, right? ;-) (So sorry I wasted peoples time, and thanks for all the help)

closed 2019-11-14 by bvarner

pbrkr, 2019-11-14:
@bvarner I've done that before! I always advise starting from `core-image-base` now.

kraj, 2019-11-14:
@pbrkr perhaps this should be documented if we have not done it. Recommending core-image-base for general use and use core-image-minimal for special usecases like initramfs or BSP bringup or tiny images etc.

agherzan, 2019-11-15:
I've heard a lot of people getting bitten but this. Sounds like a gap in the docs.

wbochra, 2021-04-05:
so what is the image you used and it worked with vc4graphics

kraj, 2021-04-05:
> so what is the image you used and it worked with vc4graphics

core-image-weston, core-image-x11 should be good to try if you want to exercise graphics.

wbochra, 2021-04-05:
I am curentlly working with core-image-sato does is support graphics ?
another thing please how can I enabel the  vc4graphics ?

kraj, 2021-04-05:
> I am curentlly working with core-image-sato does is support graphics ?
> another thing please how can I enabel the vc4graphics ?

yes sato image should work too, vc4graphics is on by default if you are using dunfell or newer release
there is an option to disable it instead. which is to set `DISABLE_VC4GRAPHICS = "1"` in local.conf as documented here 

https://github.com/agherzan/meta-raspberrypi/blob/master/docs/extra-build-config.md#vc4

wbochra, 2021-04-05:
thank you so much that's what I did exaclt bu I got this error:
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'libgl' (but /home/ubuntu/poky/meta/recipes-gnome/gtk+/gtk+3_3.24.25.bb RDEPENDS on or otherwise requires it)
mesa-gl RPROVIDES libgl but was skipped: PREFERRED_PROVIDER_virtual/libgl set to mesa, not mesa-gl
mesa RPROVIDES libgl but was skipped: PREFERRED_PROVIDER_virtual/libgbm set to mesa-gl, not mesa
NOTE: Runtime target 'libgl' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libgl']
NOTE: Runtime target 'libnss-mdns' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libnss-mdns', 'avahi', 'gtk+3', 'libgl']
NOTE: Runtime target 'packagegroup-base-extended' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-base-extended', 'libnss-mdns', 'avahi', 'gtk+3', 'libgl']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-base-extended', 'libnss-mdns', 'avahi', 'gtk+3', 'libgl']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

wbochra, 2021-04-05:
if it is enabeld by default how can I activate it and work with it ?? I am using raspberry pi3 with ethernet cabel

kraj, 2021-04-05:
we have
```
conf/machine/include/rpi-default-providers.inc:PREFERRED_PROVIDER_virtual/libgl ?= "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "mesa-gl", d)}"
```

which should set it to `mesa-gl` when you set `DISABLE_VC4GRRAPHICS = "1"`

so I wonder what your settings are

wbochra, 2021-04-06:
this is my bblayers.conf because I had tha eroor once whan i didn't add oppenembedded layers
/home/ubuntu/poky/meta \
  /home/ubuntu/poky/meta-poky \
  /home/ubuntu/poky/meta-yocto-bsp \
  /home/ubuntu/meta-raspberrypi \
  /home/ubuntu/meta-openembedded/meta-oe \
  /home/ubuntu/meta-openembedded/meta-python \
  /home/ubuntu/meta-openembedded/meta-multimedia \
  /home/ubuntu/meta-openembedded/meta-networking \

wbochra, 2021-04-06:
PREFERRED_PROVIDER_virtual/libgl ?= "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "mesa-gl", d)}" here is my settings

wbochra, 2021-04-06:
I even added those layers to the bblayers.conf meta-xfce, meta-gnome and meta-perl

wbochra, 2021-04-06:
the error now :
Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'libgl' (but /home/ubuntu/poky/meta/recipes-gnome/gtk+/gtk+3_3.24.25.bb RDEPENDS on or otherwise requires it)
mesa-gl RPROVIDES libgl but was skipped: PREFERRED_PROVIDER_virtual/libgl set to mesa, not mesa-gl
mesa RPROVIDES libgl but was skipped: PREFERRED_PROVIDER_virtual/libgbm set to mesa-gl, not mesa
NOTE: Runtime target 'libgl' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libgl']
NOTE: Runtime target 'libnss-mdns' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libnss-mdns', 'avahi', 'gtk+3', 'libgl']
NOTE: Runtime target 'packagegroup-base-extended' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-base-extended', 'libnss-mdns', 'avahi', 'gtk+3', 'libgl']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-base-extended', 'libnss-mdns', 'avahi', 'gtk+3', 'libgl']

agherzan, 2021-04-06:
@wbochra can you attach your `local.conf` (without the comments), attach the `bblayers.conf` and tell us what revisions you use for the layers?

wbochra, 2021-04-06:
[conf.pdf](https://github.com/agherzan/meta-raspberrypi/files/6263800/conf.pdf)
what do you mean here ( what revisions you use for the layers?)

agherzan, 2021-04-06:
Your build uses a set of layers (including this one). What revisions do you use for all the layers included in the build?

agherzan, 2021-04-06:
I'm checking now a build on rpi3-64 with sato image using master branches.

agherzan, 2021-04-06:
I've just built for RaspberryPi3 64bit on `core-image-sato` using `master` for all layers and default configuration (`local.conf`). Please start from this and incrementally build on top.

wbochra, 2021-04-06:
@agherzan does your build secced and can please tell me how can I find the revisions in the layer.conf ??

agherzan, 2021-04-06:
Yes, it did succeed just fine. The revisions are the `git` revisions of each layer checkout you have cloned.

wbochra, 2021-04-06:
I cloned the layers from this link git://git.openembedded.org/meta-openembedded

wbochra, 2021-04-06:
@agherzan  did you write this line DISABLE_VC4GRRAPHICS = "1"  to your local.conf ?? it is the line that causes my error

agherzan, 2021-04-06:
You mean `DISABLE_VC4GRAPHICS` and no, it was left undefined.

wbochra, 2021-04-06:
yeah ofcourse the build will be just fine when you set the DISABLE_VC4GRAPHICS variabel to 1 in your local.conf the problems happens 
ERROR: Nothing RPROVIDES 'libgl' (but /home/ubuntu/poky/meta/recipes-gnome/gtk+/gtk+3_3.24.25.bb RDEPENDS on or otherwise requires it)
mesa-gl RPROVIDES libgl but was skipped: PREFERRED_PROVIDER_virtual/libgl set to mesa, not mesa-gl
mesa RPROVIDES libgl but was skipped: PREFERRED_PROVIDER_virtual/libgbm set to mesa-gl, not mesa
NOTE: Runtime target 'libgl' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libgl']
NOTE: Runtime target 'libnss-mdns' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libnss-mdns', 'avahi', 'gtk+3', 'libgl']
NOTE: Runtime target 'packagegroup-base-extended' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-base-extended', 'libnss-mdns', 'avahi', 'gtk+3', 'libgl']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-base-extended', 'libnss-mdns', 'avahi', 'gtk+3', 'libgl']

kraj, 2021-04-06:
I added `DISABLE_VC4GRAPHICS = "1"` to `conf/local.conf` and  set `MACHINE=raspberrypi4` then `bitbake core-image-sato` works, so I can not reproduce it on master branch, 

ps: I use yoe Distro so things might be different in your setup that we don't know.

kraj, 2021-04-06:
> Current build config and layers:
>
> ```
> Build Configuration:
> BB_VERSION           = "1.42.0"
> BUILD_SYS            = "x86_64-linux"
> NATIVELSBSTRING      = "universal"
> TARGET_SYS           = [...]

Please do not mix branches. your build config seems to use `warrior` release for some layers and `master` for meta-raspberrypi, you should use matching branch for all layers.

kraj, 2021-04-06:
@wbochra please post your `Build Configuration` that Bitbake prints at the top when build starts.

wbochra, 2021-04-07:
@kraj when I bitbake with DISABLE_VC4GRAPHICS = "1" in my local.conf I don't even get to the build configuration the buil crach instantelly so this is my configuration when I remove  DISABLE_VC4GRAPHICS = "1" from my local.conf which works just fine `OTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.49.2"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.2+snapshot-ea455ca8671d3bc2a1097989bfaabe92f3ca37ab"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:ea455ca8671d3bc2a1097989bfaabe92f3ca37ab"
meta-raspberrypi     = "master:d1f191ed3018e0b130311e443f014e30d2f5ed97"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      
meta-perl            
meta-gnome           
meta-xfce            = "master:ba2576ed79a09879f714ccf7f4fa32259c18ab8c"
meta-test            = "master:ea455ca8671d3bc2a1097989bfaabe92f3ca37ab"
`

wbochra, 2021-04-07:
Hi @kraj  I just changed my local.conf to a new one (default in every thing) with just DISABLE_VC4GRAPHICS = "1" and MACHINE=raspberrypi3-64 I still have the same problem

wbochra, 2021-04-07:
I am using yocto-3.2.1