agherzan/meta-raspberrypi #424: core-image-sato build fails (Nothing PROVIDES 'faad2')

AustinHaigh, 2019-06-10:
**Description**

Building poky with the sato UI (using the command `bitbake core-image-sato`) fails.

**Steps to reproduce the issue:**
1. `git clone https://github.com/agherzan/meta-raspberrypi`
2. `source ./oe-init-build-env rpi-build`
3. `bitbake-layers add-layer ../meta-raspberrypi`
4. add `MACHINE?="raspberrypi"` to `conf/local.conf`
5. `bitbake core-image-sato`

**Describe the results you received:**
The final `bitbake` command fails with the following output:

```Loading cache: 100% |########################################################################################################################################################################| Time: 0:00:00
Loaded 1339 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'faad2' (but /mnt/hdd/yocto/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.16.0.bb DEPENDS on or otherwise requires it)
NOTE: Runtime target 'gst-examples' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['gst-examples', 'gstreamer1.0-plugins-bad', 'faad2']
NOTE: Runtime target 'packagegroup-core-x11-sato' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-sato', 'gst-examples', 'gstreamer1.0-plugins-bad', 'faad2']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-core-x11-sato', 'gst-examples', 'gstreamer1.0-plugins-bad', 'faad2']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

**Describe the results you expected:**
The final `bitbake` command should successfully generate a firmware image.

**Additional details (revisions used, host distro, etc.):**
I'm using the master branch of poky and meta-raspberrypi, both pulled around 18:00 GMT 2019/06/10, and building on Ubuntu Server 18.04.2 LTS.

AustinHaigh, 2019-06-10:
I didn't add all dependencies to bblayers.conf

closed 2019-06-10 by AustinHaigh

kraj, 2019-06-13:
in your local.conf add
```
LICENSE_FLAGS_WHITELIST_append = " commercial_faad2 "
```

TonyMcKnight22, 2019-11-28:
> I forgot to add the dependencies to bblayers.conf

Hi Austin,

I met the same issue as yours when "bitbake core-image-sato".
Could you please let me know more details about how to "add the dependencies to bblayers.conf" to solve the issue?
Thanks!

Best Regards

coolicelee, 2020-01-09:
> I forgot to add the dependencies to bblayers.conf

Hi Austin
Could you please say more detail about how to add the dependencies, I have same issue.
Thank you
BR

coolicelee, 2020-01-09:
> > I forgot to add the dependencies to bblayers.conf
> 
> Hi Austin,
> 
> I met the same issue as yours when "bitbake core-image-sato".
> Could you please let me know more details about how to "add the dependencies to bblayers.conf" to solve the issue?
> Thanks!
> 
> Best Regards

Any answer?
I have same problem.

kraj, 2020-01-09:
you need `meta-oe` in `bblayers.conf` and add it license whitelist as I suggested above