agherzan/meta-raspberrypi #425: linux-raspberrypi 4.19 cannot be fetched on warrior branch

gaurav-xebrium, 2019-06-12:
**Description**
Unable to build linux-raspberry-4.19 on warrior branch for machine `raspberrypi3-64`.

I am currently trying to build core-image-sato for Raspberry Pi 3 Model B, but linux-kernel 4.19 cannot be fetched. I get error `Unable to find revision ab8652c03fa081b27de7e28a74c2536cb2aa3e5b in branch rpi-4.19.y even from upstream`
**Steps to reproduce the issue:**
1.Edit the line `PREFERRED_VERSION_linux-raspberrypi` in `rpi-default-versions.inc` to have version string "4.19%".
2. Set `MACHINE` to `raspberrypi3-64` in `local.conf`.
3. Execute `bitbake core-image-sato` or `bitbake linux-raspberrypi`

**Additional details (revisions used, host distro, etc.):**
Using Ubuntu 18.04, Layer branch warrior.

agherzan, 2019-06-12:
Use master as a workaround. I'll back-port the changes in warrior too.

gaurav-xebrium, 2019-06-12:
alright, will use master, Thanks!

agherzan, 2019-06-14:
@gaurav-xebrium `warrior` branch has now the fix for this.

closed 2019-06-14 by agherzan