agherzan/meta-raspberrypi #428: OMXPlayer build fails

TomMettam, 2019-06-16:
OMXPlayer build fails when targeting Raspberry Pi 3b+ from meta-raspberrypi master

`/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r4/recipe-sysroot/usr/include/bits/byteswap.h:34:29: sorry, unimplemented: Thumb-1 hard-float VFP ABI`

c0ffee, 2019-06-16:
```
diff --git recipes-multimedia/omxplayer/omxplayer_git.bb recipes-multimedia/omxplayer/omxplayer_git.bb
index aad2fed..1d539fc 100644
--- recipes-multimedia/omxplayer/omxplayer_git.bb
+++ recipes-multimedia/omxplayer/omxplayer_git.bb
@@ -80,6 +80,8 @@ export INCLUDES = "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", " -D_
                   "
 export DIST = "${D}"
 
+ARM_INSTRUCTION_SET = "arm"
+
 do_compile() {
     # Needed for compiler test in ffmpeg's configure
     mkdir -p tmp
```

fixed it for me

c0ffee, 2019-06-16:
i didnt create a pull request because i was not sure if it is a good idea to disable thumb for all people for this package ;) but sadly so far i didnt find a better solution

TomMettam, 2019-06-16:
Will "arm" work for all pi variants?

c0ffee, 2019-06-16:
all pi variants are arm yes and support the basic arm instruction set. thumb instruction set reduces the sizes and could increase the performance, I'm using omxplayer with it and couldnt discover any problems with it yet.

yocto is using by default here thumb2 instruction set for this machine, I have no clue why it complains about thumb1 in this case, maybe we can get it to work with thumb2 somehow...

kraj, 2019-06-17:
> OMXPlayer build fails when targeting Raspberry Pi 3b+ from meta-raspberrypi master
> 
> `/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r4/recipe-sysroot/usr/include/bits/byteswap.h:34:29: sorry, unimplemented: Thumb-1 hard-float VFP ABI`

the fix to disable thumb ISA may be a broad-brush here, Please check if omxplayer build it modifying -mcpu/march options to use something that is < armv6 which would mean -mthumb would point to thumb1 and that could result in such errors. What is the full compiler cmdline for this particular build failures?

agherzan, 2019-06-19:
I agree. We should find the root cause of this.

closed 2019-06-20 by agherzan

c0ffee, 2019-06-21:
can we pls cherry-pick this in warrior?

agherzan, 2019-06-21:
@c0ffee Could you please send a PR with this commit targeting the warrior branch?

PR #430 (2019-06-20): omxplayer: remove hardcoded tune and arch from Makefile
Fixes: #428

**- What I did**
Removed the hardcoded arch and tune settings from Makefile 
**- How I did it**
Patching the Makefile
commit ca11a291 (2019-06-20): omxplayer: remove hardcoded tune and arch from Makefile

Fixes: #428
PR #431 (2019-06-24): omxplayer: remove hardcoded tune and arch from Makefile
Fixes: #428

**- What I did**
Backported the fix for compiling recent omxplayer with thumb support
**- How I did it**
commit 6c736949 (2019-06-24): omxplayer: remove hardcoded tune and arch from Makefile

Fixes: #428