agherzan/meta-raspberrypi #432: RaspberryPi 4

Romaxx, 2019-06-24:
Hi, RaspberryPi 4 includes new concepts (SPI boot eprom, PCIe, Integrated Ethernet, dual HDMI), any suggestions/roadmap to begin the integration of RPI4 with this layer ? Thk

agherzan, 2019-06-24:
Waiting for hardware.

Romaxx, 2019-06-24:
some people have received the rpi4 some weeks ago under NDA, nobody in this situation here ?

agherzan, 2019-06-24:
I'm not part of that elite :)

agherzan, 2019-06-27:
And now I am: I have the board and will probably push an initial support today.

diegosueiro, 2019-07-03:
@agherzan ,

Just to give you a feedback that I have a Rpi4-1G booting with ag/rpi4-64 branch.

Do you have any idea of the effort to get u-boot working on Rpi4?

agherzan, 2019-07-03:
@diegosueiro I plan to look into uboot support as soon as I sort out the kerne. I still fail to make the board mount the SD card root. What is your `local.conf` configuration?

diegosueiro, 2019-07-03:
@agherzan ,

I think the important bits in my local.conf are (nothing special I suppose):

```
MACHINE = "raspberrypi4-64"
ENABLE_UART = "1"
BOOT_SPACE = "80192"
CMDLINE_DEBUG = "earlyprintk loglevel=8"
PACKAGE_CLASSES = "package_ipk"
DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_initscripts = ""
IMAGE_INSTALL_append = " kernel-modules"
```

And I'm building the core-image-minimal.

This is the output for `dmesg | grep -i mmc`:
```
[    0.000000] Kernel command line: 8250.nr_uarts=1 cma=64M bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0
 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:04:10:4D vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x4
0000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait    ear
lyprintk loglevel=8
[    0.762989] mmc-bcm2835 fe300000.mmcnr: could not get clk, deferring probe
[    2.140844] mmc-bcm2835 fe300000.mmcnr: mmc_debug:0 mmc_debug2:0
[    2.146990] mmc-bcm2835 fe300000.mmcnr: DMA channel allocated
[    2.178809] sdhci-iproc fe340000.emmc2: Linked as a consumer to regulator.1
[    2.200882] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    2.208050] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.215204] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.220807] mmc0: SDHCI controller on fe340000.emmc2 [fe340000.emmc2] using ADMA
[    2.231023] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    2.239623] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.246994] Waiting for root device /dev/mmcblk0p2...
[    2.333896] mmc1: new high speed SDIO card at address 0001
[    2.354564] mmc0: new ultra high speed DDR50 SDXC card at address 1234
[    2.362507] mmcblk0: mmc0:1234 SA64G 57.7 GiB
[    2.368976]  mmcblk0: p1 p2
[    2.407468] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    4.469399] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    7.966555] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.bin failed with 
error -2
```

agherzan, 2019-07-03:
@diegosueiro I found the issue but I'm really curious how come it works for you. Do you happen to have any additional kernel config fragment in your build setup?

diegosueiro, 2019-07-03:
@agherzan,

No I don't have any changes. The only difference is that you are using the 4G and I'm using the 1GB. Right?

agherzan, 2019-07-03:
The key bit is here:

```
[    2.178809] sdhci-iproc fe340000.emmc2: Linked as a consumer to regulator.1
```

My build, and also the upstream defconfig doesn't have the driver for SDHCI iproc. So in my case that never gets probed because obviously there is nothing to. Can you grep for this config (CONFIG_MMC_SDHCI_IPROC) in the defconfig that ends up in WORKDIR and also the one that ends up in the build directory as `.config`? Somehow you have that driver enabled.

I pushed a fix to rpi-linux as to get this fix upstream: https://github.com/raspberrypi/linux/pull/3045 .

diegosueiro, 2019-07-03:
@agherzan,

The `defconfig` in the WORKDIR and the `.config` both have the `CONFIG_MMC_SDHCI_IPROC=y`.

In the previous tests, I had added by hand a defconfig that was the same as the raspbian. But when I updated my local copy of the meta-raspberrypi with your ag/rpi4-64 branch I removed all my tweaks. I think yocto didn't replace my copied defconfig in the WORKDIR folder.

agherzan, 2019-07-03:
@diegosueiro  Well that somehow messed up your build. But at least we have an explanation now. I'm currently making sure all machines still build and I'll merge this support.

diegosueiro, 2019-07-05:
@agherzan,

In order to have the USB-C OTG working (with dtoverlay=dwc2) I had to add the following in the `recipes-kernel/linux/linux-raspberrypi.inc` file:
```
do_compile_append_raspberrypi4-64() {
    cc_extra=$(get_cc_option)
    oe_runmake dtbs CC="${KERNEL_CC} $cc_extra " LD="${KERNEL_LD}" ${KERNEL_EXTRA_ARGS}
}
```

agherzan, 2019-07-11:
@diegosueiro can you explain a little what is that doing? I miss context.

diegosueiro, 2019-07-11:
@agherzan,

I don't know either. :-(
I just replicated the same lines of the raspberrypi3-64.
https://github.com/agherzan/meta-raspberrypi/commit/6c4de0b5fe44b8e661f1391ee8540a7f04d75315

pauldotknopf, 2019-07-12:
Hey guys, just jumping in here. Would you consider the latest bits usable for Raspberry Pi 4? What all is working/not working?

agherzan, 2019-07-12:
@pauldotknopf Fell free to try it. The current known limitation is 1G mem on 64bit. Everything else should work as much as upstream (kernel, uboot etc) works.

agherzan, 2019-07-12:
Uboot support: https://github.com/agherzan/meta-raspberrypi/pull/437

Romaxx, 2019-07-13:
Other limitation of 64bit Os on previous versions of PI was the use of DSI (Display Serial Interface) and CSI (Camera Serial Interface) lanes due to the 32-bit interface to Videocore IV GPU, Raspberrypi 4 use VideoCore VI, 32bits also, but with a new MMU, have someone tested DSI and CSI port with 64bits Build on Raspberrypi 4 ? Thk

agherzan, 2019-07-25:
@Romaxx I haven't tested many peripherals. If you find any that you test and don't work as expected please create a new issue. Thanks.

closed 2019-07-25 by agherzan