agherzan/meta-raspberrypi #444: ERROR: No recipes available for:  u-boot and gstreamer?

grbesd1, 2019-07-18:
ERROR: No recipes available for:
 Projects/raspberrypi/rpi_yocto/poky/meta-raspberrypi/recipes-bsp/u-boot/u-boot_2019.07.bbappend
 Projects/raspberrypi/rpi_yocto/poky/meta-raspberrypi/recipes-multimedia/gstreamer/gstreamer1.0-omx_1.16%.bbappend

I'm using meta-raspberrypi in yocto setup.  While running bitbake core-image-base i'm getting missing recipe error.

**Steps to reproduce the issue:**
mkdir rpi_yocto
cd rpi_yocto
git clone git://git.yoctoproject.org/poky
cd poky
git clone git://git.openembedded.org/meta-openembedded
git clone git://git.yoctoproject.org/meta-raspberrypi
source oe-init-build-env rpi64-build
echo 'MACHINE = "raspberrypi3-64"' >> conf/local.conf
echo 'PREFERRED_VERSION_linux-raspberrypi = "4.19"' >> conf/local.conf

Added following lines:
BBLAYERS ?= " \
  /Projects/raspberrypi/rpi_yocto/poky/meta \
  /Projects/raspberrypi/rpi_yocto/poky/meta-poky \
  /Projects/raspberrypi/rpi_yocto/poky/meta-yocto-bsp \
  /Projects/raspberrypi/rpi_yocto/poky/meta-raspberrypi \
  /Projects/raspberrypi/rpi_yocto/poky/meta-openembedded/meta-oe \
  /Projects/raspberrypi/rpi_yocto/poky/meta-openembedded/meta-multimedia \
  /Projects/raspberrypi/rpi_yocto/poky/meta-openembedded/meta-networking \
  /Projects/raspberrypi/rpi_yocto/poky/meta-openembedded/meta-python \

bitbake core-image-base

HOST: Ubuntu 18 LTS
kernel: 4.15

agherzan, 2019-07-18:
I believe you don't have an up to date poky and you are using this layer's master HEAD.
As you can see, here http://git.yoctoproject.org/cgit.cgi/poky/tree/meta/recipes-bsp/u-boot/u-boot_2019.07.bb poky has the needed recipe.

agherzan, 2019-07-18:
I will close this issue for now. Feel free to comment if you encounter any more issues.

closed 2019-07-18 by agherzan