agherzan/meta-raspberrypi #446: Raspeberrypi4-64, usb no power

iohe, 2019-07-19:
**Description**
Keyboard is not powered, kernel reports errors 
**Steps to reproduce the issue:**
1. build meta-raspberry @ 6531d1d (HEAD -> ag/rpi4-uboot-update, origin/ag/rpi4-uboot-update) u-boot: Update patches for RPi4
2.
3.

**Describe the results you received:**
Errors encountered in kernel :
[    2.530074] usb 1-1: new high-speed USB device number 4 using xhci_hcd
[    2.550489] usb 1-1: device descriptor read/8, error -61
[    2.678539] usb 1-1: device descriptor read/8, error -61
[    2.914059] usb 1-1: new high-speed USB device number 5 using xhci_hcd
[    2.934469] usb 1-1: device descriptor read/8, error -61
[    3.062515] usb 1-1: device descriptor read/8, error -61
[    3.170269] usb usb1-port1: unable to enumerate USB device

**Describe the results you expected:**
Usb keyboard to work.

**Additional information you deem important (e.g. issue happens only occasionally):**
Always happens.
It did not happen with meta-raspberry@ 1e329d36c99718194d20e6216f1c13199e124bb7

**Additional details (revisions used, host distro, etc.):**
layers used:
-poky@97c5921666
-meta-openembedded@27d1d3dfa
- meta-raspberry @ 6531d1d
l-ocal.conf has:
MACHINE ?= "raspberrypi4-64"
DISTRO_FEATURES_append = " opengl"
DISTRO_FEATURES_append = " polkit"
-Image : core-image-x11
Other packages in image: xfce4

[output.yocto.txt](https://github.com/agherzan/meta-raspberrypi/files/3411975/output.yocto.txt)
[dmesg.txt](https://github.com/agherzan/meta-raspberrypi/files/3411976/dmesg.txt)

agherzan, 2019-07-19:
What power supply are you using? Are you testing it with an official charger?

iohe, 2019-07-19:
Official power source for rpi4.
usb is powered and no issue present if i build a couple of days old commit, as specified.  I suspect something wrong on firmware side.

agherzan, 2019-07-19:
That is worrying. Would you be able to give it a try with an older firmware and test that theory?

iohe, 2019-07-19:
[dmesg.txt](https://github.com/agherzan/meta-raspberrypi/files/3412283/dmesg.txt)
This works ok

iohe, 2019-07-21:
Using https://github.com/agherzan/meta-raspberrypi/commit/8d5fc54f41dab42d765644ea953bf451ced78f74 i can attach my Razer Blackwidw Lite keyboard and type. 
Using 69da331154852f5d62d6cf7a69c4ddb34f5564d0 it does not go more than colored screen at boot.
Using 0d641c8fa918b1c5ecce2cb9918c47bb3c4e1510, and RPI_USE_U_BOOT = "1" it does not appear again.
Using a USBtester that shows Voltage and amps, it never shows amps above 0.
Is anyone else able to reproduce issue ?

agherzan, 2019-07-22:
The relevant kernel issue is: https://github.com/raspberrypi/linux/issues/3093

agherzan, 2019-07-24:
@iohe if you are willing to compromise on 1Gb of ram, here is a PR ^

iohe, 2019-07-24:
ok for now, will revisit later for full 4g. 
Thank you very much!

PR #450 (2019-07-25): raspberrypi4-64.conf: Limit RAM to 3G
The PCI controller on the Raspberry Pi 4 acts as a DMA controller and can
only address the first 3Gb[1] of the memory space. On 32bit this was
addressed by implementing bouncing buffers[2] but this currently has no
support for arm64. In order to have working USB, as a workaround, we
limit the ram to 3G.

Fixes #446

[1] https://github.com/raspberrypi/linux/issues/3093
[2] https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/pci/controller/pcie-brcmstb-bounce.c
commit fafad9b0 (2019-07-25): raspberrypi4-64.conf: Limit RAM to 3G

The PCI controller on the Raspberry Pi 4 acts as a DMA controller and can
only address the first 3Gb[1] of the memory space. On 32bit this was
addressed by implementing bouncing buffers[2] but this currently has no
support for arm64. In order to have working USB, as a workaround, we
limit the ram to 3G.

Fixes #446

[1] https://github.com/raspberrypi/linux/issues/3093
[2] https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/pci/controller/pcie-brcmstb-bounce.c
PR #451 (2019-07-26): [warrior backport] raspberrypi4-64.conf: Limit RAM to 3G
The PCI controller on the Raspberry Pi 4 acts as a DMA controller and can
only address the first 3Gb[1] of the memory space. On 32bit this was
addressed by implementing bouncing buffers[2] but this currently has no
support for arm64. In order to have working USB, as a workaround, we
limit the ram to 3G.

Fixes #446

[1] https://github.com/raspberrypi/linux/issues/3093
[2] https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/pci/controller/pcie-brcmstb-bounce.c
commit e9e3ad19 (2019-07-26): raspberrypi4-64.conf: Limit RAM to 3G

The PCI controller on the Raspberry Pi 4 acts as a DMA controller and can
only address the first 3Gb[1] of the memory space. On 32bit this was
addressed by implementing bouncing buffers[2] but this currently has no
support for arm64. In order to have working USB, as a workaround, we
limit the ram to 3G.

Fixes #446

[1] https://github.com/raspberrypi/linux/issues/3093
[2] https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/pci/controller/pcie-brcmstb-bounce.c
closed 2019-07-24 by iohe