agherzan/meta-raspberrypi #453: No HDMI output on Raspberry Pi 4

mozcelikors, 2019-07-25:
Hi, I recently got a Pi 4, 4GB version. Previously I had Raspberry Pi 3B+, with perfectly running everything I require, using Yocto. I have a Qt app which runs using eglfs platform plugin running on Pi 3B+, but not working on Pi 4. I wanted to port my work there to the meta-raspberrypi's raspberrypi4 machine. Here's what I did:

**bblayers.conf**
```
BBLAYERS ?= " \
  ${BSPDIR}/meta \
  ${BSPDIR}/meta-poky \
  ${BSPDIR}/meta-yocto-bsp \
  ${BSPDIR}/meta-openembedded/meta-oe \
  ${BSPDIR}/meta-openembedded/meta-multimedia \
  ${BSPDIR}/meta-openembedded/meta-networking \
  ${BSPDIR}/meta-openembedded/meta-python \
  ${BSPDIR}/meta-openembedded/meta-gnome \
  ${BSPDIR}/meta-openembedded/meta-xfce \
  ${BSPDIR}/meta-qt5 \
  ${BSPDIR}/meta-qt5-extra \
  ${BSPDIR}/meta-raspberrypi \
```

**local.conf**
```
MACHINE_FEATURES_append = "vc4graphics"
DISTRO_FEATURES_append = " bluez5 bluetooth wifi polkit "
DISTRO_FEATURES_remove = "systemd"
IMAGE_FEATURES += "splash"
SPLASH = "psplash-raspberrypi"
LICENSE_FLAGS_WHITELIST_append = " commercial "
RPI_KERNEL_DEVICETREE_OVERLAYS_append = " overlays/ads7846.dtbo "
```

**rpi-config_%.bbappend**
```
echo "hdmi_group=2" >> ${DEPLOYDIR}/bcm2835-bootfiles/config.txt
echo "hdmi_mode=87" >> ${DEPLOYDIR}/bcm2835-bootfiles/config.txt
echo "hdmi_cvt 800 480 60 6 0 0 0" >> ${DEPLOYDIR}/bcm2835-bootfiles/config.txt
echo "hdmi_drive=1" >> ${DEPLOYDIR}/bcm2835-bootfiles/config.txt
```
(Whether I have these changes or not, does not affect the result)

**myimage.bb**
```
include recipes-core/images/core-image-minimal.bb
IMAGE_INSTALL_append = " packagegroup-core-x11 packagegroup-core-boot packagegroup-xfce-base "
IMAGE_INSTALL_append = "  kernel-modules "
IMAGE_INSTALL_append = " linux-firmware-rpidistro-bcm43455 bluez-firmware-rpidistro-bcm4345c0-hcd"
... and more
```

When I try to either **run my app on OpenGL with eglfs**, or **when I simply run Xorg (XFCE), I get no display output on both HDMI's**. Only hdmi output I get is the output from psplash splash screen, which looks like its bars are not working.

Just for example, I get following on app output:
```
QT_QPA_EGLFS_NO_LIBINPUT=1 QT_LOGGING_RULES="qt.qpa.*=true" /opt/myapp -platform eglfs
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/var/volatile/tmp/runtime-root'
qt.qpa.egldeviceintegration: EGL device integration plugin keys: ("eglfs_emu", "eglfs_kms_egldevice", "eglfs_kms", "eglfs_x11")
qt.qpa.egldeviceintegration: EGL device integration plugin keys (sorted): ("eglfs_kms", "eglfs_emu", "eglfs_kms_egldevice", "eglfs_x11")
qt.qpa.egldeviceintegration: Trying to load device EGL integration "eglfs_kms"
qt.qpa.eglfs.kms: New DRM/KMS via GBM integration created
qt.qpa.egldeviceintegration: Using EGL device integration "eglfs_kms"
qt.qpa.eglfs.kms: platformInit: Opening DRM device
qt.qpa.input: udev device discovery for type QFlags<QDeviceDiscovery::QDeviceType>(Device_DRM|Device_VideoMask)
qt.qpa.input: Found matching devices ("/dev/dri/card1", "/dev/dri/card0")
qt.qpa.eglfs.kms: Found the following video devices: ("/dev/dri/card1", "/dev/dri/card0")
qt.qpa.eglfs.kms: Using "/dev/dri/card1"
qt.qpa.eglfs.kms: Using backend-provided DRM device /dev/dri/card1
qt.qpa.eglfs.kms: Creating GBM device for file descriptor 6 obtained from "/dev/dri/card1"
qt.qpa.eglfs.kms: Querying EGLDisplay
qt.qpa.eglfs.kms: Found 3 planes
qt.qpa.eglfs.kms: plane 0: id = 28 countFormats = 10 possibleCrtcs = 0x1 supported formats = XR24 AR24 RG16 RG24 BG24 YU16 YU12 YV12 NV12 NV21 
qt.qpa.eglfs.kms:   property 0: id = 7 name = 'type'
[... 282 lines cut ...]
qt.qpa.eglfs.kms:   property 1: id = 21 name = 'MODE_ID'
qt.qpa.eglfs.kms:   type is BLOB
qt.qpa.eglfs.kms:   property 2: id = 18 name = 'OUT_FENCE_PTR'
qt.qpa.eglfs.kms:   type is RANGE, value is 0, possible values are:
qt.qpa.eglfs.kms:     0
qt.qpa.eglfs.kms:     18446744073709551615
qt.qpa.eglfs.kms: Output HDMI1 can use 3 planes: 28 35 42 
qt.qpa.eglfs.kms: Output eglfs plane is: 28
qt.qpa.eglfs.kms: EDID data for output "HDMI1": identifier '', manufacturer 'Samsung Electric Company', model 'SyncMaster-', serial '', physical size: 160.00x90.00
qt.qpa.eglfs.kms: Creating new global GBM mouse cursor
qt.qpa.eglfs.kms: Initializing cursor atlas from ":/cursor.json"
qt.qpa.eglfs.kms: Sorted screen list: QVector(OrderedScreen(QPlatformScreen=0xcee930 ("HDMI1") : 2147483647 / QPoint(0,0) / primary: false))
qt.qpa.eglfs.kms: Adding QPlatformScreen 0xcee930 ( "HDMI1" ) to QPA with geometry QRect(0,0 1920x1080) and isPrimary= false
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
qt.qpa.eglfs.kms: Creating gbm_surface for screen HDMI1
qt.qpa.eglfs.kms: Got native format 34325258 from eglGetConfigAttrib() with return code true
Draw call returned Invalid argument.  Expect corruption.
qt.qpa.eglfs.kms: Adding FB, size 1920x1080, DRM format 0x34325258
qt.qpa.eglfs.kms: Setting mode for screen HDMI1
qt.qpa.eglfs.kms: Adding FB, size 1920x1080, DRM format 0x34325258

```
I use warrior branch for all standard layers and meta-raspberrypi layer. Does meta-raspberrypi has HDMI support? Is there something I am missing here? Could we please get a little help for the HDMI output?
Thank you very much in advance

xuhcc, 2019-07-27:
Hi @mozcelikors 
Did you try to run RPi 3 with the warrior branch? Yesterday I built an image for my RPi 3 with the latest commit from the warrior and encountered similar issues with HDMI display. Now I'm building it from https://github.com/agherzan/meta-raspberrypi/commit/b76252c98520b79e012495f8022f238f66c20b05 and it's working. So may be this problem is not specific to RPi 4.

mozcelikors, 2019-07-27:
Hi @xuhcc, 
I unfortunately can't get any HDMI output still with commit [b76252c ](https://github.com/agherzan/meta-raspberrypi/commit/b76252c98520b79e012495f8022f238f66c20b05) using my Raspberry Pi 4.

agherzan, 2019-07-27:
@mazcelikors can you please try the current master's or warrior's HEAD?

mozcelikors, 2019-07-27:
Hi  @agherzan,
Thanks for your awesome work with meta-raspberrypi, I have to say.
Previously I tried with warrior's HEAD and unfortunately was unable to get HDMI output. I checked and there is no new commit at warrior's HEAD so I doubt it will change things.
I get the following for the master's HEAD:

ERROR: No recipes available for:
```
.../meta-raspberrypi/recipes-bsp/u-boot/u-boot_2019.07.bbappend
.../meta-raspberrypi/recipes-multimedia/gstreamer/gstreamer1.0-omx_1.16%.bbappend
```

Maybe /dev/kmsg output also helps here:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.58 (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 SMP Wed Jul 24 17:27:49 UTC 2019
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000000fc00000
[    0.000000] On node 0 totalpages: 901120
[    0.000000]   DMA zone: 1152 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 131072 pages, LIFO batch:31
[    0.000000]   HighMem zone: 770048 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e0 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39628 r8192 d21812 u69632
[    0.000000] pcpu-alloc: s39628 r8192 d21812 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 899968
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M  smsc95xx.macaddr=DC:A6:32:03:2E:E3 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0  root=/dev/mmcblk0p2 rootfstype=ext4 rootwait logo.nologo splash rootflags=commit=120,data=writeback bootdelay=0 silent=1 elevator=deadline noatime nodiratime data=writeback quiet consoleblank=0  console=tty3 loglevel=0 vt.global_cursor_default=0
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 3293440K/3604480K available (8192K kernel code, 661K rwdata, 2492K rodata, 2048K init, 852K bss, 48896K reserved, 262144K cma-reserved, 3080192K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                   lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 662 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 853 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 29088 entries in 86 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000017] Switching to timer-based delay loop, resolution 18ns
[    0.000234] Console: colour dummy device 80x30
[    0.000257] console [tty3] enabled
[    0.000297] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000319] pid_max: default: 32768 minimum: 301
[    0.000581] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000597] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001326] CPU: Testing write buffer coherency: ok
[    0.001720] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002340] Setting up static identity map for 0x200000 - 0x20003c
[    0.002494] rcu: Hierarchical SRCU implementation.
[    0.003343] smp: Bringing up secondary CPUs ...
[    0.004187] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005157] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006079] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[... 243 lines cut ...]
[    2.144360] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[    2.148947] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    2.148967] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[    2.153210] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[    2.153224] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[    2.227567] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    2.412454] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    2.856748] urandom_read: 3 callbacks suppressed
[    2.856755] random: dd: uninitialized urandom read (512 bytes read)
[... cut ...]
[    3.695895] brcmfmac: power management disabled
[    3.696277] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    4.232027] bcmgenet fd580000.genet eth0: Link is Down
[    5.346774] Bluetooth: Core ver 2.22
[    5.346857] NET: Registered protocol family 31
[    5.346867] Bluetooth: HCI device and connection manager initialized
[    5.347155] Bluetooth: HCI socket layer initialized
[    5.347172] Bluetooth: L2CAP socket layer initialized
[    5.347214] Bluetooth: SCO socket layer initialized
```

mozcelikors, 2019-07-28:
@agherzan I have switched everything to master branch and now able to get HDMI output. However, graphics are incredibly slow on Qt/eglfs (0.5 FPS). On Raspberry Pi 3, I used to get 30 FPS with the same set up. I read that it might be due to hardware/software acceleration difference.

kraj, 2019-07-30:
@mozcelikors try changing  
`
VC4DTBO ?= "vc4-fkms-v3d"
`
to
`
VC4DTBO ?= "vc4-kms-v3d"
`
in
`conf/machine/raspberrypi4-64.conf` or `conf/machine/raspberrypi4.conf`

mozcelikors, 2019-07-30:
@kraj Thanks for the response. Is vc4-kms-v3d recently enabled? I tried it before and I remember not getting any helpful result. In fact, I got no HDMI signal at all using that.

kraj, 2019-07-30:
It regresssed for one kernel update we did for 4.19 ( at least for rpi3) but then it was fixed. I don't yet have rpi4 so dont know if it is fixed there

mozcelikors, 2019-07-30:
@kraj @agherzan Hi again, I built my image using everything master, and still no improvement.
I doubt vc4-kms-v3d works with Raspberry Pi 4. I get the following on /dev/kmsg:
```
[    2.631929] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[    3.671876] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[    4.711880] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[    5.751878] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[    6.791880] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[   17.111887] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:116:crtc-2] flip_done timed out
[   27.351889] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:116:crtc-2] flip_done timed out
[   37.591881] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CONNECTOR:44:Composite-1] flip_done timed out
[   47.831878] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:113:plane-20] flip_done timed out
[   58.071882] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:116:crtc-2] flip_done timed out
[   58.075229] Console: switching to colour frame buffer device 90x30
[   63.701916] udevd[118]: worker [130] /devices/platform/soc/soc:gpu is taking a long time
[   68.311878] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:116:crtc-2] flip_done timed out
[   78.551882] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CONNECTOR:44:Composite-1] flip_done timed out
[   88.791877] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:113:plane-20] flip_done timed out
[   99.031876] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:116:crtc-2] flip_done timed out
[  101.191924] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  101.384956] vc4_hdmi fe902000.hdmi: ASoC: can't open interface fe902000.hdmi: -19
[  101.385701] vc4_hdmi fe902000.hdmi: ASoC: can't open interface fe902000.hdmi: -19
[  101.387023] vc4_hdmi fe902000.hdmi: ASoC: can't open interface fe902000.hdmi: -19
[  101.390424] vc4_hdmi fe902000.hdmi: ASoC: can't open interface fe902000.hdmi: -19
[  101.390823] vc4_hdmi fe902000.hdmi: ASoC: can't open interface fe902000.hdmi: -19
[  105.351899] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  106.391877] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  107.432017] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  108.471875] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  109.511882] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  110.551892] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  120.791914] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:116:crtc-2] flip_done timed out
[  131.041893] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CONNECTOR:44:Composite-1] flip_done timed out
[  141.281930] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:113:plane-20] flip_done timed out
[  151.511893] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:116:crtc-2] flip_done timed out
[  152.631916] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  153.671885] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  154.711891] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  155.751891] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  156.791890] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  162.791888] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  163.831887] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  164.871890] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  165.911890] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  166.951889] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  167.511904] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:116:crtc-2] flip_done timed out
[  177.751903] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:113:plane-20] flip_done timed out
[  178.161894] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  179.191914] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  180.231915] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  181.271883] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  182.311883] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  187.991900] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:116:crtc-2] flip_done timed out
[  193.511912] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  194.551941] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  195.591908] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  196.631892] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  197.671910] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  208.871907] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  209.911913] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  210.951910] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  211.991912] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  213.031911] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  224.231913] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  225.271912] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  226.311916] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  227.351913] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  228.391911] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  239.601893] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  240.631925] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[... 7 lines cut ...]
[  259.111925] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[  270.311911] i2c-bcm2835 fe805000.i2c: i2c transfer timed out

```
[... cut ...]  it used to run with great performance then.

What I tried:

Raspberry Pi 3 B and Raspberry Pi 3 B +
meta-raspberrypi sumo
vc4-kms-v3d
gpu_mem=128
Excellent performance on Qt/OpenGLES/EGLFS

Raspberry Pi 4
meta-raspberrypi master
vc4-fkms-v3d
Very very poor performance on Qt/OpenGLES/EGLFS

Raspberry Pi 4
meta-raspberrypi master
vc4-kms-v3d
No HDMI output (+ Aforementioned DRM/HDMI/I2C errors)

I guess I will have to wait until vc4-kms-v3d is ported. Anything I can help with, please let me know.

agherzan, 2019-07-31:
@mozcelikors are you working on 32 or 64 bit machine?

mozcelikors, 2019-07-31:
@agherzan I use x86_64 host and raspberry pi 4 32 bit (raspberrypi4.conf)

agherzan, 2019-07-31:
@mozcelikors That is strange because we basically have the same setup as Raspbian. Can you try to reproduce on raspbian?

mozcelikors, 2019-07-31:
@agherzan I dont have Buster and Qt Raspbian SDK installed at the moment, but I can try to reproduce it in a few days.

peel70, 2019-08-04:
hello @agherzan, I've the same problem as @mozcelikors; with warrior's branch no hdmi output from qt, so I'll try the master's branch and now I have output from qt but the performances are poor. I've changed 
VC4DTBO ?= "vc4-fkms-v3d"
to
VC4DTBO ?= "vc4-kms-v3d"
in
conf/machine/raspberrypi4-64.conf or conf/machine/raspberrypi4.conf

as suggested by @kraj but the result in the same.

In the config.txt of the rpi there is this line:
# Enable VC4 Graphics
dtoverlay=vc4-fkms-v3d

so I think VC4DTBO ?= "vc4-fkms-v3d" is correct.

maybe is too soon for GPU acceleration in QT via eglfs_kms?

thank you for your kindly attention

mozcelikors, 2019-08-04:
Hi @peel70 ,
Please do share with us if you stumble upon a solution. I am also on the lookout.
Just a note: If you use -platform linuxfb --no-sand-box, you will get a lot better performance than eglfs at the moment. Still very very disappointing though. This issue basically renders Raspberry Pi 4 useless for people who would like to use it for Qt-based head units.

peel70, 2019-08-05:
Hi @mozcelikors,
I'm investigating gstreamer pipeline with gst-launch-1.0 to check if the problem is in the QT.
I've tried to compile QT with kms flag and create a json for eglfs_kms configuration without success.

agherzan, 2019-08-05:
@kraj @mozcelikors https://github.com/raspberrypi/linux/issues/3020 So fkms is the only option for rpi4. We have to figure out why is it so slow. And the first step would be to see how it behaves on raspbian.

mozcelikors, 2019-08-05:
@peel70 @kraj @agherzan I will try it on raspbian but I wont be able to do it most possibly until next week due to my work load at the moment. If any one could do it earlier, please try to do so and share results with us. Thanks

peel70, 2019-08-06:
Hi @mozcelikors , @agherzan , @kraj 
Today I've cross compiled QT 5.12.4 on my desktop for RPI4 with -device linux-rasp-pi3-vc4-g++ with this configuration:

Building on: linux-g++ (x86_64, CPU features: mmx sse sse2)
Building for: devices/linux-rasp-pi3-vc4-g++ (arm, CPU features: neon)
Target compiler: gcc 4.8.3
Configuration: cross_compile compile_examples enable_new_dtags largefile neon precompile_header shared rpath release c++11 concurrent dbus reduce_exports stl
Build options:
  Mode ................................... release
  Optimize release build for size ........ no
  Building shared libraries .............. yes
  Using C standard ....................... C11
  Using C++ standard ..................... C++11
  Using ccache ........................... no
  Using gold linker ...................... no
  Using new DTAGS ........................ yes
  Using precompiled headers .............. yes
  Using LTCG ............................. no
  Target compiler supports:
    NEON ................................. yes
  Build parts ............................ libs
Qt modules and options:
  Qt Concurrent .......................... yes
  Qt D-Bus ............................... yes
  Qt D-Bus directly linked to libdbus .... yes
  Qt Gui ................................. yes
  Qt Network ............................. yes
  Qt Sql ................................. yes
  Qt Testlib ............................. yes
  Qt Widgets ............................. yes
  Qt Xml ................................. yes
Support enabled for:
  Using pkg-config ....................... yes
  udev ................................... yes
  Using system zlib ...................... yes
Qt Core:
  DoubleConversion ....................... yes
    Using system DoubleConversion ........ yes
  GLib ................................... yes
  iconv .................................. yes
  ICU .................................... no
  Tracing backend ........................ <none>
  Logging backends:
    journald ............................. no
    syslog ............................... no
    slog2 ................................ no
  Using system PCRE2 ..................... yes
Qt Network:
  getifaddrs() ........................... yes
  IPv6 ifname ............................ yes
  libproxy ............................... no
  Linux AF_NETLINK ....................... yes
  OpenSSL ................................ yes
    Qt directly linked to OpenSSL ........ no
  OpenSSL 1.1 ............................ yes
  DTLS ................................... yes
  SCTP ................................... no
  Use system proxies ..................... yes
Qt Gui:
  Accessibility .......................... yes
  FreeType ............................... yes
    Using system FreeType ................ yes
  HarfBuzz ............................... yes
    Using system HarfBuzz ................ yes
  Fontconfig ............................. yes
  Image formats:
    GIF .................................. yes
    ICO .................................. yes
    JPEG ................................. yes
      Using system libjpeg ............... yes
[... 74 lines cut ...]
1) It is not possible to launch the app from QtCreator with these errors
QML debugging is enabled. Only use this in a safe environment.
Setting framebuffer size is only available with DRM atomic API
Failed to move cursor on screen HDMI1: -13
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
Could not set cursor on screen HDMI1: -13
failed to export dumb buffer: Permission denied
Failed to create scanout resource
failed to export dumb buffer: Permission denied
Failed to create scanout resource
QEGLPlatformContext: eglSwapBuffers failed: 3003
Could not lock GBM surface front buffer!

2) The Application wont work without X11 either in EGLFS, LINUXFB or other.

3) It's working only launch from the RPI with desktop.

I've addedd pi user to the render group but nothing change

I've tried gstreamer (without X11) with gst-launch with kmssink and the CPU is under the 8% in a full hd video but the graphics performances are poor.

So I don't know where the problem is.

regards to all

mozcelikors, 2019-08-07:
@peel70 @agherzan  @kraj 
I wonder why you have:
EGLFS Raspberry Pi ................... no
With your device selection, I would assume eglfs_brcm would be available.

If you did this on Raspbian Buster (which I assume you did) and you say that it's only working with X11; then I think our presumptions are correct, and this is indeed an issue with Raspberry Pi 4 at the moment.

Could you please create a question to Raspberry Pi Forums regarding this question and link it here?
Kind Regards.

peel70, 2019-08-07:
Hello @mozcelikors , @agherzan , @kraj 
finally I'm near to the solution (I think).

I've just rebuild the cross compile with Raspian Burster Lite (no X11) and follow an issue on the Qt bug reports https://bugreports.qt.io/browse/QTBUG-76889?filter=-3

It's normal that with the  linux-rasp-pi3-vc4-g+'s spec is not enabled eglfs_bcrm because in this case (like RPI3-64bit) is using kms (Using EGL device integration "eglfs_kms" say activating the filter rules in the QLoggin Category)

Now with the cross compile I've QT5.12.4 accelerated, the user must be in the "render" group for activating the hardware acceleration as sayed in the Qt bug report, and it work.

I think the problem is the same in poky distribution, the render group is not created in the build, but listen to the privileges of /etc/dri folders

root@raspberrypi4:/dev/dri# ls -al
drwxr-xr-x  3 root root        120 Aug  7 07:46 .
drwxr-xr-x 16 root root       3720 Aug  7 07:46 ..
drwxr-xr-x  2 root root        100 Aug  7 07:46 by-path
crw-rw----  1 root video  226,   0 Aug  7 07:46 card0
crw-rw----  1 root video  226,   1 Aug  7 07:46 card1
crw-rw-rw-  1 root render 226, 128 Aug  7 07:46 renderD128

I'm not practiced very well in YOCTO, I've tried to create group manually but the graphics performance are the same.

The other problem of the black screen is because the QtQuick in this case works different and is necessary to use the QScreen to create the QQuickView and attach the QML.

See https://doc.qt.io/qt-5/embedded-linux.html to see how use QScreen.
I've tried and its working.

So how to fix this with yocto I'm not able but I think the solution is above.
At the moment the video is not accelerated in the cross compiled image too (tried H265) in QML and GStreamer using kmssink.

Best regards
Enrico

mozcelikors, 2019-08-07:
@peel70 I tried to reproduce your solution
```
root@raspberrypi4:~# groupadd render
root@raspberrypi4:~# usermod -a -G render root
root@raspberrypi4:~# groups root
render root
```
In the application side,
```
QVector<QQuickView *> views;
	for (QScreen *screen : app.screens()) {
		QQuickView *view = new QQuickView;
		view->setScreen(screen);
		view->setResizeMode(QQuickView::SizeRootObjectToView);
		view->setSource(QUrl("qrc:/qml/main.qml"));
		QObject::connect(view->engine(), &QQmlEngine::quit, qGuiApp, &QCoreApplication::quit);
		views.append(view);
		view->showFullScreen();
	}
```

Then when I try with vc4-fkms-v3d on Yocto, I get no performance improvement like you.
There must be that is done in Raspbian that we are missing.

@agherzan @kraj  I did not see in meta-raspberrypi sources any reference to "linux-rasp-pi3-vc4-g++". Qt compilation is done with -xplatform linux-oe-g++ (qtbase_git.bb). Maybe this could be the problem. In order to have hardware acceleration we might have to compile qt with -device linux-rasp-pi3-vc4-g++ and set the other configure flags accordingly for "-device" compilation and not "-xplatform" compilation. The Raspbian compilation that @peel70 tried is I assume something like that for the Raspbian:

 ./configure -release -opengl es2 -device linux-rasp-pi3-vc4-g++ \
   -device-option CROSS_COMPILE=~/raspbian/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf- \
   -sysroot ~/raspbian/sysroot \
   -prefix /usr/local/qt5pi -extprefix ~/raspbian/qt5pi -hostprefix ~/raspbian/qt5 \
   -v -nomake examples -nomake tests -no-use-gold-linker

Maybe we could tailor this in qtbase_git.bbappend in meta-raspberrypi layer.
Does this make sense ?

peel70, 2019-08-07:
Hi @mozcelikors ,
I found a potential solution after finding the  https://bugreports.qt.io/browse/QTBUG-76889?filter=-3 and the relative guide made by Mark at this link http://wapel.de/?p=641
I've checked the Qt's Yocto log.do_configure and the cross compile config.summary and both build 

EGLFS EGLDevice ...................... yes
    EGLFS GBM ............................ yes

that is for the eglfs_kms because instead of EGLFS on Broadcom it use EGLFS on Mesa (if I understand)
So I'm not sure that the problem is in the Qt compilation or in the system build relative to the render group for PI user, not sure necessary for root.

I can confirm that all reported in the https://doc.qt.io/qt-5/embedded-linux.html works. I was able to configure the json and clone HDMI1 to HDMI2 or with QScreen play two different QML at the same time with CPU under the 7% (full HD not 4K), maybe a difference on that line

QQuickView *view = new QQuickView;
to 
QQuickView *view = new QQuickView();

the problem, for me, is that the H265 video si not accelerated (and is what I need), in gstreamer 1.14 and 1.17 I think there is something missing somewhere (maybe OpenMax need manual compilation with the new VC library?)

mozcelikors, 2019-08-07:
@peel70 You were able to do this on Raspbian, is that correct?

> I can confirm that all reported in the https://doc.qt.io/qt-5/embedded-linux.html works. I was able to configure the json and clone HDMI1 to HDMI2 or with QScreen play two different QML at the same time with CPU under the 7% (full HD not 4K), maybe a difference on that line

Also you mean that gstreamer acceleration does not work in Raspbian, right?

I am currently interested in pure Yocto solution since my distro developments depend on Qt-EGLFS. I am hoping to find a solution in Yocto, and I thought I'd give this a shot. I will let you know if it helps.

peel70, 2019-08-08:
Hi @mozcelikors,
yes I confirm its working on Raspbian, there is a problem on QtMultimedia with gstreamer with the video acceleration (like the previous RPI3,2,1,0)
I hoping that this will be fixed.

agherzan, 2019-08-08:
I tried to compare it against rpi3 with kmscube:

```
root@raspberrypi3-64:~# kmscube
Using display 0x55758f81f0 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_a"
  display extensions: "EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 2.0 Mesa 19.1.1"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_i"
===================================

root@raspberrypi4-64:~# kmscube
Using display 0x55a3d701e0 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_a"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_"
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.1 Mesa 19.1.1"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "VMware, Inc."
  renderer: "softpipe"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 e
GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_text[   18.710025] random: 7 urandom warning(s) mig
ure_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL"
===================================
Using modifier ffffffffffffff
failed to set mode: Invalid argument
```
It takes the wrong renderer on rpi4.

agherzan, 2019-08-08:
I think the core of the issue (or at least part of it) is that mesa doesn't have the right driver enabled (v3d).

peel70, 2019-08-08:
Hi @agherzan ,
this is the output of the Raspian Burster

pi@raspberrypi:~/kmscube $ ./kmscube
Using display 0x1ef1710 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.0 Mesa 19.1.0-devel"
  shading language version: "OpenGL ES GLSL ES 3.00"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_texture_query_lod "
===================================

gitgonewithwind, 2019-08-08:
> I think the core of the issue (or at least part of it) is that mesa doesn't have the right driver enabled (v3d).

Thank you. I got it.

agherzan, 2019-08-08:
I'm onto something. I'll update soon.

gitgonewithwind, 2019-08-08:
> I'm onto something. I'll update soon.

good news. I will try it.

peel70, 2019-08-08:
hi @agherzan ,
this is the output from Raspian Burster

```
pi@raspberrypi:~/kmscube $ ./kmscube
Using display 0x1ef1710 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query
 EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions 
EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm 
EGL_MESA_platform_surfaceless"

display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import 
EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs 
EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync 
EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image 
EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image 
EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base 
EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync 
EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync 
EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export 
EGL_WL_bind_wayland_display "

===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.0 Mesa 19.1.0-devel"
  shading language version: "OpenGL ES GLSL ES 3.00"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays 
GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 
GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer 
GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D 
GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear 
GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode 
GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil 
GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level 
GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth 
GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync 
GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg 
GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth 
GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range 
GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map 
GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float 
GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_integer_mix 
GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture 
GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp 
GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex 
GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_EXT_float_blend GL_KHR_no_error 
GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 
GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_texture_query_lod 
"
===================================
```

agherzan, 2019-08-08:
@peel70 I've done that too. I know that it's picked correctly there.

agherzan, 2019-08-08:
@peel70 but do  you get any output on hdmi?

peel70, 2019-08-08:
@agherzan if you mean on Raspian yes and QML is accelerated, if you mean with the master branch of Yocto, yes but not accelerated

agherzan, 2019-08-08:
OK. So you are using rpi4 (32b right?). Have you tried kmscube on Yocto?

peel70, 2019-08-08:
right rpi4 32bit. No I've not tried kmscube on Yocto

peel70, 2019-08-08:
@agherzan is not installed on my Yocto's image, I've to rebuild it to try, I've noticed that there is not the /opt/vc folder in the image. Maybe the problem is this?

gitgonewithwind, 2019-08-08:
I'm going to try kmscube

kraj, 2019-08-08:
@agherzan probably we should disable vc4 graphics for 32bit rpi4 for now.

agherzan, 2019-08-08:
@kraj It's really annoying because it works on raspbian. With vc4 fkms. I just miss something which makes mesa select software renderer.

gitgonewithwind, 2019-08-08:
in yocto:

raspberrypi4:/home/myuser# export DISPLAY=:0                                                                                                                                                                
raspberrypi4:/home/myuser# kmscube                                                                                                                                                                          
drmModeGetResources failed: Invalid argument                                                                                                                                                                
failed to initialize legacy DRM

peel70, 2019-08-08:
@gitgonewithwind with rpi4 and kms you don't need to export DISPLAY it's not working if you export DISPLAY

gitgonewithwind, 2019-08-08:
my start up script:
#!/bin/sh                                                                                                                                                                                                   
                                                                                                                                                                                                            
export DISPLAY=:0                                                                                                                                                                                           
                                                                                                                                                                                                            
xset s off                                                                                                                                                                                                  
xset dpms 0 0 0                                                                                                                                                                                             
xset -dpms         
..

kernel  boot up output:

....

starting mountd: done                                                                                                                                                                                       
Starting system log daemon...0                                                                                                                                                                              
Starting kernel log daemon...0                                                                                                                                                                              
Starting crond: OK                                                                                                                                                                                          
Modifiers specified, but DRI is too old                                                                                                                                                                     
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.                                                                                                                                            
The XKEYBOARD keymap compiler (xkbcomp) reports:                                                                                                                                                            
> Warning:          Unsupported high keycode 372 for name <I372> ignored                                                                                                                                    
>                   X11 cannot support keycodes above 255.                                                                                                                                                  
>                   This warning only shows for the first high keycode.                                                                                                                                     
Errors from xkbcomp are not fatal to the X server                                                                                                                                                           
[   38.050521] ------------[ cut here ]------------                                                                                                                                                         
[   38.055342] WARNING: CPU: 0 PID: 416 at drivers/gpu/drm/vc4/vc4_firmware_kms.c:899 vc4_crtc_atomic_flush+0x98/0x9c [vc4]                                                                                 
[   38.066414] Modules linked in: nfsd ipv6 i2c_dev brcmfmac brcmutil sha256_generic vc4 bcm2835_v4l2(C) snd_soc_core bcm2835_codec(C) cfg80211 v4l2_mem2mem snd_compress spidev joydev snd_pcm_dmaengine bd
[   38.110057] CPU: 0 PID: 416 Comm: Xorg Tainted: G         C        4.19.58 #1                                                                                                                            
[   38.117300] Hardware name: BCM2835                                                                                                                                                                       
[   38.120780] [<c021322c>] (unwind_backtrace) from [<c020d680>] (show_stack+0x20/0x24)                                                                                                                     
[   38.128648] [<c020d680>] (show_stack) from [<c09d815c>] (dump_stack+0xbc/0x100)                                                                                                                          
[   38.136076] [<c09d815c>] (dump_stack) from [<c0222ca8>] (__warn.part.0+0xcc/0xe8)                                                                                                                        
[   38.143681] [<c0222ca8>] (__warn.part.0) from [<c0222e5c>] (warn_slowpath_null+0x54/0x5c)                                                                                                                
[... cut ...]
raspberrypi4:/home/myuser# kmscube                                                                                                                                                                          
drmModeGetResources failed: Invalid argument                                                                                                                                                                
failed to initialize legacy DRM                                                                                                                                                                             
raspberrypi4:/home/myuser#

peel70, 2019-08-08:
@gitgonewithwind  I've build Yocto without X11 like Burster Ligth to have eglfs_kms performance

mozcelikors, 2019-08-08:
Let's please focus on the issue at hand. Our purpose is to improve the performance of OpenGL/ES based applications on meta-raspberrypi. @agherzan seems to be close to a solution, that is mesa selecting software renderer on Yocto as opposed to Raspbian, we should stick to it. I think unrelated questions should be asked on different issues to avoid flooding.

gitgonewithwind, 2019-08-08:
> Burster Ligth

I think I need X11 to run chromium. is it right?

gitgonewithwind, 2019-08-08:
> @gitgonewithwind I've build Yocto without X11 like Burster Ligth to have eglfs_kms performance

/usr/bin/xli -onroot -center -zoom 100 /home/root/logo.png 
my company log can show. but very slow to show and display. think it's the same issue with yours. I am not sure.

agherzan, 2019-08-09:
Posted an update in https://github.com/agherzan/meta-raspberrypi/issues/461 as I reckon that's the root of these issues. TLDR: Not fixed yet.

peel70, 2019-08-15:
Hi @agherzan 
I want to summarize my tests:

1) Raspbian Buster Light + Qt 5.12.4 Cross Compiled
- kmscube works
- Qt see two QScreen and I'm able to mirror or put different contents on the two screens at the same time on **/dev/dri/card1**

2) Bitbake Yocto Master branch, meta-raspberrypi vc4 branch + @kraj hardcoded options Qt 5.12.3

-  64 bit image kmscube not working (maybe my mistake in the build), Qt5 perfomance bad

-  32 bit image kmscube works only in **/dev/dri/card0**, Qt5 performance good but Qt see only 1 Screen and work on **/dev/dri/card0** the secondary screen remain with the shaded rectangle as boot. Try to pass a screen configuration as reported here https://doc.qt.io/qt-5/embedded-linux.html cause application crash; tvservice -l  command see both devices

Maybe a problem in the linux raspberry firmware?

agherzan, 2019-08-15:
@peel70 https://github.com/agherzan/meta-raspberrypi/pull/476/files should make this work.

peel70, 2019-08-15:
@agherzan I will try in this days and give you a feedback.

gitgonewithwind, 2019-08-15:
@agherzan 
the testing result for your update ag/rpi4-graphics branch is same as yesterday:

pberrypi4:/home/myuser# kmscube -D /dev/dri/card1
Using display 0x2d9a90 with EGL version 1.4

EGL information:
version: "1.4"
vendor: "Mesa Project"
client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_dev"
display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_"

OpenGL ES 2.x information:
version: "OpenGL ES 3.0 Mesa 19.1.1"
shading language version: "OpenGL ES GLSL ES 3.00"
vendor: "Broadcom"
renderer: "V3D 4.2"
extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_form"

failed to set mode: Permission denied
raspberrypi4:/home/myuser# kmscube -D /dev/dri/card0
drmModeGetResources failed: Invalid argument
failed to initialize legacy DRM
raspberrypi4:/home/myuser# kmscube
drmModeGetResources failed: Invalid argument
failed to initialize legacy DRM

I try again, my card1 is same as your card0, card0 is same as your card1. but I have a utput:
failed to set mode: Permission denied what's that? why?

root is same.
-------------------------
bitbake  core-image-full-cmdline and bitbake core-image-x11   for kmscube  test results are same,chromium no display.
but bitbake core-image-x11  image  can show  touch screen calibration  vim  and vim.
now I am going to add chromium in local.conf for bitbake core-image-x11  to test.

agherzan, 2019-08-15:
@gitgonewithwind what user are you running that command as?

gitgonewithwind, 2019-08-15:
@agherzan 
root

agherzan, 2019-08-15:
I have not tried chromium but I can say that kmscube and qt5 demos work as expected.

gitgonewithwind, 2019-08-15:
> I have not tried chromium but I can say that kmscube and qt5 demos work as expected.

so you mean chromium should work well? I am compiling the chromium-x11... in bitbake core-image-x11 and default local.conf ...

agherzan, 2019-08-15:
chromium is on top of X which I haven't tried. I only worked over eglfs.

gitgonewithwind, 2019-08-15:
> chromium is on top of X which I haven't tried. I only worked over eglfs.

can you try later? I have spent more than 3 weeks, I worry about my manager is not happy. thanks

agherzan, 2019-08-15:
Sadly I can't because I don't have the time for it.

[7 more replies]

closed 2019-08-16 by agherzan