agherzan/meta-raspberrypi #456: rpi-test-image unbuildable because of crda (new issue)

MarcoBorrini99, 2019-07-28:
**Description**

With lastest master, bitbake fails to build rootfs for a Rpi4 rpi-test-image because of crda.

**Steps to reproduce the issue:**
1. clone meta-raspberrypi master branch
2. clone meta-openembedded master branch
3. create building env:
```
[marco@Port371]: ~/src/poky>$ source oe-init-build-env rpi-build
```
4. edit conf/local.conf adding:
```
MACHINE ??= "raspberrypi4"
PACKAGE_CLASSES ?= "package_ipk"
```
5. edit conf/bblayers.conf as:
```
BBLAYERS ?= " 							\
  /home/marco/src/poky/meta 					\
  /home/marco/src/poky/meta-poky 				\
  /home/marco/src/poky/meta-yocto-bsp 				\
  /home/marco/src/poky/meta-openembedded/meta-oe 		\
  /home/marco/src/poky/meta-openembedded/meta-multimedia 	\
  /home/marco/src/poky/meta-openembedded/meta-networking 	\
  /home/marco/src/poky/meta-openembedded/meta-python 		\
  /home/marco/src/poky/meta-raspberrypi				\
  "
```
6. start buildig image
`[marco@Port371]: ~/src/poky/rpi-build>$ bitbake rpi-test-image`

**Describe the results you received:**
```
[marco@Port371]: ~/src/poky/rpi-build>$ bitbake rpi-test-image
WARNING: Host distribution "arch" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |#################################################################################################| Time: 0:00:00
Loaded 3381 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.43.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi4"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7+snapshot-20190728"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "master:1a4e4fb6b0a9d54641bd4193e95311d1f822a9ca"
meta-oe              
meta-multimedia      
meta-networking      
meta-python          = "master:5f74c657c06b79394ddd5e532e1e10ee49a9f9c5"
meta-raspberrypi     = "master:6c175288cc81554b765c99453dc96fd4570205dd"

Initialising tasks: 100% |############################################################################################| Time: 0:00:02
Sstate summary: Wanted 2 Found 0 Missed 2 Current 1324 (0% match, 99% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
ERROR: rpi-test-image-1.0-r0 do_rootfs: Unable to install packages. Command '/home/marco/src/poky/rpi-build/tmp/work/raspberrypi4-poky-linux-gnueabi/rpi-test-image/1.0-r0/recipe-sysroot-native/usr/bin/opkg --volatile-cache -f /home/marco/src/poky/rpi-build/tmp/work/raspberrypi4-poky-linux-gnueabi/rpi-test-image/1.0-r0/opkg.conf -t /home/marco/src/poky/rpi-build/tmp/work/raspberrypi4-poky-linux-gnueabi/rpi-test-image/1.0-r0/temp/ipktemp/ -o /home/marco/src/poky/rpi-build/tmp/work/raspberrypi4-poky-linux-gnueabi/rpi-test-image/1.0-r0/rootfs  --force_postinstall --prefer-arch-to-version   install packagegroup-base-extended packagegroup-core-boot packagegroup-rpi-test psplash-raspberrypi run-postinsts' returned 1:
Collected errors:
 * Solver encountered 1 problem(s):
 * Problem 1/1:
 *   - package packagegroup-rpi-test-1.0-r0.all requires crda, but none of the providers can be installed
 * 
 * Solution 1:
 *   - do not ask to install a package providing packagegroup-rpi-test

 * Solution 2:
 *   - do not ask to install a package providing packagegroup-base-extended

ERROR: rpi-test-image-1.0-r0 do_rootfs: 
ERROR: rpi-test-image-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/marco/src/poky/rpi-build/tmp/work/raspberrypi4-poky-linux-gnueabi/rpi-test-image/1.0-r0/temp/log.do_rootfs.29879
ERROR: Task (/home/marco/src/poky/meta-raspberrypi/recipes-core/images/rpi-test-image.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3752 tasks of which 3751 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/marco/src/poky/meta-raspberrypi/recipes-core/images/rpi-test-image.bb:do_rootfs
Summary: There was 1 WARNING message shown.
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
```

**Describe the results you expected:**
A not failing process, crda seems to me already be referenced by above added layers.

**Additional information you deem important (e.g. issue happens only occasionally):**
Same error if I try to build for Rpi3

**Additional details (revisions used, host distro, etc.):**
Each cloned repository are master branches, Host distro is ArchLinux (I know, it's not officiali supported) and also a virtualized Ubuntu 18.04.02; to me seems a different issue from #78

agherzan, 2019-07-30:
@MarcoBorrini99 Can you retest with master's HEAD? It should be fixed now.

MarcoBorrini99, 2019-07-30:
> @MarcoBorrini99 Can you retest with master's HEAD? It should be fixed now.

I've just deleted previous build directory and started from scratch with no luck, tomorrow I'll try with a new and clean yocto installation.

```
[marco@Port371]: ~/src/poky/rpi-build>$ bitbake rpi-test-image
WARNING: Host distribution "arch" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |##########################################################################################################################################################################################################| Time: 0:00:00
Loaded 3383 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.43.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi4"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7+snapshot-20190730"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "master:1a4e4fb6b0a9d54641bd4193e95311d1f822a9ca"
meta-oe              
meta-multimedia      
meta-networking      
meta-python          = "master:2bdd4a599558c93385642b17b27d5cce31911dd6"
meta-raspberrypi     = "master:414831c102849246fecab0f0cf7dcb9a3494d3fc"

Initialising tasks: 100% |#####################################################################################################################################################################################################| Time: 0:00:02
Sstate summary: Wanted 2 Found 0 Missed 2 Current 1288 (0% match, 99% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
ERROR: rpi-test-image-1.0-r0 do_rootfs: Unable to install packages. Command '/home/marco/src/poky/rpi-build/tmp/work/raspberrypi4-poky-linux-gnueabi/rpi-test-image/1.0-r0/recipe-sysroot-native/usr/bin/opkg --volatile-cache -f /home/marco/src/poky/rpi-build/tmp/work/raspberrypi4-poky-linux-gnueabi/rpi-test-image/1.0-r0/opkg.conf -t /home/marco/src/poky/rpi-build/tmp/work/raspberrypi4-poky-linux-gnueabi/rpi-test-image/1.0-r0/temp/ipktemp/ -o /home/marco/src/poky/rpi-build/tmp/work/raspberrypi4-poky-linux-gnueabi/rpi-test-image/1.0-r0/rootfs  --force_postinstall --prefer-arch-to-version   install packagegroup-base-extended packagegroup-core-boot packagegroup-rpi-test psplash-raspberrypi run-postinsts' returned 1:
Collected errors:
 * Solver encountered 1 problem(s):
 * Problem 1/1:
 *   - package packagegroup-rpi-test-1.0-r0.all requires wireless-regdb, but none of the providers can be installed
 * 
 * Solution 1:
 *   - do not ask to install a package providing packagegroup-rpi-test

 * Solution 2:
 *   - do not ask to install a package providing packagegroup-base-extended

ERROR: rpi-test-image-1.0-r0 do_rootfs: 
ERROR: rpi-test-image-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/marco/src/poky/rpi-build/tmp/work/raspberrypi4-poky-linux-gnueabi/rpi-test-image/1.0-r0/temp/log.do_rootfs.22752
ERROR: Task (/home/marco/src/poky/meta-raspberrypi/recipes-core/images/rpi-test-image.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3652 tasks of which 3651 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/marco/src/poky/meta-raspberrypi/recipes-core/images/rpi-test-image.bb:do_rootfs
Summary: There was 1 WARNING message shown.
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
```

kraj, 2019-07-30:
it seems that I introduced a typo it should be `wireless-regdb-static`

MarcoBorrini99, 2019-07-30:
> it seems that I introduced a typo it should be `wireless-regdb-static`

Yes, I can boot my rpi4 now; I only had to disable wiringpi (git.dragon.net seems to be not rachable) and bigbuckbunny-* files (too long to download). Thanks, now I can continue to work on my embedded project.

closed 2019-08-03 by MarcoBorrini99