agherzan/meta-raspberrypi #460: bthelper fails due to RF-killed Bluetooth on thud

sashko, 2019-08-02:
```

Bluetooth radio happens to be disabled on boot which leads to `bthelper@hci0` service failing.

   Active: [[0;1;31mfailed[[0m (Result: exit-code) since Fri 2019-08-02 07:33:11 UTC; 8min ago
  Process: 802 ExecStart=/usr/bin/bthelper hci0 [[0;1;31m(code=exited, status=1/FAILURE)[[0m
 Main PID: 802 (code=exited, status=1/FAILURE)

Aug 02 07:33:11 localhost systemd[1]: Started Raspberry Pi bluetooth helper.
Aug 02 07:33:11 localhost bthelper[802]: Can't init device hci0: Operation not possible due to RF-kill (132)
Aug 02 07:33:11 localhost systemd[1]: [[0;1;39m[[0;1;31m[[0;1;39mbthelper@hci0.service: Main process exited, code=exited, status=1/FAILURE[[0m
Aug 02 07:33:11 localhost systemd[1]: [[0;1;39m[[0;1;31m[[0;1;39mbthelper@hci0.service: Failed with result 'exit-code'.[[0m
```

**Steps to reproduce the issue:**
1. Use latest version of `thud`
2. Build an image
3. Execute `systemctl status bthelper@hci0.service` or `systemctl --no-pager --failed`

**Describe the results you received:**

`bthelper@hci0` service failed.

**Describe the results you expected:**

`bthelper@hci0` service does not fail if Bluetooth radio is disabled.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

kraj, 2019-08-03:
Can you also reproduce this on master please ?

sashko, 2019-08-03:
> Can you also reproduce this on master please ?

Sure, I will try to find time to do some.

Do you have an idea why would this happen though?

kraj, 2019-08-03:
I think it could be kernel version

agherzan, 2019-08-06:
Master and warrior the current maintained (up to date branches). Also, you are not mentioning any machines on which you saw this issue. I have tested rpi3 and rpi4 on warrior/master and bt works as expected. 

What does `rfkill list` say?

sashko, 2019-08-06:
> Also, you are not mentioning any machines on which you saw this issue.

Raspberry Pi 3 B+.

> What does `rfkill list` say?

```
0: phy0: wlan
	Soft blocked: yes
	Hard blocked: no
1: hci0: bluetooth
	Soft blocked: yes
	Hard blocked: no
```

agherzan, 2019-08-06:
I see. Try to replicate first on master. Is updating to warrior an option for you?

sashko, 2019-08-08:
> I see. Try to replicate first on master. Is updating to warrior an option for you?

Same issue on `Warrior`.

agherzan, 2021-10-30:
@sashko this looks like an old issue. Is there any progress on it? We don't maintain thud anymore so I'll be closing this but feel free to reopen if you reproduce on current master.

closed 2021-10-30 by agherzan