agherzan/meta-raspberrypi #461: RaspberryPi 4 DRM not working with YoE

psivasubramanian, 2019-08-02:
**Description**
Hi,
Our team is trying to bring up Rpi4 using Yoe build (master branch) instructions, but facing below issue with DRM. Rpi4 boots up properly and displays command terminal through HDMI display. Though both DRM card0 and card1 are available, graphics rendering is not happening even with kmscube (tried westeros wayland compositor also). 

**Describe the results you received:**
root@raspberrypi4:~# kmscube -D /dev/dri/card1
Using display 0x1f39ef0 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_image_dma_buf_export "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.1 Mesa 19.1.1"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "VMware, Inc."
  renderer: "softpipe"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_OES_viewport_array GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_copy_image GL_EXT_draw_buffers_indexed GL_EXT_draw_elements_base_vertex GL_EXT_gpu_shader5 GL_EXT_primitive_bounding_box GL_EXT_render_snorm GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_buffer GL_EXT_texture_cube_map_array GL_EXT_texture_norm16 GL_EXT_texture_view GL_KHR_context_flush_control GL_NV_image_formats GL_OES_copy_image GL_OES_draw_buffers_indexed GL_OES_draw_elements_base_vertex GL_OES_gpu_shader5 GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_buffer GL_OES_texture_cube_map_array GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_blend_func_extended GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_clip_cull_distance GL_EXT_disjoint_timer_query GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_EXT_texture_compression_bptc GL_KHR_parallel_shader_compile GL_EXT_texture_sRGB_R8 GL_EXT_depth_clamp GL_EXT_texture_query_lod "
===================================
Using modifier ffffffffffffff
failed to set mode: Invalid argument

Tried with "card0" also as
root@raspberrypi4:~# kmscube -D /dev/dri/card0
drmModeGetResources failed: Invalid argument
failed to initialize legacy DRM

**Steps to reproduce the issue:**

1. git clone git@github.com:YoeDistro/yoe-distro.git and cd yoe-distro
2. **.** raspberrypi4-envsetup.sh 
3. yoe_setup
4. yoe_add_layer git@github.com:WebPlatformForEmbedded/meta-wpe master
5. Add the following lines to conf/local.conf
DISTRO_FEATURES_remove_rpi = "x11"
6. Add "kmscube" to wpe-westeros-image (meta-wpe/recipes-core/images/wpe-westeros-image.bb)
7. Build an image: bitbake wpe-westeros-image
8. Insert SD card & run lsblk (note sd card device, and substitute for /dev/sdX below)
9. yoe_install_image /dev/sdX wpe-westeros-image
10. Boot Rpi4 with SD card & run kmscube as follows

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

kraj, 2019-08-03:
@psivasubramanian  did you try setting 
```
conf/machine/raspberrypi4.conf:VC4DTBO ?= "vc4-fkms-v3d"
```
to `vc4-kms-v3d`

mozcelikors, 2019-08-04:
@kraj  I am having the same problem with @psivasubramanian and using vc4-kms-v3d completely breaks things in Raspberry Pi 4 (No HDMI output and devicetree complains about a lot of things). I feel like it definitely should be properly ported to Raspberry Pi 4. @psivasubramanian Please do share your results with us as well.

psivasubramanian, 2019-08-05:
@kraj Yes @mozcelikors, switching to vc4-kms-v3d make things worse on Rpi4 (No HDMI output).

agherzan, 2019-08-05:
@mozcelikors is this a duplicate of #453?

mozcelikors, 2019-08-05:
@agherzan I didnt try with his distro but I think root cause might be the same.

agherzan, 2019-08-09:
Here is a small update. The issue seems to be that mesa doesn't select the right driver and falls back to software renderer in Yocto. On Raspbian, this doesn't happen. Tested with kmscube. RaspberryPi 4 uses Videocore 6 which needs a new mesa driver, v3d. Also, on the kernel side there was another issue where the rpi4-64 didn't enable the v3d driver (the 32bit counterpart does/did). All these are needed but even with these I can't persuade mesa to use the broadcom renderer.

I worked on this yesterday and sadly I'll have to postpone this as I'll travel for the next 5 days. If nobody has any progress, I'll get back to it as soon as. If anyone looks into it, keep in mind that both vc4 and v3d drivers are needed. This is because the kernel side does the 3D in v3d and the drm in v4. On older Raspberry Pis both happened in vc4. 

I will push the wip branch which includes some component updates and the bits I'm mentioning above.

aurelian17, 2019-08-09:
@agherzan please specify the wip branch. I will try to give it a try this weekend if possible.

agherzan, 2019-08-09:
@aurelian17 That is great. I just pushed the branch as well: https://github.com/agherzan/meta-raspberrypi/tree/ag/rpi4-graphics . Keep us updated.

aurelian17, 2019-08-10:
Can someone please post the output of the following command on RPI3: cat /sys/dev/char/226\:128/device/uevent

kraj, 2019-08-11:
@aurelian17 
```
root@raspberrypi3-64:~# cat /sys/dev/char/226:128/device/uevent
cat: '/sys/dev/char/226:128/device/uevent': No such file or directory
```

peel70, 2019-08-12:
@aurelian17 hello,
today I've build both 32 and 64 bit image for RPI4 with the new branch poster by @agherzan but the output of the kmscube is this:

```
root@raspberrypi4-64:~# kmscube
Using display 0x5581497240 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_debug EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_image_dma_buf_export "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.1 Mesa 19.1.1"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "VMware, Inc."
  renderer: "softpipe"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_OES_viewport_array GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_copy_image GL_EXT_draw_buffers_indexed GL_EXT_draw_elements_base_vertex GL_EXT_gpu_shader5 GL_EXT_primitive_bounding_box GL_EXT_render_snorm GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_buffer GL_EXT_texture_cube_map_array GL_EXT_texture_norm16 GL_EXT_texture_view GL_KHR_context_flush_control GL_NV_image_formats GL_OES_copy_image GL_OES_draw_buffers_indexed GL_OES_draw_elements_base_vertex GL_OES_gpu_shader5 GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_buffer GL_OES_texture_cube_map_array GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_blend_func_extended GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_clip_cull_distance GL_EXT_disjoint_timer_query GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_EXT_texture_compression_bptc GL_KHR_parallel_shader_compile GL_EXT_texture_sRGB_R8 GL_EXT_depth_clamp GL_EXT_texture_query_lod "
===================================
Using modifier ffffffffffffff
failed to set mode: Invalid argument
```
In both the  images there's not the opt/vc fodfer, I think is important for che management of the VideoCore.
Maybe is may mistake in the local.conf?

Regards
Enrico

gitgonewithwind, 2019-08-12:
> @aurelian17 hello,
> today I've build both 32 and 64 bit image for RPI4 with the new branch poster by @agherzan but the output of the kmscube is this:
>
> ```
> root@raspberrypi4-64:~# kmscube
> Using [...]

but for me, the output is same as before(MACHINE = "raspberrypi4"
):

raspberrypi4:/etc/X11/Xsession.d# kmscube
drmModeGetResources failed: Invalid argument
failed to initialize legacy DRM

gitgonewithwind, 2019-08-12:
in branch ag/rpi4-graphics

raspberrypi4:/home/myuser# cat /sys/dev/char/226\:0/device/uevent               
DRIVER=v3d                                                                      
OF_NAME=v3d                                                                     
OF_FULLNAME=/v3dbus/v3d@7ec04000                                                
OF_COMPATIBLE_0=brcm,2711-v3d                                                   
OF_COMPATIBLE_N=1                                                               
MODALIAS=of:Nv3dT<NULL>Cbrcm,2711-v3d

vrziny, 2019-08-12:
@agherzan, @psivasubramanian, @gitgonewithwind 
I found some clues in Mesa after digging into it. Mesa seems to introduce KMSRO since 19.0 for better infrastructure.
In short, add v3d and kmsro to bbappend file like a line below if you use Mesa 19.x.
GALLIUMDRIVERS_append_rpi = ",vc4,v3d,kmsro"
kmscube and weston work very well now.

kraj, 2019-08-12:
This is a common driver, it should be applied to main recipe in oe-core, and infact vc4 as well should move there, we should only apply v3d here

vrziny, 2019-08-13:
@kraj I totally agree with your opinion. This is just one of several ways to enable h/w acceleration for 3d application like kmscube. It should be applied to their recipe. I hope it can help who struggle with it at this time.

peel70, 2019-08-13:
Hi @vrziny,
I've modified bbappend as you and rebuild image raspberrypi4-64 (no X11 or weston) and the problem is not fixed, kmscube don't work.
Maybe something wrong in my configuration?

Raspbian Buster ligth (no X11) works good

vrziny, 2019-08-13:
Hi @peel70,
Your error log comes from failure of drmModeSetCrtc. Have you run modetest? You should compare the result of modetest and dmesg with raspbian. Actually I'm not using YoE, but it's based on yocto project(32 bit).

peel70, 2019-08-13:
hi @vrziny,
I'm using yocto too, I've checked on 64bit and is not working, with 32bit its working now.
I think there are other changes to do.
I'm using Qt5.12.3 and try a test app; it's accelerated, but Qt only see 1 Screen, but tvservice show both. With the original raspbian burster light and this Qt's guide https://doc.qt.io/qt-5/embedded-linux.html I was able to use both hdmi with different QML for each.
I know that the device is new and warrior branch too, but something work now.

gitgonewithwind, 2019-08-13:
hi @peel70 @vrziny 
I modified the file as GALLIUMDRIVERS_append_rpi = ",vc4,v3d,kmsro" .    in yocto  branch ag/rpi4-graphics
when starting up, my company logo display is faster than before as normal. but the x11 chromium still does not display. my issue is #467.   so you patch is useful  at somewhere for @vrziny and for my logo display. thanks

but:

raspberrypi4:/home/myuser# kmscube                                                                                                                                                          
drmModeGetResources failed: Invalid argument                                                                                                                                                
failed to initialize legacy DRM

vrziny, 2019-08-13:
@gitgonewithwind,
Have you try "kmscube -D /dev/dri/card1"?

peel70, 2019-08-13:
Hi @vrziny,
this is my output of the above command:
```
root@raspberrypi4:~# kmscube -D /dev/dri/card1
drmModeGetResources failed: Invalid argument
failed to initialize legacy DRM

root@raspberrypi4:~# kmscube -D /dev/dri/card0
Using display 0x1354b98 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_debug EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.0 Mesa 19.1.1"
  shading language version: "OpenGL ES GLSL ES 3.00"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_texture_query_lod "
===================================
root@raspberrypi4:~#
```

It seems to work on card0 and not on card1
I'm not at office at the moment and I cannot make other tests

gitgonewithwind, 2019-08-13:
@peel70 

raspberrypi4:/home/myuser# kmscube                                              
drmModeGetResources failed: Invalid argument                                    
failed to initialize legacy DRM                                                 
raspberrypi4:/home/myuser# kmscube -D /dev/dri/card1                            
Using display 0x1521a90 with EGL version 1.4                                    
===================================                                             
EGL information:                                                                
  version: "1.4"                                                                
  vendor: "Mesa Project"                                                        
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_dev"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_"
===================================                                             
OpenGL ES 2.x information:                                                      
  version: "OpenGL ES 3.0 Mesa 19.1.1"                                          
  shading language version: "OpenGL ES GLSL ES 3.00"                            
  vendor: "Broadcom"                                                            
  renderer: "V3D 4.2"                                                           
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_form"
===================================                                             
failed to set mode: Permission denied                                           
raspberrypi4:/home/myuser# kmscube -D /dev/dri/card0                            
drmModeGetResources failed: Invalid argument                                    
failed to initialize legacy DRM

gitgonewithwind, 2019-08-13:
@peel70 

### raspberrypi4:/home/myuser# kmscube
drmModeGetResources failed: Invalid argument                                    
failed to initialize legacy DRM                                                 
raspberrypi4:/home/myuser# kmscube -D /dev/dri/card1                            
Using display 0x1521a90 with EGL version 1.4                                    
===================================                                             
EGL information:                                                                
  version: "1.4"                                                                
  vendor: "Mesa Project"                                                        
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_dev"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_"
===================================                                             
OpenGL ES 2.x information:                                                      
  version: "OpenGL ES 3.0 Mesa 19.1.1"                                          
  shading language version: "OpenGL ES GLSL ES 3.00"                            
  vendor: "Broadcom"                                                            
  renderer: "V3D 4.2"                                                           
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_form"
===================================                                             
failed to set mode: Permission denied                                           
raspberrypi4:/home/myuser# kmscube -D /dev/dri/card0                            
drmModeGetResources failed: Invalid argument                                    
failed to initialize legacy DRM

gitgonewithwind, 2019-08-13:
@peel70 

raspberrypi4:/home/myuser# sudo kmscube                                         
drmModeGetResources failed: Invalid argument                                    
failed to initialize legacy DRM                                                 
raspberrypi4:/home/myuser# sudo kmscube -D /dev/dri/card0                       
drmModeGetResources failed: Invalid argument                                    
failed to initialize legacy DRM                                                 
raspberrypi4:/home/myuser# sudo kmscube -D /dev/dri/card1                       
Using display 0x1c590a0 with EGL version 1.4                                    
===================================                                             
EGL information:                                                                
  version: "1.4"                                                                
  vendor: "Mesa Project"                                                        
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_dev"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_"
===================================                                             
OpenGL ES 2.x information:                                                      
  version: "OpenGL ES 3.0 Mesa 19.1.1"                                          
  shading language version: "OpenGL ES GLSL ES 3.00"                            
  vendor: "Broadcom"                                                            
  renderer: "V3D 4.2"                                                           
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_form"
===================================                                             
failed to set mode: Permission denied                                           
raspberrypi4:/home/myuser#

peel70, 2019-08-13:
@gitgonewithwind 
don't know, I've rebuild the image with @vrziny suggestions and something is working.
Have you try with root user (I see this difference from me)?
I've read that the user must be in the "**render**" group and root is all enabled by default

gitgonewithwind, 2019-08-13:
@peel70
sorry, I used bitbake core-image-x11 to build the image.
I changed to HDMI2 port  to show as my post before. I think my output is same as your.
let me roll back

gitgonewithwind, 2019-08-13:
@peel70  
I think you QT display issue is same as my logo display issue. now I use the patch  GALLIUMDRIVERS_append_rpi = ",vc4,v3d,kmsro"  . my logo display is correct! for me It fixed!

but x11 chromioun no display. it's another issue.

gitgonewithwind, 2019-08-13:
@peel70 

pberrypi4:/home/myuser# kmscube -D /dev/dri/card1
Using display 0x2d9a90 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_dev"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_"
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.0 Mesa 19.1.1"
  shading language version: "OpenGL ES GLSL ES 3.00"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_form"
===================================
failed to set mode: Permission denied
raspberrypi4:/home/myuser# kmscube -D /dev/dri/card0
drmModeGetResources failed: Invalid argument
failed to initialize legacy DRM
raspberrypi4:/home/myuser# kmscube
drmModeGetResources failed: Invalid argument
failed to initialize legacy DRM

----------------
I try again, my card1 is same as your car0, car0 is same as your card1.

peel70, 2019-08-13:
@gitgonewithwind the suggestion worked for me and for my qt test app.
But the secondary HDMI is not working ( and when configure a json file to qt for the second display management, cause a crash of the application.
We are not in the correct post because we are not speaking about YoE but I think the problems are the same.
Maybe something wrong in you local.conf? 
I've removed X11 and wayland from distro features and added opengl

gitgonewithwind, 2019-08-13:
@peel70 
I need run chromium, so I think I need x11. tomorrow I will try your local.conf ( remove X11 and wayland from distro features and added opengl). now it's time to go home.
see you.
the suggestion is useful for my logo display. but chromium no display which is another issue.

peel70, 2019-08-13:
@gitgonewithwind, with my suggestion you are not able to run chromium

agherzan, 2019-08-14:
@vrziny Very good catch with `kmsro`. I'll give it a try tomorrow.

gitgonewithwind, 2019-08-14:
> @gitgonewithwind, with my suggestion you are not able to run chromium
Thanks @peel70  I got it.

aurelian17, 2019-08-15:
GALLIUMDRIVERS_append_rpi = ",vc4,v3d,kmsro" does the trick for the 32bit version. But the 64bit still doesn't work.
And I think that for 32bit is working because of some userland parts that were added to the build.

agherzan, 2019-08-15:
kmscube works with both 32 and 64bit machines with https://github.com/agherzan/meta-raspberrypi/pull/476. @kraj I would hardcode `kmsro` for now so I can backport the patch easily on warrior as well. As soon as the change lands in poky we change.

vrziny, 2019-08-16:
@agherzan That's a good step forward. I'm glad it helped.

agherzan, 2019-08-16:
Fixed by https://github.com/agherzan/meta-raspberrypi/pull/476

closed 2019-08-16 by agherzan