agherzan/meta-raspberrypi #464: Error in do compile for cairo_1.16.0.bb

thesillywhat, 2019-08-06:
**Description**
poky-warrior-21.0.1/meta/recipes-graphics/cairo/cairo_1.16.0bb: do_compile failed with exit code 1

I am trying to create an image for Raspberrypi cm3 64 bits. I followed the steps listed in lissue#97 to create an image for 64 bits. Since there is no machine.conf file available for CM3 64bits.
 

**Steps to reproduce the issue:**
1. add KERNEL_DEVICETREE_append = "broadcom/bcm2710-rpi-cm3.dtb" to local.conf
2. Set MACHINE ?="raspberrypi3-64"
3. Run bitbake core-image-base

**Describe the results you received:**
blue@blue:~/yacto/rpi-qt5_64/build$ bitbake core-image-base
Parsing recipes: 100% |######################################################################################################################################################################| Time: 0:01:24
Parsing of 2264 .bb files complete (0 cached, 2264 parsed). 3343 targets, 124 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7.1"
TUNE_FEATURES        = "aarch64 cortexa53 crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       
meta-oe              
meta-qt5-warrior     
meta-raspberrypi-warrior 
meta-rpi_custom      
meta-python          
meta-networking      = "<unknown>:<unknown>"

Initialising tasks: 100% |###################################################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 1179 Found 254 Missed 925 Current 0 (21% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: cairo-1.16.0-r0 do_compile: oe_runmake failed
ERROR: cairo-1.16.0-r0 do_compile: Function failed: do_compile (log file is located at /home/blue/yacto/rpi-qt5_64/build/tmp/work/aarch64-poky-linux/cairo/1.16.0-r0/temp/log.do_compile.17778)
ERROR: Logfile of failure stored in: /home/blue/yacto/rpi-qt5_64/build/tmp/work/aarch64-poky-linux/cairo/1.16.0-r0/temp/log.do_compile.17778

**Describe the results you expected:**
I was expecting a successful image build since I was able to create an image for CM3 32 bits.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

//////Poky-warrior 21.0.1/////

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7.1"
TUNE_FEATURES        = "aarch64 cortexa53 crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       
meta-oe              
meta-qt5-warrior     
meta-raspberrypi-warrior 
meta-rpi_custom      
meta-python          
meta-networking      = "<unknown>:<unknown>"

kraj, 2019-08-06:
@thesillywhat Please post the exact error, it must be in ` /home/blue/yacto/rpi-qt5_64/build/tmp/work/aarch64-poky-linux/cairo/1.16.0-r0/temp/log.do_compile.17778`

thesillywhat, 2019-08-06:
[log.do_compile.17778.txt](https://github.com/agherzan/meta-raspberrypi/files/3473438/log.do_compile.17778.txt)

kraj, 2019-08-06:
```
/home/blue/yacto/rpi-qt5_64/build/tmp/work/aarch64-poky-linux/cairo/1.16.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/8.3.0/ld: cannot find -lEGL
/home/blue/yacto/rpi-qt5_64/build/tmp/work/aarch64-poky-linux/cairo/1.16.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/8.3.0/ld: cannot find -lGLESv2
collect2: error: ld returned 1 exit status
```

so it seems to not having virtual/libegl in DEPENDS ?

thesillywhat, 2019-08-06:
Is that something I need to add in my local.conf ?

kraj, 2019-08-06:
what does your `./recipes-graphics/cairo/cairo_%.bbappend` in meta-raspberrypi look like ?

kraj, 2019-08-06:
and please post complete build configuration. its missing the critical parts on the right hand side.

thesillywhat, 2019-08-06:
That is my complete build configuration. That is what I see in my terminal window.  Is that wrong?

This is what I see in my bbappend file

PACKAGECONFIG_GLESV2 = " ${@bb.utils.contains('DISTRO_FEATURES', 'x11', '', 'glesv2', d)}"

PACKAGECONFIG_append_rpi = "${@bb.utils.contains('MACHINE_FEATURES', 'vc4graphics', '', ' egl ${PACKAGECONFIG_GLESV2}', d)}"

thesillywhat, 2019-08-06:
I also see that I am missing target FPU for the previous machine configuration it would populate as
target FPU = "hard"

kraj, 2019-08-06:
FPU is always `hard` for armv8, but I use Yoe-Distro and I do not see this build error. So unless I can reproduce it, it wont be easy to suggest what could be going on

thesillywhat, 2019-08-06:
I really need this to work. Can I remove this recipe ? Or is this something integral to the whole image.

agherzan, 2019-08-06:
Are you using user land or vc4?

kraj, 2019-08-06:
With 64 bit you only use vc4 graphics

agherzan, 2019-08-06:
Exactly. But if sets the provider for gles userland, it will just provide nothing because userland will not produce gles and Co. I've seen this exact error in the past when I was using userland providers by mistake.

thesillywhat, 2019-08-07:
I am not using vc4graphics. In my local.conf. I am removing vc4graphics from my machine features.
I have attached my local.conf file here
[local_conf.txt](https://github.com/agherzan/meta-raspberrypi/files/3477201/local_conf.txt)

thesillywhat, 2019-08-07:
So I am using vc4graphics now and I have restarted the process will keep this thread updated with the result

kraj, 2019-08-07:
`MACHINE_FEATURES_remove="vc4graphics" ` that's your problem, remove it from your local.conf
`

thesillywhat, 2019-08-07:
That did work.

closed 2019-08-07 by thesillywhat