agherzan/meta-raspberrypi #467: x11 in raspberry pi 4  chromium can not display on weida hi-tech cool touch screen in Yotco project

gitgonewithwind, 2019-08-07:
**Description**

**Steps to reproduce the issue:**
1.
my local.config: some codes are my application. the local.conf almost are same as the file in raspberry pi3

MACHINE = "raspberrypi4"
DISTRO ?= "nodistro"
PACKAGE_CLASSES ?= "package_rpm"
DL_DIR = "/opt/Downloads"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
# display language support
CORE_IMAGE_EXTRA_INSTALL += " \
	source-han-sans-cn-fonts \
	source-han-sans-kr-fonts \
	ttf-arphic-uming \
	"

# rpi
ENABLE_I2C = "1"
KERNEL_MODULE_AUTOLOAD += "i2c-dev"
ENABLE_SPI_BUS = "1"
ENABLE_UART = "1"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
# touchscreen
WAVESHARE_1024X600_C_2_1 = "1"
PACKAGECONFIG_pn-xserver-nodm-init = "nocursor"

# fti
CORE_IMAGE_EXTRA_INSTALL += " \
	bg-logo \
	fti-locator \
	fti-touchscreen \
	psplash-fortress \
	"
INHERIT += "extrausers"
ROOT_PASSWORD = "MarShall51"
EXTRA_USERS_PARAMS = "usermod -P ${ROOT_PASSWORD} root;"
EXTRA_USERS_PARAMS += "useradd -P ${ROOT_PASSWORD} myuser;"
EXTRA_USERS_PARAMS += "usermod -G video myuser;"
#KERNEL_DEVICETREE_append = " overlays/vc4-fkms-v3d.dtbo"
# rt
#PREFERRED_PROVIDER_virtual/kernel = "linux-raspberrypi-rt"
#KERNEL_DEVICETREE_remove = "bcm2708-rpi-0-w.dtb"

# base
IMAGE_FEATURES += "ssh-server-openssh package-management hwcodecs splash"
DISTRO_FEATURES_append = " pam"
CORE_IMAGE_EXTRA_INSTALL += " \
	packagegroup-core-full-cmdline \
	kernel-modules util-linux \
	bash bc cpio gawk tree rsync ldd vim nano \
	diffutils coreutils findutils iputils \
	trace-cmd i2c-tools e2fsprogs-resize2fs \
	"

# gui/gl
IMAGE_FEATURES += "x11"
LICENSE_FLAGS_WHITELIST = "commercial"
MACHINE_FEATURES_append = " vc4graphics"
DISTRO_FEATURES_append = " opengl x11"
DISTRO_FEATURES_remove = "wayland"
PACKAGECONFIG_append_pn-mesa-gl = " gbm"
RDEPENDS_packagegroup-core-x11-utils_remove_pn-packagegroup-core-x11 = "xinput-calibrator"
CORE_IMAGE_EXTRA_INSTALL += " \
	packagegroup-core-x11 \
	xf86-video-modesetting \
	xserver-xorg-extension-glx \
	xterm xsetroot openbox \
	fontconfig fontconfig-utils \
	chromium-x11 \
	"

# languages
CORE_IMAGE_EXTRA_INSTALL += " \
	packagegroup-core-lsb-perl \
	packagegroup-core-lsb-python \
	nodejs nodejs-npm \
	"

WARN_QA_append = " version-going-backwards"
ERROR_QA_remove = "version-going-backwards"
BB_DANGLINGAPPENDS_WARNONLY = "yes"

INHERIT += "rm_work"
INHERIT += "buildhistory"
INHERIT += "image-buildinfo"
#INHERIT += "distrodata"
BUILDHISTORY_COMMIT = "1"
BUILDHISTORY_PUSH_REPO = "origin"

PACKAGE_CLASSES ?= "package_ipk"
EXTRA_IMAGE_FEATURES ?= "tools-debug tools-profile"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

PACKAGECONFIG_append_pn-qemu-system-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
CONF_VERSION = "1"

2. run : bitbake  -c clean core-image-full-cmdline

3. .config I change to preempty. if I keep the old .config. the problem are same.

**Describe the results you received:**

my app(nodejs ) is running well.
chromium is running well.
but chromium not display on touchscreen
in dmesg,  the kernel find the touchscreen

**Describe the results you expected:**
chromium  need display in touchscreen

**Additional information you deem important (e.g. issue happens only occasionally):**
I change my config.txt to same as the raspberry pi 3. but issue is same.
config.txt

#dtparam=pwr_led_gpio=35
# Enable SPI bus
dtparam=spi=on
# Enable I2C
dtparam=i2c1=on
dtparam=i2c_arm=on
# Enable UART
enable_uart=1
# Enable VC4 Graphics
dtoverlay=vc4-kms-v3d,cma-256
# Waveshare "C" 1024x600 7" Rev2.1 IPS capacitive touch screen
max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1

**Additional details (revisions used, host distro, etc.):**

gitgonewithwind, 2019-08-07:
do I need change some default config for x11 xorg in /etc/X11 ?

gitgonewithwind, 2019-08-08:
I changed  VC4DTBO ?= "vc4-fkms-v3d" to VC4DTBO ?= "vc4-kms-v3d" in conf/machine/raspberrypi4.conf.  error is still there. it's similar to #453    

[  297.602249] i2c-bcm2835 fe805000.i2c: i2c transfer timed out                                                                                                                                             
[  298.642262] i2c-bcm2835 fe805000.i2c: i2c transfer timed out                                                                                                                                             
[  299.682255] i2c-bcm2835 fe805000.i2c: i2c transfer timed out                                                                                                                                             
[  300.722254] i2c-bcm2835 fe805000.i2c: i2c transfer timed out                                                                                                                                             
[  301.762254] i2c-bcm2835 fe805000.i2c: i2c transfer timed out                                                                                                                                             
[  302.802252] i2c-bcm2835 fe805000.i2c: i2c transfer timed out                                                                                                                                             
Modifiers specified, but DRI is too old                                                                                                                                                                     
[  313.442276] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:116:crtc-2] flip_done timed out                                                                                                  
                                                                                                                                                                                                            
[  323.682272] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CONNECTOR:44:Composite-1] flip_done timed out

agherzan, 2019-08-08:
Don't use `vc4-kms-v3d` on rpi4. It will not work (at least for now).

gitgonewithwind, 2019-08-08:
> Don't use `vc4-kms-v3d` on rpi4. It will not work (at least for now).

thanks.
but anybody found solution?

agherzan, 2019-08-08:
Not yet. I wasn't able to run even `kmscube`.

gitgonewithwind, 2019-08-08:
> Not yet. I wasn't able to run even `kmscube`.

Thank your reply so soon.
one more question: this upstream is the official  source code for raspberry pi, is it? I think so, but now     I am confused. I mean if there is no solution, It's hard for me to find it.
I look forward to your reply very soon.
Thank you.

agherzan, 2019-08-08:
In not official as in maintained by the foundation but it is the main upstream one (if not the only).

gitgonewithwind, 2019-08-08:
Thank you very much for your reply so soon.
so there is no solution in the world up to now.
I keep going to find it. anyway, it's hard to me. I almost have spent more than two weeks.
do you have any idea about what cause the issue as you guess? thanks.

gitgonewithwind, 2019-08-08:
I used the updated master branch to build  MACHINE = "raspberrypi3" .
 the image does not work in raspberry 3.  no display same as raspberry 4.

but linux-raspberrypi-rt_4.4  works well . I used before.

kraj, 2019-08-08:
> I used the updated master branch to build MACHINE = "raspberrypi3" .
> the image does not work in raspberry 3. no display same as raspberry 4.
> 
> but linux-raspberrypi-rt_4.4 works well . I used before.

For now 
rpi4 class
`DTBO="vc4-fkms-v3d"`
rpi3 class
`DTBO="vc4-kms-v3d"`

if you do not override thes are defaults too.

gitgonewithwind, 2019-08-08:
/etc/X11/Xsession.d# cat 01logo.shbg                                                                                                                                                           
#!/bin/sh                                                                                                                                                                                                   
/usr/bin/xset s off                                                                                                                                                                                         
/usr/bin/xset dpms 0 0 0                                                                                                                                                                                    
/usr/bin/xset -dpms                                                                                                                                                                                         
/usr/bin/xli -onroot -center -zoom 100 /home/root/logo.png    

-----------------------
it can show my copmay's logo at start up, but  slow!
===============
raspberrypi4:/etc/X11/Xsession.d# cat 01logo.shbg                                                                                                                                                           
#!/bin/sh                                                                                                                                                                                                   
/usr/bin/xset s off                                                                                                                                                                                         
/usr/bin/xset dpms 0 0 0                                                                                                                                                                                    
/usr/bin/xset -dpms                                                                                                                                                                                         
/usr/bin/xli -onroot -center -zoom 100 /home/root/logo.png                                                                                                                                                  
raspberrypi4:/etc/X11/Xsession.d# kmscube                                                                                                                                                                   
drmModeGetResources failed: Invalid argument                                                                                                                                                                
failed to initialize legacy DRM

gitgonewithwind, 2019-08-08:
my init startup script in /etc/init.d

cat fti-touchscreen.sh                                                                                                                                                           
#!/bin/sh                                                                                                                                                                                                   
                                                                                                                                                                                                            
export DISPLAY=:0                                                                                                                                                                                           
                                                                                                                                                                                                            
xset s off                                                                                                                                                                                                  
xset dpms 0 0 0                                                                                                                                                                                             
xset -dpms                                                                                                                                                                                                  
                                                                                                                                                                                                            
pushd /home/myuser                                                                                                                                                                                          
sudo node /home/myuser/node/server.js &                                                                                                                                                                     
popd                                                                                                                                                                                                        
                                                                                                                                                                                                            
sleep 7                                                                                                                                                                                                     
sudo -u myuser chromium --gpu-no-context-lost --enable-gpu-rasterization --start-fullscreen --kiosk --no-first-run --incognito http://localhost:3300/rpi.html > /dev/null 2>&1 &

agherzan, 2019-08-15:
@Sakaki I know you run X on rpi4-64 in gentoo. Have you ever seen `Draw call returned Invalid argument.  Expect corruption.` when starting X?

agherzan, 2019-08-15:
@sakaki- I know you run X on rpi4-64 in gentoo. Have you ever seen `Draw call returned Invalid argument.  Expect corruption.` when starting X?

agherzan, 2019-08-16:
@gitgonewithwind In order to make you happy I tested X as well. If you build master `meta-rasperrypi` + master `poky` you will be able to run X and accelerated demos (like `glxgears`). **There is a bug in warrior** where this doesn't work (as of yet). 

Here is a demo `local.conf` with which I built a `core-image-basic`:

```
MACHINE ?= "raspberrypi4-64"
DL_DIR = "/work/yocto/shared-downloads"
SSTATE_DIR = "/work/yocto/shared-sstate"
GPU_MEM = "256"
ENABLE_UART = "1"
RPI_USE_U_BOOT = "1"
LICENSE_FLAGS_WHITELIST = "commercial"
IMAGE_FSTYPES = "rpi-sdimg.gz"
IMAGE_INSTALL_append = " packagegroup-core-x11-base mesa-demos kmscube cinematicexperience"
DISTRO_FEATURES_append = " opengl"
DISTRO_FEATURES_append = " x11"
DISTRO_FEATURES_remove = " wayland"
DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"

# Everything from here on is default from poky

DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_rpm"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
```

Here is a sample log of running `glxgears`:

```
root@raspberrypi4-64:~# DISPLAY=:0 glxgears
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
309 frames in 5.0 seconds = 61.738 FPS
300 frames in 5.0 seconds = 59.994 FPS
[...]
```

I'm also closing this issue now. X11 is confirmed to work and is accelerated.

sakaki-, 2019-08-16:
> @sakaki- I know you run X on rpi4-64 in gentoo. Have you ever seen Draw call returned Invalid argument. Expect corruption. when starting X?

Not that I recall.

gitgonewithwind, 2019-08-16:
@agherzan @sakaki- 
thank you so much. I am going to test.

gitgonewithwind, 2019-08-16:
@agherzan 

root@raspberrypi4:~# DISPLAY=:0 glxgears                                        
Error: couldn't open display :0                                                 
root@raspberrypi4:~# kmscube                                                    
drmModeGetResources failed: Invalid argument                                    
failed to initialize legacy DRM                                                 
root@raspberrypi4:~# kmscube -D /dev/dri/card0                                  
drmModeGetResources failed: Invalid argument                                    
failed to initialize legacy DRM                                                 
root@raspberrypi4:~# kmscube -D /dev/dri/card1                                  
Using display 0xe0bb70 with EGL version 1.4                                     
===================================                                             
EGL information:                                                                
  version: "1.4"                                                                
  vendor: "Mesa Project"                                                        
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_dev"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_"
===================================                                             
OpenGL ES 2.x information:                                                      
  version: "OpenGL ES 3.0 Mesa 19.1.1"                                          
  shading language version: "OpenGL ES GLSL ES 3.00"                            
  vendor: "Broadcom"                                                            
  renderer: "V3D 4.2"                                                           
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_form"

------
I use you local.conf, but 32bit(my old app  code look like 32bit, not me to write it)   

 glxgears  not display

kmscube -D /dev/dri/card1   works well!

I am going to try 64bit.

agherzan, 2019-08-16:
@gitgonewithwind Have you started X first?

gitgonewithwind, 2019-08-16:
@agherzan 
sorry ,how to start x?  I do not know before

agherzan, 2019-08-16:
There are many ways - please search online. But for the sake of this test just run `X &` and afterwards the `glxgears` app.

gitgonewithwind, 2019-08-16:
@agherzan 
Thank you so much. works well!
I am going to add my company apps.
Thanks again. have a good weekend!

gitgonewithwind, 2019-08-19:
@agherzan 
sorry, chromium still can not work!   
DISPLAY=:0 glxgears  works well,  my local.conf,core-image-full-cmdline, 32bit

my nodjs code works well. I can access http://localhost:3300/rpi.html from my host machines. so the error is not from my the other code (nodejs webserver)
but I run chromium,Segmentation fault, no display on touchscreen.
I am trying to find the reasons.

******
DISPLAY=:0 sudo -u myuser chromium --gpu-no-context-lost --enable-gpu-rasterization --start-fullscreen --kiosk --no-first-run --incognito http://localhost:3300/rpi.html > /de 
[2] 1033                                                                                                                                                                                                  
raspberrypi4:/home/myuser#                                                                                                                                                                                
raspberrypi4:/home/myuser#                                                                                                                                                                                
[2]+  Segmentation fault      DISPLAY=:0 sudo -u myuser chromium --gpu-no-context-lost --enable-gpu-rasterization --start-fullscreen --kiosk --no-first-run --incognito http://localhost:3300/rpi.html > 1
raspberrypi4:/home/myuser#

gitgonewithwind, 2019-08-20:
@agherzan 
I got it. I mean I do not want  to close this issue. if closed, maybe nobody care it. thank you so much.

agherzan, 2019-08-20:
That makes sense. I'll reopen it.

xuhcc, 2019-08-20:
@gitgonewithwind 

Try to run chromium with `--disable-gpu` flag. I had similar issue with QtWebEngine on RPi and disabling GPU acceleration solved it.

gitgonewithwind, 2019-08-20:
> @gitgonewithwind
> 
> Try to run chromium with `--disable-gpu` flag. I had similar issue with QtWebEngine on RPi and disabling GPU acceleration solved it.

@xuhcc 

not work with --disable-gpu. thanks

I simply run : sudo -u myuser chromium     in my old source code in pi3, works well. 
but in current pi4 still  it's  Segmentation fault. I am going to add components  one by one in local.conf to try.
 
raspberrypi4:/home/myuser# sudo -u myuser chromium                                                                                                                                                        
Segmentation fault

gitgonewithwind, 2019-08-20:
strace sudo -u myuser chromium

output:
...
wait4(8848, [{WIFSIGNALED(s) && WTERMSIG(s) == SIGSEGV}], WNOHANG|WSTOPPED, NULL) = 8848                                                                                                                  
rt_sigaction(SIGHUP, {sa_handler=0x1ee69, sa_mask=~[KILL STOP RTMIN RT_1], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                         
rt_sigaction(SIGINT, {sa_handler=0x1ee69, sa_mask=~[KILL STOP RTMIN RT_1], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                         
rt_sigaction(SIGQUIT, {sa_handler=0x1ee69, sa_mask=~[KILL STOP RTMIN RT_1], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                        
rt_sigaction(SIGUSR1, {sa_handler=0x1ee69, sa_mask=~[KILL STOP RTMIN RT_1], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                        
rt_sigaction(SIGUSR2, {sa_handler=0x1ee69, sa_mask=~[KILL STOP RTMIN RT_1], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                        
rt_sigaction(SIGPIPE, {sa_handler=SIG_IGN, sa_mask=~[KILL STOP RTMIN RT_1], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                        
rt_sigaction(SIGALRM, {sa_handler=0x1ee69, sa_mask=~[KILL STOP RTMIN RT_1], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                        
rt_sigaction(SIGTERM, {sa_handler=0x1ee69, sa_mask=~[KILL STOP RTMIN RT_1], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                        
rt_sigaction(SIGCHLD, {sa_handler=0x1ee69, sa_mask=~[KILL STOP RTMIN RT_1], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                        
rt_sigaction(SIGCONT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                                                        
rt_sigaction(SIGTSTP, {sa_handler=0x1ee69, sa_mask=~[KILL STOP RTMIN RT_1], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6dcad51}, NULL, 8) = 0                                                        
close(4)                                = 0                                                                                                                                                               
close(5)                                = 0                                                                                                                                                               
geteuid32()                             = 0                                                                                                                                                               
getegid32()                             = 1000                                                                                                                                                            
setresuid32(-1, 1000, 0)                = 0                                                                                                                                                               
keyctl(KEYCTL_REVOKE, 588763748)        = 0                                                                                                                                                               
setreuid32(-1, 0)                       = 0                                                                                                                                                               
getuid32()                              = 0                                                                                                                                                               
openat(AT_FDCWD, "/etc/login.defs", O_RDONLY|O_LARGEFILE) = 3                                                                                                                                             
fstat64(3, {st_mode=S_IFREG|0644, st_size=11046, ...}) = 0                                                                                                                                                
read(3, "#\n# /etc/login.defs - Configurat"..., 4096) = 4096                                                                                                                                              
read(3, "ommented out and\n# set TTYPERM t"..., 4096) = 4096                                                                                                                                              
read(3, "56, SHA256-based algorithm will "..., 4096) = 2854                                                                                                                                               
close(3)                                = 0                                                                                                                                                               
[... cut ...]
kill(8839, SIGSEGV)                     = 0                                                                                                                                                               
--- SIGSEGV {si_signo=SIGSEGV, si_code=SI_USER, si_pid=8839, si_uid=0} ---                                                                                                                                
+++ killed by SIGSEGV +++                                                                                                                                                                                 
Segmentation fault

gitgonewithwind, 2019-08-20:
my chromium version:  chromium-x11-76.0.3809.100
I open a issue at meta-browser too: https://github.com/OSSystems/meta-browser/issues/299
maybe the issue is from chromium itself. 
I am going to add the old version to test.

gitgonewithwind, 2019-08-20:
I use following local.conf for pi4 64bit  to generate the image.  it works well to run sudo -u myuser chromium.
but 32bit can not work!
I am going to add more my apps and test more

local.conf

 1 MACHINE ?= "raspberrypi4-64"
  2 DL_DIR = "/opt/Downloads"
  3 #SSTATE_DIR = "/work/yocto/shared-sstate"
  4 GPU_MEM = "256"
  5 ENABLE_UART = "1"
  6 RPI_USE_U_BOOT = "1"
  7 LICENSE_FLAGS_WHITELIST = "commercial"
  8 #IMAGE_FSTYPES = "rpi-sdimg.gz"
  9 IMAGE_INSTALL_append = " packagegroup-core-x11-base mesa-demos kmscube cinem    aticexperience chromium-x11 strace"
 10 DISTRO_FEATURES_append = " opengl"
 11 DISTRO_FEATURES_append = " x11"
 12 DISTRO_FEATURES_remove = " wayland"
 13 DISTRO_FEATURES_append = " systemd"
 14 VIRTUAL-RUNTIME_init_manager = "systemd"
 15 DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
 16 
 17 INHERIT += "extrausers"
 18 ROOT_PASSWORD = "MarShall51"
 19 EXTRA_USERS_PARAMS = "usermod -P ${ROOT_PASSWORD} root;"
 20 EXTRA_USERS_PARAMS += "useradd -P ${ROOT_PASSWORD} myuser;"
 21 EXTRA_USERS_PARAMS += "usermod -G video myuser;"
 22 
 23 # Everything from here on is default from poky
 24 
 25 DISTRO ?= "poky"
 26 PACKAGE_CLASSES ?= "package_rpm"
 27 EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
 28 USER_CLASSES ?= "buildstats image-mklibs image-prelink"
 29 PATCHRESOLVE = "noop"
 30 BB_DISKMON_DIRS = "\
 31     STOPTASKS,${TMPDIR},1G,100K \
 32     STOPTASKS,${DL_DIR},1G,100K \
 33     STOPTASKS,${SSTATE_DIR},1G,100K \
 34     STOPTASKS,/tmp,100M,100K \
 35     ABORT,${TMPDIR},100M,1K \
 36     ABORT,${DL_DIR},100M,1K \
 37     ABORT,${SSTATE_DIR},100M,1K \
 38     ABORT,/tmp,10M,1K"
 39 PACKAGECONFIG_append_pn-qemu-native = " sdl"
 40 PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"

gitgonewithwind, 2019-08-22:
pi4 64bit core-image-base  works well.
but 32bit not work!

agherzan, 2021-10-30:
Close for no activity. We can reopen if we refresh on the current master.

closed 2021-10-30 by agherzan