agherzan/meta-raspberrypi #469: recipes-devtools/wiringPi does not work, as it needs to be updated to 2.52 for RPi4

mozcelikors, 2019-08-10:
As suggested in http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/, Raspberry Pi 4 is supported in 2.52 version of wiringPi (Raspberry Pi 4 is not supported in other versions). I think this should be updated on meta-raspberrypi as well. Furthermore, gpio tool gives segmentation fault, possibly there needs to be upgrades there as well.

agherzan, 2019-08-15:
@mozcelikors Good shout. Are you able to test and put up a patch for this?

agherzan, 2019-08-15:
Also, see https://github.com/agherzan/meta-raspberrypi/issues/473

mozcelikors, 2019-08-15:
@agherzan Hi, a patch to the existent recipe, or modifying the existing recipe to 2.54? Since git repo is also going to be deprecated, I'm not so sure how to write recipe for .deb package, maybe if there is a tar somewhere we can use it I think. If you can give me a bit more detail what might be the best for approach for meta-raspberrypi, I can see what I can do.

agherzan, 2019-08-15:
That's why I pointed you to the deprecation issue too. I think we should just blacklist it for now. Hoping that someone will pick up the maintenance.

theonlyzby, 2019-09-14:
I found this arm64 fork ( https://github.com/Vicinzu/wiringPi-arm64 ), which can be installed into aarch64 container ... but it has not update up 2.52 to support Raspberry Pi 4 :-(

ykla, 2019-10-29:
Anyone have sources about writing 2.52?

pbrkr, 2019-11-14:
WiringPi is dead after e050601, closing this.

closed 2019-11-14 by pbrkr

wikmas, 2020-08-31:
https://github.com/WiringPi/WiringPi

wikmas, 2020-08-31:
WiringPi 2.60 is here https://github.com/WiringPi/WiringPi
It works on Raspberry Pi 4.

agherzan, 2020-09-02:
What is the state of development with this project?

ykla, 2020-09-14:
It seems that has stopped develop. But It works on raspberry pi 4 with others help. @agherzan

agherzan, 2020-09-15:
Is the rpi4 support included in a specific release?