agherzan/meta-raspberrypi #473: WiringPi has been deprecated

proffalken, 2019-08-13:
**Description**

WiringPi has been deprecated by it's author: http://wiringpi.com/wiringpi-deprecated/

As reported in https://github.com/brocaar/lora-gateway-os/issues/18, git.drogon.com no longer presents a git server, instead it just presents an error.

This means that when you come to build an image that relies on Wiring Pi to be installed (such as the LoRa-Gateway-OS repo) then it fails at https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-devtools/wiringpi/wiringpi_git.bb#L14 with the following error:

```
ERROR: wiringpi-git-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/lora-gateway-os/layers/openembedded/openembedded-core/scripts:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot/usr/bin/crossscripts:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot-native/usr/sbin:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot-native/usr/bin:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot-native/sbin:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot-native/bin:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/lora-gateway-os/bitbake/bin:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/hosttools"; export HOME="/media/yocto/jenkins"; LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror git://git.drogon.net/wiringPi /media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/downloads/git2/git.drogon.net.wiringPi --progress failed with exit code 128, output:

Cloning into bare repository '/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/downloads/git2/git.drogon.net.wiringPi'...

fatal: unable to connect to git.drogon.net:

git.drogon.net[0: 188.246.205.22]: errno=Connection refused

git.drogon.net[1: 2a03:9800:10:7b::2]: errno=Network is unreachable

ERROR: wiringpi-git-r0 do_fetch: Fetcher failure for URL: 'git://git.drogon.net/wiringPi'. Unable to fetch URL from any source.

ERROR: wiringpi-git-r0 do_fetch: Function failed: base_do_fetch

ERROR: Logfile of failure stored in: /media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/temp/log.do_fetch.22117

NOTE: recipe wiringpi-git-r0: task do_fetch: Failed
```

**Steps to reproduce the issue:**
1. Try to build an image that uses the WiringPi library

**Describe the results you received:**

See error message in description

**Describe the results you expected:**

The image to build

**Additional information you deem important (e.g. issue happens only occasionally):**

There are no other mirrors out there that are newer than 12 months old (https://github.com/WiringPi/WiringPi was last updated over a year ago for 2.46, the latest release is 2.52), and the author recommends that we find an alternative GPIO library to use instead.

**Additional details (revisions used, host distro, etc.):**

agherzan, 2019-08-14:
That is news to me @proffalken . I think we should go ahead and remove the recipe.

brocaar, 2019-08-14:
@agherzan I'm not sure if (directly) removing the recipe would be the right approach. I understand that the author is not going to continue the development (see http://wiringpi.com/wiringpi-deprecated/), but it doesn't mean that we are not able to use the already released versions.

> I will make a final release of wiringPi available soon – with the sources, but that’s that. No more public releases. I’ll still be maintaining it for my own uses and clients, but for everyone else, please look at for alternative GPIO library for on-going projects.

https://git.drogon.net/ currently returns:

> git.drogon.net is **temporarily unavailable**. Please look for alternatives for wiringPi, etc. -Gordon

It does say it is temporarily unavailable, not permanently. Hopefully it will come back.

This topic might also be interesting to keep an eye on. As this package is also included in the Raspbian distribution, this deprecation is going to affect more people: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=248500. Maybe somebody else will step up as a new maintainer?

agherzan, 2019-08-14:
I read it as a permanent removal.

proffalken, 2019-08-14:
@agherzan yeah, have to agree with you here, I think the removal of the git repo is potentially temporary, but I get the feeling from that blog post that halting future development is more permanent than we might hope. :(

proffalken, 2019-08-14:
FWIW, I do think that removing it immediately might be a bit premature, perhaps keep an eye on the situation and see how it develops.

@brocaar the only thing I've not been able to figure out so far is whether it will affect the ability of lora-gateway-os to interact with some of the boards.

I'm pretty sure things like the IC880A run via SPI, but I can't tell whether your code is reliant on the underlying WiringPi libraries to work with these boards.

agherzan, 2019-08-14:
Let's make it at least RPI123 only compatible.

kraj, 2019-08-14:
blacklist it. Add some like below to the recipe
```
PNBLACKLIST[wiringpi] = "Since upstream has abandoned it https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=248500"
```

pbrkr, 2019-11-14:
WiringPi is dead after e050601, closing this.

closed 2019-11-14 by pbrkr