agherzan/meta-raspberrypi #474: rpm-basic-image build with 'rt' kernel fails

Supriyaj29, 2019-08-14:
**Description**

I am trying to build custom image for raspberrypi-cm3. I have setup yocto environment using worrier branch. 

I am overriding following variables in local.conf:
MACHINE ?= "raspberrypi-cm3"
RPI_USE_U_BOOT = "1"
IMAGE_INSTALL_append = " kernel-image kernel-devicetree"
IMAGE_FSTYPES_remove += " rpi-sdimg"
IMAGE_FSTYPES = "ext4"
KERNEL_BOOTCMD = "bootz"
KERNEL_IMGTYPE = "zImage"
PREFERRED_PROVIDER_virtual/kernel ?= "linux-raspberrypi-rt"

Bitbake rpi-basic-image fails with following error:

NOTE: make -j 2 HOSTCC=gcc  -isystem/home/ubuntu/yocto/cros-build/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/linux-raspberrypi-rt/1_4.14.81+gitAUTOINC+acf578d07d-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/ubuntu/yocto/cros-build/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/linux-raspberrypi-rt/1_4.14.81+gitAUTOINC+acf578d07d-r0/recipe-sysroot-native/usr/lib                         -L/home/ubuntu/yocto/cros-build/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/linux-raspberrypi-rt/1_4.14.81+gitAUTOINC+acf578d07d-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/ubuntu/yocto/cros-build/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/linux-raspberrypi-rt/1_4.14.81+gitAUTOINC+acf578d07d-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/ubuntu/yocto/cros-build/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/linux-raspberrypi-rt/1_4.14.81+gitAUTOINC+acf578d07d-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/ubuntu/yocto/cros-build/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/linux-raspberrypi-rt/1_4.14.81+gitAUTOINC+acf578d07d-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/ubuntu/yocto/cros-build/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/linux-raspberrypi-rt/1_4.14.81+gitAUTOINC+acf578d07d-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/ubuntu/yocto/cros-build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 HOSTCPP=gcc  -E bcm2708-rpi-zero-w.dtb
|   CC      scripts/mod/empty.o
|   CC      scripts/mod/devicetable-offsets.s
|   MKELF   scripts/mod/elfconfig.h
|   HOSTCC  scripts/mod/modpost.o
|   CHK     scripts/mod/devicetable-offsets.h
|   HOSTCC  scripts/mod/sumversion.o
|   HOSTCC  scripts/mod/file2alias.o
|   HOSTLD  scripts/mod/modpost
| make[3]: *** No rule to make target 'arch/arm/boot/dts/bcm2708-rpi-zero-w.dtb'.  Stop.
| arch/arm/Makefile:343: recipe for target 'bcm2708-rpi-zero-w.dtb' failed

**Additional information you deem important (e.g. issue happens only occasionally):**
None
**Additional details (revisions used, host distro, etc.):**
poky - warrior branch

This is the first time I am using yocto build system and haven't yet understood all the steps. Would really appreciate, if someone could point out if there is anything wrong with this setup.

agherzan, 2019-08-14:
This should be fixed by https://github.com/agherzan/meta-raspberrypi/pull/457

fgiancane8, 2019-08-16:
@agherzan , 

actually it is not fixed, I just updated the standard kernel and not the -rt one. It happens that the -rt version is maintained as a separate branch with different point release, so we either have to back port again the patch on 4.14-rt series or include in SRC_URI the patch I posted before.

agherzan, 2019-08-17:
Right. Missed the fact that this was on a rt branch. The easiest way would be to work with upstream on this. But I'm not sure if they maintain the 4.14 branch anymore. What about the 4.19 one? We should update if we haven't don't already.

fgiancane8, 2019-08-18:
There is a 4.19.y-rt branch on raspberrypi/linux repository, and it seems to have RT_PATCH included and currently maintained.

However, in my opinion, I would not remove the 4.14 series. I am experiencing issues with 4.19 series especially with the MIPI DSI official display which is working correctly on 4.14 and not on 4.19 (no way of making it work so far, tested like 3 weeks ago). 

My suggestion would be to back port the patch on 4.14.y-rt also and then provide 4.19.y-rt for those not relying on external displays.

agherzan, 2019-08-18:
We have now a 4.19 rt recipe: https://github.com/agherzan/meta-raspberrypi/pull/478 . And about the 4.14 I agree to backport the dt rename fix.

fgiancane8, 2019-08-19:
PR sent to raspberrypi/linux also for 4.14.y-rt series. In the meantime, please find attached the patch here.

fgiancane8, 2019-08-19:
[0001-Backport-BCM270X_DT-Rename-Pi-Zero-W-DT-files.patch.zip](https://github.com/agherzan/meta-raspberrypi/files/3515828/0001-Backport-BCM270X_DT-Rename-Pi-Zero-W-DT-files.patch.zip)

PR #482 (2019-08-22): linux-raspberrypi-rt: update to latest commit for 4.14 series
Update linux kernel 4.14 series recipe for building with latest -rt
branch.
This fixes #474.
commit e52158c9 (2019-08-22): linux-raspberrypi-rt: update to latest commit for 4.14 series

Update linux kernel 4.14 series recipe for building with latest -rt
branch.
This fixes #474.
closed 2019-08-22 by agherzan