agherzan/meta-raspberrypi #479: Broken graphics on poky provided mesa version in warrior

agherzan, 2019-08-18:
**Description**

When building a image for RaspberryPi (tested on 4) on warrior, HDMI shows no output.

**Steps to reproduce the issue:**
1. checkout warrior meta-raspberrypi and poky
2. build a image that includes for example kmscube
3. run kmscube

**Describe the results you received:**
```
root@raspberrypi4:~# kmscube -D /dev/dri/card0
drmModeGetResources failed: Invalid argument
failed to initialize legacy DRM
root@raspberrypi4:~# kmscube -D /dev/dri/card1
Using display 0xc6f648 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_debug EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 2.0 Mesa 19.0.1"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d "
===================================
Draw call returned Invalid argument.  Expect corruption.
```

App works but there is no HDMI output.

**Additional information you deem important (e.g. issue happens only occasionally):**
Updating to master fixes the issue.
Also, locally updating mesa in poky warrior (to the version in master) fixes the issue.

sirus20x6, 2019-10-31:
do you get display of your console? I might have the same issue. I get display output while booting and then black screen instead of a shell

agherzan, 2019-11-01:
@sirus20x6 I can't remember to be honest but it sounds right.

sirus20x6, 2019-11-01:
@agherzan ok because I've been building zeus branch and I get display until the shell is supposed to show up. After that the screen is blank with backlight on. Other mini hdmi port is also black. I didn't see anything super obvious in system.journal, but there were a few red lines about 
`[drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
Oct 29 16:26:50 product kernel: vc4-drm soc:gpu: [drm:drm_fb_helper_fbdev_setup] *ERROR* Failed to set fbdev configuration
`
and then a few lines later
`brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Oct 29 16:26:56 product kernel[222]: [   17.680036] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Oct 29 16:26:56 product kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: May  2 2019 02:46:17 version 7.45.189 (r714228 CY) FWID 01-e1db26e2
Oct 29 16:26:56 product kernel[222]: [   17.694362] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: May  2 2019 02:46:17 version 7.45.189 (r714228 CY) FWID 01-e1db26e2
`

I tried a few things like adding dtoverlay=vc4-fkms-v3d,cma-265 to config.txt . a suggestion I saw floating around the web, but it didn't help the display at all.

[system.journal.txt](https://github.com/agherzan/meta-raspberrypi/files/3798126/system.journal.txt)

kraj, 2019-11-01:
remove cma-265

sirus20x6, 2019-11-01:
thats how my first few boots were. I added the ,cma-265 after finding a suggestion to try it online. either way it doesn't work

sirus20x6, 2019-11-04:
it appears that zeus and master use the same mesa so I'm confused as to what's going on here.

sirus20x6, 2019-11-18:
does anyone have any kind of graphics output working on pi 4 on zeus?

agherzan, 2021-10-30:
Closing as worrior is not a supported branch anymore.

closed 2021-10-30 by agherzan