agherzan/meta-raspberrypi #481: GStreamer-based QML Video does not work on EGLFS on Pi 4, but works on Pi 3

mozcelikors, 2019-08-19:
I am interested in developing/porting apps to Pi, and now switching to Pi 4. I am unable to run Qt's official example video player, I suspect there is still driver anomalities with meta-raspberrypi, regarding graphical infrastructure, when it comes to Pi 4.

**How to reproduce:**
_Build image with:_
```
PACKAGECONFIG_qtbase_append = " examples "
PACKAGECONFIG_qtmultimedia_append = " examples "
IMAGE_INSTALL_append = " qtbase-examples "
```
_From Pi,_
```
/usr/share/examples/multimedia/video/qmlvideo# ./qmlvideo -platform eglfs
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/var/volatile/tmp/runtime-root'
Setting framebuffer size is only available with DRM atomic API
```
_Checked configuration:_

meta-raspberrypi master HEAD @ b112816e95ec59447a82a3a0ee7842c547dda335
poky  master HEAD @ 5f0d31ce4653f7a76a5226fa379a285ceca19d63

From the opened app select an mp4 file, and hit "Video"; you won't be able to see any video displayed. Output is given above. With the same approach, one can display video on Pi 3.

agherzan, 2019-08-19:
What is your full `local.conf`? Also, is this 32 or 64bit?

mozcelikors, 2019-08-19:
This is 32 bit.  Here are my `local.conf` additions:
```

ENABLE_SPI_BUS="1"
ENABLE_I2C="1"
MACHINE="raspberrypi4"
PACKAGE_CLASSES = "package_rpm"
TOOLCHAIN_HOST_TASK_append = " nativesdk-cmake "
MACHINE_FEATURES_append = "vc4graphics"
DISTRO_FEATURES_append = " bluez5 bluetooth wifi polkit "
DISTRO_FEATURES_remove = "systemd"
IMAGE_FEATURES += "splash"
LICENSE_FLAGS_WHITELIST_append = " commercial "
```

agherzan, 2019-08-19:
What is your distro?

mozcelikors, 2019-08-19:
Hi, My distro is poky

gitgonewithwind, 2019-08-20:
@agherzan 
my chromium still does not work! please look at #467 thanks. I try to figure out.

agherzan, 2019-08-20:
@gitgonewithwind I kindly ask you to stop hijacking random threads. As you can see this is another issue. If you haven't received feedback yet, it is most probably because it's not something people are working on right now and there is not an obvious solution to your problem. So please be patient and in the meanwhile try to dig into the problem as much as you can.

agherzan, 2019-08-20:
@aurelian17 @kraj Might know more about Gst and QML. I haven't tried anything in that regards.

aurelian17, 2019-08-20:
I'll take a look over it. It will help me a lot having exactly the same configuration as yours. So please share local.conf and bblayers.conf.

mozcelikors, 2019-08-20:
@aurelian17 Hi, thanks. I will do best I can. Let me know if you might need anything else.

meta-raspberrypi master HEAD @ b112816
poky master HEAD @ 5f0d31ce4653f7a76a5226fa379a285ceca19d63

local.conf

```
DISTRO = "poky"
ENABLE_SPI_BUS="1"
ENABLE_I2C="1"
MACHINE="raspberrypi4"
PACKAGE_CLASSES = "package_rpm"
MACHINE_FEATURES_append = "vc4graphics"
DISTRO_FEATURES_append = " bluez5 bluetooth wifi polkit "
# Do not use systemd
DISTRO_FEATURES_remove = "systemd"
IMAGE_FEATURES += "splash"
LICENSE_FLAGS_WHITELIST_append = " commercial "
RPI_KERNEL_DEVICETREE_OVERLAYS_append = " overlays/ads7846.dtbo "
```

bblayers.conf

```
BBLAYERS ?= " \
  ${BSPDIR}/meta \
  ${BSPDIR}/meta-poky \
  ${BSPDIR}/meta-yocto-bsp \
  ${BSPDIR}/meta-openembedded/meta-oe \
  ${BSPDIR}/meta-openembedded/meta-multimedia \
  ${BSPDIR}/meta-openembedded/meta-networking \
  ${BSPDIR}/meta-openembedded/meta-python \
  ${BSPDIR}/meta-openembedded/meta-gnome \
  ${BSPDIR}/meta-openembedded/meta-xfce \
  ${BSPDIR}/meta-qt5 \
  ${BSPDIR}/meta-qt5-extra \
  ${BSPDIR}/meta-raspberrypi \
  "
```

Core Image components from recipes

```
IMAGE_INSTALL_append = " gstreamer1.0-plugins-good \
                         gstreamer1.0-plugins-bad \
                         gstreamer1.0-plugins-ugly "

IMAGE_INSTALL_append = "  packagegroup-qt5-toolchain-target  "
IMAGE_INSTALL_append = " qtdeclarative qtdeclarative-qmlplugins qtquickcontrols qtquickcontrols-qmlplugins qtmqtt "
IMAGE_INSTALL_append = " qtbase-examples qtmultimedia-examples qtdeclarative-examples "
```

mozcelikors, 2019-08-25:
@aurelian17 Any good news yet ?

xuhcc, 2019-08-25:
@mozcelikors Does standalone gstreamer work for you on RPi4? I'm trying to play video with `gst-play-1.0`, it is starting without errors but video is not showing.

peel70, 2019-08-25:
Hi @mozcelikors  @agherzan @aurelian17 I've the same issue as you but I've configured a qmultimedia_git.bbappend with 
PACKAGECONFIG_append = " gstreamer" 
because QMultimedia must be compiled with gstreamer flag.

in the plugin-bad of gstreamer I've also included PACKAGECONFIG_append = " faad faac kms"

In gstreamer I've also addedd gstreamer1.0-omx because, is the only that is able tu use the VC for render the video in GPU
But at the moment is not working accelerated, I've opened an issue on freedesktop about this.

I've tried to recompile gstreamer on a raspbian burster light but without success at the moment

See this https://gitlab.freedesktop.org/gstreamer/gst-omx/issues/31#note_208543
"OMX is the only way to hardware decode video on the Pi 3. The new Pi 4 has video4linux decoders but not the previous versions"

Try to configure GST_DEBUG_DUMP_DOT_DIR before gst-play or gst-launch to check the chain that gstreamer is using to render the video or use gst-launch and specify the chain I think that in this new release of the system/firmware you cannot use eglfsink but kmssink

By the way I was not able to give video accelerated with pi3 but I can share my experience to try to find a solution because I also need the video acceleration in Pi4

mozcelikors, 2019-08-26:
@xuhcc In my experience, I get no video as well, stuck at Prerolling...

gst-play-1.0 mymovie.mp4
Press 'k' to see a list of keyboard shortcuts.
Now playing /run/media/sdb1/mymovie.mp4
Prerolling...

Note that I need to run it under EGLFS. X11 doesn't cut it for me. So I also tried ! eglfsink and ! kmssink. Same behavior.

peel70, 2019-08-26:
@xuhcc nice to hear this, have you check the CPU load during the video play?

xuhcc, 2019-08-26:
@mozcelikors 

I had the same issue with preroll (also without X11). Today I managed to get it working by constructing the pipeline manually:

```
gst-launch-1.0 -v filesrc location=myvideo.mkv ! matroskademux ! h264parse ! avdec_h264 ! kmssink
```

1080p video is playing fine now, but 4K is super slow.

That's my `gstreamer1.0-plugins-bad_1.16.0.bbappend` (thanks @peel70):

```
PACKAGECONFIG_remove = "vulkan"
PACKAGECONFIG_append = " faad faac kms "
```

peel70, 2019-08-26:
@xuhcc as I suspected gstreamer is using sw decoding and not hw decoding.
RPI4 has h265 builtin decoding. 
With KODI I was able to play 4k H265 with low CPU load with RPI4.
To be sure about gstreamer chain, read this guide https://embeddedartistry.com/blog/2018/2/22/generating-gstreamer-pipeline-graphs and configure GST_DEBUG_DUMP_DOT_DIR so you are able to see a graphically gstreamer chain.

I think there's no gstreamer plugin that decode h265 in GPU with zero copy.
I was hoping in gstreamer-omx but no luck at the moment.

Consider that in RPI Buster Full,  launching chromium and go to chrome://gpu you'll see no hw video decode so the problem is another in my opinion, maybe is too soon to give all works fine?
Gstreamer bad had also libde265 dec, and you need to IMAGE_INSTALL_append = " libde265" but I'have not tested yet.

xuhcc, 2019-08-26:
@peel70 Yes, indeed the CPU load is high. When I play the same video with omxplayer, the CPU load is insignificant.

mozcelikors, 2019-08-29:
Neither from QML application, nor from entering gst-play-1.0 can I play a .mp4 video at the moment.
I've added to bad plugins:
PACKAGECONFIG_remove = "vulkan"
PACKAGECONFIG_append = " faad faac kms "

and to the image:
IMAGE_INSTALL_append = " gstreamer1.0-omx "
IMAGE_INSTALL_append = " libde265 "

I play with: 
GST_DEBUG_DUMP_DOT_DIR=/home/root gst-launch-1.0 -v filesrc location=mymovie.mp4  ! matroskademux ! h264parse ! avdec_h264 ! kmssink
WARNING: erroneous pipeline: no element "avdec_h264"
I think this is because this pipeline is for .mkv extension. 
I then tried:
GST_DEBUG_DUMP_DOT_DIR=/home/root gst-launch-1.0 -v filesrc location=mymovie.mp4  ! kmssink
and
GST_DEBUG_DUMP_DOT_DIR=/home/root gst-launch-1.0 -v filesrc location=mymovie.mp4

I get no video on the screen, no gstreamer debug on /home/root directory.

I'm wondering if we should also open a bug in Qt, if eglfssink is not supported anymore, new Qt version needs to use kmssink.

peel70, 2019-08-29:
@mozcelikors I've used master branch instead of warrior to build rpi4 image without X11 with Qt5.12.3 and a QML application works fine and accelerated read this
https://doc.qt.io/qt-5/embedded-linux.html
you have to launch application with -platform eglfs but it use eglfs_kms instead of eglfs_brcm (like before warrior and pi3)
Use setenv in main.cpp and enable debug rules
setenv("QT_LOGGING_RULES", "*=true")
If you have correctly build the image you'll see that eglfs_kms is used.
At the moment opengl is accelerated but not video at the moment

Verify with kmscube before launching Qt to check MESA and OpenGL driver works.

for Gstreamer use gst_inspect-1.0 | grep omx to check gstreamer-omx is installed
the gstreamer pipeline is not correct if you are playing mp4 you have not to use matroskademux
try 
GST_DEBUG_DUMP_DOT_DIR=/home/root gst-launch-1.0 -v filesrc location=mymovie.mp4 ! qtdemux ! h264parse ! omxh264dec ! videoconvert ! kmssink

I'm not behind rpi so I cannot test the pipeline, but I see video but at 3/4 fps

let me know

mozcelikors, 2019-08-29:
@peel70 A QML Application works fine, but does qmlvideo example work in your case? (check the issue topic)
QML application also works fine with me, I also run with -platform eglfs, Everything works fine, except that QML-based video doesn't show up. (Qt internally uses GStreamer)
I will check your pipeline

peel70, 2019-08-29:
@mozcelikors 
I've QML video work with Qmultimedia MediaPlayer and VideoOut but fps are poor
gstreamer is working too
have you tried with gst-play-1.0 too?

mozcelikors, 2019-08-29:
@peel70 Interesting. gst-play-1.0 also gives nothing. Maybe my master branches are a bit outdated and I need to pull the latest. I will give it a go with the very latest.

peel70, 2019-08-29:
@mozcelikors let me take some time to check on my build

mozcelikors, 2019-08-29:
@peel70 Sure, thanks for the help

peel70, 2019-08-30:
@mozcelikors sorry my mistake
I've tried 
gst-launch-1.0 videotestsrc ! kmssink
and its working

I've played mkv files and not mp4 because I wanted to try the h265 built-in decoder but it works at 3/4 fps QMediaplayer is able to play mkv but with the same performance

I was able ti play mp4 with this command but is a sw render (I've installed gstreamer-1.0-libav):
`gst-launch-1.0 -e -vvv filesrc location="/home/root/test.mp4" ! qtdemux ! h264parse ! avdec_h264 ! kmssink`

I was able to convert mp4 to mkv with this command
`gst-launch-1.0 -v filesrc location="/home/root/test.mp4" ! parsebin name="demux" ! h264parse ! queue ! matroskamux name="mux" ! filesink location="1.mkv" demux. ! aacparse ! queue ! mux.`

I was able to play accelerated in GPU with this pipeline:
`gst-launch-1.0 -e -vvv filesrc location=big_buck_bunny_720p_h264.mov ! qtdemux ! queue ! h264parse ! v4l2h264dec capture-io-mode=4 ! kmssink`

but
`GST_DEBUG="omx:2" gst-launch-1.0 -e -vvv filesrc location=bbb_sunflower_1080p_30fps_normal.mp4 ! qtdemux ! h264parse ! omxh264dec ! videoconvert ! kmssink`

I give this error:
¡0:00:00.332141903   634  0x1430d20 ERROR                    omx gstomx.c:1892:gst_omx_component_set_parameter:<omxh264dec-omxh264dec0> Set video_decode parameter at index 0x02000001: Bad parameter (0x80001005)
0:00:00.332392237   634  0x1430d20 ERROR                    omx gstomx.c:2072:gst_omx_port_update_port_definition:<omxh264dec-omxh264dec0> Updated video_decode port 130 definition: Bad parameter (0x80001005)`

mozcelikors, 2019-09-01:
gst-launch-1.0 videotestsrc ! kmssink works.
But playing through /usr/share/examples/multimedia/qmlvideo still does not work for me.
I've built everything fresh, with the latest git repos.

peel70, 2019-09-02:
Hi I’m out for 1 week,
Before qmlvideo try what work for me.
If you have compiled qmultimedia with gstreamer flag, what works on gst works on qmultimedia. From qt 5.11.4 if I’m not wrong it’s possibile to specify gst pipeline in qmediaplayer too in the source property

mozcelikors, 2019-09-02:
@peel70 Hi,
I compiled qmultimedia with gstreamer everytime, When I use everything master, it uses 5.13.0.
Could you please specify which HEADs you used and which branch for,
poky
meta-openembedded
meta-raspberrypi
meta-qt5
meta-qt5-extra
Thanks. I will try other commands when I get home.

peel70, 2019-09-02:
If I remember well 5.12.3
Can you try a little qtquick app and enable qtdebug and watch what happening when playing video?
Try mp4 mov and mkv

peel70, 2019-09-02:
setenv("QT_LOGGING_RULES", "*=true")

mozcelikors, 2019-09-02:
I enabled logs when I used qmlvideo example:
```

QT_LOGGING_RULES="qt.qpa.*=true" /usr/share/examples/multimedia/video/qmlvideo/qmlvideo -platform eglfs
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/var/volatile/tmp/runtime-root'
qt.qpa.egldeviceintegration: EGL device integration plugin keys: ("eglfs_emu", "eglfs_kms_egldevice", "eglfs_kms", "eglfs_x11")
qt.qpa.egldeviceintegration: EGL device integration plugin keys (sorted): ("eglfs_kms", "eglfs_emu", "eglfs_kms_egldevice", "eglfs_x11")
qt.qpa.egldeviceintegration: Trying to load device EGL integration "eglfs_kms"
qt.qpa.eglfs.kms: New DRM/KMS via GBM integration created
qt.qpa.egldeviceintegration: Using EGL device integration "eglfs_kms"
qt.qpa.eglfs.kms: platformInit: Opening DRM device
qt.qpa.input: udev device discovery for type QFlags<QDeviceDiscovery::QDeviceType>(Device_DRM|Device_VideoMask)
qt.qpa.input: Found matching devices ("/dev/dri/card1", "/dev/dri/card0")
qt.qpa.eglfs.kms: Found the following video devices: ("/dev/dri/card1", "/dev/dri/card0")
qt.qpa.eglfs.kms: Using "/dev/dri/card1"
qt.qpa.eglfs.kms: Using backend-provided DRM device /dev/dri/card1
qt.qpa.eglfs.kms: Creating GBM device for file descriptor 6 obtained from "/dev/dri/card1"
qt.qpa.eglfs.kms: Querying EGLDisplay
qt.qpa.eglfs.kms: Found 3 planes
qt.qpa.eglfs.kms: plane 0: id = 28 countFormats = 10 possibleCrtcs = 0x1 supported formats = XR24 AR24 RG16 RG24 BG24 YU16 YU12 YV12 NV12 NV21 
qt.qpa.eglfs.kms:   property 0: id = 7 name = 'type'
qt.qpa.eglfs.kms:   type is ENUM, value is 1, possible values are:
qt.qpa.eglfs.kms:     enum 0: Overlay - 0
qt.qpa.eglfs.kms:     enum 1: Primary - 1
qt.qpa.eglfs.kms:     enum 2: Cursor - 2
qt.qpa.eglfs.kms:   property 1: id = 16 name = 'FB_ID'
qt.qpa.eglfs.kms:   type is OBJECT
qt.qpa.eglfs.kms:   property 2: id = 17 name = 'IN_FENCE_FD'
qt.qpa.eglfs.kms:   type is SIGNED_RANGE, value is -1, possible values are:
qt.qpa.eglfs.kms:     -1
qt.qpa.eglfs.kms:     2147483647
qt.qpa.eglfs.kms:   property 3: id = 19 name = 'CRTC_ID'
qt.qpa.eglfs.kms:   type is OBJECT
qt.qpa.eglfs.kms:   property 4: id = 12 name = 'CRTC_X'
qt.qpa.eglfs.kms:   type is SIGNED_RANGE, value is 0, possible values are:
qt.qpa.eglfs.kms:     -2147483648
qt.qpa.eglfs.kms:     2147483647
qt.qpa.eglfs.kms:   property 5: id = 13 name = 'CRTC_Y'
qt.qpa.eglfs.kms:   type is SIGNED_RANGE, value is 0, possible values are:
qt.qpa.eglfs.kms:     -2147483648
qt.qpa.eglfs.kms:     2147483647
qt.qpa.eglfs.kms:   property 6: id = 14 name = 'CRTC_W'
qt.qpa.eglfs.kms:   type is RANGE, value is 800, possible values are:
qt.qpa.eglfs.kms:     0
qt.qpa.eglfs.kms:     2147483647
qt.qpa.eglfs.kms:   property 7: id = 15 name = 'CRTC_H'
qt.qpa.eglfs.kms:   type is RANGE, value is 480, possible values are:
qt.qpa.eglfs.kms:     0
qt.qpa.eglfs.kms:     2147483647
qt.qpa.eglfs.kms:   property 8: id = 8 name = 'SRC_X'
qt.qpa.eglfs.kms:   type is RANGE, value is 0, possible values are:
qt.qpa.eglfs.kms:     0
qt.qpa.eglfs.kms:     4294967295
qt.qpa.eglfs.kms:   property 9: id = 9 name = 'SRC_Y'
qt.qpa.eglfs.kms:   type is RANGE, value is 0, possible values are:
qt.qpa.eglfs.kms:     0
qt.qpa.eglfs.kms:     4294967295
qt.qpa.eglfs.kms:   property 10: id = 10 name = 'SRC_W'
qt.qpa.eglfs.kms:   type is RANGE, value is 52428800, possible values are:
qt.qpa.eglfs.kms:     0
qt.qpa.eglfs.kms:     4294967295
qt.qpa.eglfs.kms:   property 11: id = 11 name = 'SRC_H'
qt.qpa.eglfs.kms:   type is RANGE, value is 31457280, possible values are:
qt.qpa.eglfs.kms:     0
qt.qpa.eglfs.kms:     4294967295
qt.qpa.eglfs.kms:   property 12: id = 27 name = 'IN_FORMATS'
qt.qpa.eglfs.kms:   type is BLOB
qt.qpa.eglfs.kms:   property 13: id = 30 name = 'alpha'
qt.qpa.eglfs.kms:   type is RANGE, value is 65535, possible values are:
qt.qpa.eglfs.kms:     0
[... 291 lines cut ...]
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_DOWN_WAITING ? DEBOUNCE_EVENT_TIMEOUT ? DEBOUNCE_STATE_IS_DOWN
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_DOWN ? DEBOUNCE_EVENT_RELEASE ? DEBOUNCE_STATE_IS_UP_DETECTING_SPURIOUS
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_UP_DETECTING_SPURIOUS ? DEBOUNCE_EVENT_TIMEOUT_SHORT ? DEBOUNCE_STATE_IS_UP_WAITING
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_UP_WAITING ? DEBOUNCE_EVENT_PRESS ? DEBOUNCE_STATE_IS_DOWN_DELAYING
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_DOWN_DELAYING ? DEBOUNCE_EVENT_TIMEOUT ? DEBOUNCE_STATE_IS_DOWN
qt.qpa.input: libinput: client bug: timer event1 debounce: offset negative (-35ms)
qt.qpa.input: libinput: client bug: timer event1 debounce short: offset negative (-48ms)
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_DOWN ? DEBOUNCE_EVENT_RELEASE ? DEBOUNCE_STATE_IS_UP_DETECTING_SPURIOUS
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_UP_DETECTING_SPURIOUS ? DEBOUNCE_EVENT_TIMEOUT_SHORT ? DEBOUNCE_STATE_IS_UP_WAITING
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_UP_WAITING ? DEBOUNCE_EVENT_TIMEOUT ? DEBOUNCE_STATE_IS_UP
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_UP ? DEBOUNCE_EVENT_PRESS ? DEBOUNCE_STATE_IS_DOWN_WAITING
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_DOWN_WAITING ? DEBOUNCE_EVENT_TIMEOUT ? DEBOUNCE_STATE_IS_DOWN
[... cut ...]
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_UP_WAITING ? DEBOUNCE_EVENT_TIMEOUT ? DEBOUNCE_STATE_IS_UP
qt.qpa.input: libinput: client bug: timer event1 debounce: offset negative (-894ms)
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_UP ? DEBOUNCE_EVENT_PRESS ? DEBOUNCE_STATE_IS_DOWN_WAITING
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_DOWN_WAITING ? DEBOUNCE_EVENT_RELEASE ? DEBOUNCE_STATE_IS_UP_DELAYING
qt.qpa.input: libinput: event1  - debounce state: DEBOUNCE_STATE_IS_UP_DELAYING ? DEBOUNCE_EVENT_TIMEOUT ? DEBOUNCE_STATE_IS_UP
```

peel70, 2019-09-02:
I don’t see anything about multimedia
Could you try with
export QT_LOGGING_RULES=qt.multimedia.*=true

Have you try gstreamer with .mov like me and this example too?

mozcelikors, 2019-09-02:
@peel70 I've tried .mp4
```
root@raspberrypi4:~# QT_LOGGING_RULES=qt.multimedia.*=true /usr/share/examples/multimedia/video/qmlvideo/qmlvideo -platform eglfs
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/var/volatile/tmp/runtime-root'
Setting framebuffer size is only available with DRM atomic API
qt.multimedia.video: source is QDeclarativeAudio(0xe25850)

With .mov I also cannot see the video:
```
qt.multimedia.video: media object is QMediaPlayer(0xe041c0)
qt.multimedia.video: found videonode plugin "egl" 0xfc60f8
qt.multimedia.video: updatePaintNode: no frames yet
qt.multimedia.video: Video surface format: QVideoSurfaceFormat(Format_YUV420P, QSize(480, 272), viewport=QRect(0,0 480x272), pixelAspectRatio=QSize(1, 1), handleType=NoHandle, yCbCrColorSpace=YCbCr_Undefined)
    handleType = QVariant(QAbstractVideoBuffer::HandleType, )
    pixelFormat = QVariant(QVideoFrame::PixelFormat, )
    frameSize = QVariant(QSize, QSize(480, 272))
    frameWidth = QVariant(int, 480)
    viewport = QVariant(QRect, QRect(0,0 480x272))
    scanLineDirection = QVariant(QVideoSurfaceFormat::Direction, )
    frameRate = QVariant(double, 30)
    pixelAspectRatio = QVariant(QSize, QSize(1, 1))
    sizeHint = QVariant(QSize, QSize(480, 272))
    yCbCrColorSpace = QVariant(QVideoSurfaceFormat::YCbCrColorSpace, )
    mirrored = QVariant(bool, false) all supported formats: (Format_YUV420P, Format_YV12, Format_NV12, Format_NV21, Format_UYVY, Format_YUYV, Format_RGB32, Format_ARGB32, Format_BGR32, Format_BGRA32, Format_RGB565)
```
```
qt.multimedia.video: media object is QMediaPlayer(0xdc1fa8)
qt.multimedia.video: found videonode plugin "egl" 0xfc60f8

Any ideas?

mozcelikors, 2019-09-02:
Update: gst-launch-1.0 -e -vvv filesrc location="/run/media/sda1/mymovie.mp4" ! qtdemux ! h264parse ! avdec_h264 ! kmssink works.
But qmlvideo does not.

peel70, 2019-09-02:
Ok try with a .mkv file

peel70, 2019-09-02:
Try .mov with gstreamer like me too

mozcelikors, 2019-09-03:
@peel70 Playing .mov using gst-launch-1.0 -e -vvv filesrc location=big_buck_bunny_720p_h264.mov ! qtdemux ! queue ! h264parse ! v4l2h264dec capture-io-mode=4 ! kmssink does not work for me.
But again, I'm interested in playing .mp4 files using qmlvideo. You said you were able to do it. Do you use raspberrypi4 (32-bit) ? Could you share with me your local.conf, and which HEADs and branches you use on meta-raspberrypi, poky, meta-openembedded, meta-qt5, and meta-qt5-extra? Thank you.

peel70, 2019-09-03:
No I’m Sorry the only way I was able to play mp4 was with gstreamer and avdec_h264
.mkv works in qml but not accelerated.
I think qmultimedia use  gstreamer like gstplay so to understand the automatic pipeline applyed by qmultimedia is  configure gst_debug_dot_dir and analize the pipeline.
Maybe you need, and I will try next week because I not at home, to remove omx and retry
I also think that .mp4 does not work accelerated

peel70, 2019-09-04:
I’ve found this interesting article about mp4 decode, if I understand well v4l use ffmpeg for hw decoding so we can try install ffmpeg and enable v4l
https://www.raspberrypi.org/forums/viewtopic.php?t=234778

peel70, 2019-09-04:
In my build I’ve installed ffmpeg and maybe is for this that the pipeline with .mov works for me but maybe something have to be done in the ffmpeg configuration to enable h264dec for the v4l; we have to try this too

peel70, 2019-09-04:
This is interesting too https://www.raspberrypi.org/forums/viewtopic.php?t=179670

I think that we need the support of someone expert in hw decoding and how to enable it on rpi

peel70, 2019-09-04:
Obvious question: are your videos H.264/MPEG-4?

mozcelikors, 2019-09-04:
@peel70 My videos are H.264, your standard .mp4.
I created a Qt issue, hopefully they can share their thoughts with us: https://bugreports.qt.io/browse/QTBUG-78047

peel70, 2019-09-05:
I want to share my experience:
Some years ago I’ve tried Qt’s Boot2Qt (a payment module of qt for embedded development) to try to build image for rpi3
It’s poky based (https://doc.qt.io/QtForDeviceCreation/qtee-custom-embedded-linux-image.html)
Because is a payment module obviously must be stable and tested so it’s not aligned to the latest poky releases
After I’ve built the image I’ve tried to play video and the performance was poor, so I’ve contacted the commercial that enable me the trial and I was able to spoke with the It department that told to me that the problem was not of qt but gstreamer.
Then I’ve tried another product based on RK3399 chip (rockchip); the board come with android but I was able to install Linux via poky, they develop a gstreamer plugin and with qmultimedia I was able to play mp4 and mkv gpu accelerated without changing the code. But it’s a China product with no support and for my business I can use.

I’m receiving qt newsletter and few days a go about qt for MCUs and raspberry it’s not on the list.

I’ve build a qt application that istantiate via c++ the  mediaplayer and the a class derived from videosurface for capturing the frames of the video and try to put them on a OpenGL surface but the problem is in the video decoding infact mediaplayer receive 1or 2 frames for seconds instead of 30 (for example)

In my opinion there something wrong in the gstreamer installation/plugins configuration (mov file works for me in gpu) or Raspbian have to built a gstreamer plugin that talk directly to the gpu.

We can also try to open a bug on v4l for mp4 decode.

peel70, 2019-09-05:
I’ve open an issue on v4l to have a double chance to have an answer; I’m watching your qt issue.
See https://bugzilla.kernel.org/show_bug.cgi?id=204781

peel70, 2019-09-10:
@mozcelikors I'm the same that have commented your QT BUG, please let Valentine check for another solution, as I told you in the previous posts, MediaPlayer can accept gst pipeline.
But consider a multi platform app or the fade in or out of the audio, with gst you need to demux audio and video but is difficult to control the audio volume and the audio output (HDMI / minijack)
I've just tested this from qt 5.11.3 and was a mess (for me)
Today I've restarted from scratch with the current master head but the problem is not fixed.
I've read from Raspbian docs that RPI4 cannot play in hardware decode MPEG-2 and VCI mp4 files (you can check video codec with mediainfo) and that in the new release of burster V4L must be used (the same is YOCTO) try the mp4 that I've told to Valentine if it work without the pipeline.
I've rebuild and cross compiled from burster ligth Qt5.12.4 and gstreamer 1.14.4 and the problem is the same.
So for me we have to wait for a solution that does not depend on us

mozcelikors, 2019-09-10:
@peel70 I got it. I am just curious why playbin does not work for me and it works for you. (I use Yocto, I have ffmpeg installed)

peel70, 2019-09-10:
@mozcelikors in these days I'm reading a lot of article and I have the idea where the problem is but is difficult to try a fix.

try to download this  wget -c http://mirrors.standaloneinstaller.com/video-sample/grb_2.mp4
this file have a Codec ID : isom (isom/iso2/avc1/mp41) 

the .mov have a Codec ID : qt   2005.03 (qt  )

these two files works fine and in GPU (no CPU LOAD when playing)

the files that not work for me have a Codec ID: mp42 (isom/iso2/avc1/mp41)
and they work with this pipeline:

gst-launch-1.0 -e -vvv filesrc location="/home/root/test.mp4" ! qtdemux ! h264parse ! avdec_h264 ! kmssink

avdec_h264 is from gstreamer1.0-libav (formerly ffmpeg) but is software render and, for my application I cannot use software render

as Valentine said (and I just knew) QMultimedia launch gstreamer playbin to play the file, because from the header it read the codec and build the appropriate chain to render video and audio (that you can see with the GST_DEBUG_DUMP_DOT_DIR variable)

below when I've used gstreamer for video conversion you can see the separation of audio and video, you can find example in internet for your purpose, you have to parse video and audio and put video on the videosink and audio con audiosink but you have to know how address the audio card (hdmi or minijack) and the volume

gst-launch-1.0 -v filesrc location="/home/root/test.mp4" ! parsebin name="demux" ! h264parse ! queue ! matroskamux name="mux" ! filesink location="1.mkv" demux. ! aacparse ! queue ! mux.

the only video format that work in gstreamer and qmultimedia with file source path is .mkv (this is because I need RPI4 because it have hw mkv HEVC decoder) but at the moment I wasn't able to play accelerated in GPU

peel70, 2019-09-14:
@mozcelikors after some days of test and switch back to pi3 to make other tests, finally I've got pi4 working to play mp4 in QtMultimedia too.
But at the moment I see only software decode and not hardware but is a big step (for me)
You have to try with 64bit machine and not 32bit.

Please let me know if work for you too.

I've used master branch of all the components.

Build Configuration:
BB_VERSION           = "1.43.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7+snapshot-20190914"
TUNE_FEATURES        = "aarch64 cortexa72 crc crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "master:d7441ddb0c49eafdb620dc45fb7e9a2a92790767"
meta-oe              
meta-multimedia      
meta-networking      
meta-perl            
meta-python          = "master:3061ead8cf38c47ff81d7622cbcf2b8602012212"
meta-raspberrypi     = "master:866ccc829f728f722f623571b24305d2bc9d8b17"
meta-qt5             = "master:12ea469918bcb73605b9c1e5080351f1189c0f17"

peel70, 2019-09-16:
@agherzan these are the tests that I've performed last week:

- RPI3-32bit thud GStreamer 1.14 Qt 5.11.3 the video work in QML hw accelerated via gst-omx CPU 18-20%

- RPI4-32bit master GStreamer 1.16 Qt 5.13 v4l2 decoder is active but I'm not able to play mp4 files, work on .mov with gst-launch and some particular mp4 codec but not in QMultimedia. It play only mkv on QMultimedia and Gstreamer but with 2fps

- RPI4-64bit master GStreamer 1.16 Qt5.13 v4l2 decoder is not enabled (and don't know why) all video play in both GStreamer and QMultimedia but via avdec_h264 (ffmpeg libav) software decode CPU over 250% for the same video file that play on RPI3 at 18-20% CPU

- RPI-32bit burster light with Qt 5.12.3 cross compiled and Gstreamer 1.16 (not compiled by me) same problem ad RPI4-32bit master

My plane is to try Gentoo64 bit as I read is 64bit and v4l2 enabled.

I read this article https://www.raspberrypi.org/forums/viewtopic.php?t=234778 but don't know how to try with poky.

I've not experience in low level linux firmware but I think that v4l2 codec maybe have problems with raspberry's linux firmware.

So what do you think about?

agherzan, 2019-09-16:
I don't have enough experience with this to recommend something without digging into it. If this is reproducible on Raspbian it's worth shouting on the forums as well.

xuhcc, 2019-09-16:
Apparently Kodi is the only player that can play 4K videos on RPi4 with hardware acceleration:
* https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=246837
* https://github.com/popcornmix/xbmc/tree/leia_pi4

peel70, 2019-09-16:
I've just comment the article on raspberry pi forum, but not yet approved so not visible at the moment.

oleszczyk, 2019-12-10:
@peel70 Have you found any solution for this? I am struggling with similar problems: buildroot build for Raspberry Pi 4 and QT application with video stream widget (QML MediaPlayer + VideoOutput). 
For Raspberry Pi 3B+ I used `rpi_userland` package and run QT application with EGL integration `eglfs_brcm`. On RPi 4 this configuration hangs application during start. I am trying to use EGL integration `eglfs_kms` but without any luck to get hardware acceleration.

peel70, 2019-12-10:
Hi @oleszczyk,
I've rebuilt the toolchain with the latest master (ZEUS?) with MACHINE = "raspberrypi4-64"
It seems that the problem on video4linux gstreamer's plugin is fixed.
using gst-launch-1.0 with mp4 and video4linux decoder with kmssink work in GPU (10/12% CPU load during playing of mp4 1080p at 25fps)
On QML with QtMultimedia (QML not Widgets because widgets does not work on opengl) the CPU load is near the 60% for the same video (also using gst's chain in QtMultimedia source property) this because of 2 copy RENDER CPU/VPU I suppose
At the moment there is not H265 gstreamer's HW decoder so play of H265 works in software render and unusable.
Let me know if you need further informations.

peel70, 2019-12-10:
Hi @oleszczyk,
another information, at the moment is not possible to use both video cards in kms

agherzan, 2021-10-30:
This looks like an old issue, on a branch that is not maintained anymore. I'll close it for now.

closed 2021-10-30 by agherzan