agherzan/meta-raspberrypi #483: Boot to U-Boot not working on CM3- warrior

thesillywhat, 2019-08-23:
**Description**
Raspberrypi-Cm3 wont boot up after adding RPI_USE_U_BOOT = "1"

I added the variable RPI_USE_U_BOOT = "1" and After that the device just never boots up.
I see nothing on the screen

**Steps to reproduce the issue:**
1. Added RPI_USE_U_BOOT = "1" to local.conf 

**Describe the results you received:**
Blank screen

**Describe the results you expected:**
Boot screen and then boot to the image

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
blue@blue:~/yacto/rpi-qt5/build$ bitbake qt5-image
Parsing recipes: 100% |######################################################################################################################################################################| Time: 0:01:05
Parsing of 2324 .bb files complete (0 cached, 2324 parsed). 3406 targets, 181 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi-cm3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7.1"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       
meta-qt5-warrior     
meta-oe              
meta-raspberrypi-warrior 
meta-rpi_custom      
meta-python          
meta-networking      
meta-96boards-warrior = "<unknown>:<unknown>"

Initialising tasks: 100% |###################################################################################################################################################################| Time: 0:00:05
Sstate summary: Wanted 308 Found 304 Missed 4 Current 1713 (98% match, 99% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 5574 tasks of which 5402 didn't need to be rerun and all succeeded.

thesillywhat, 2019-08-23:
Added local.conf
[local__conf.txt](https://github.com/agherzan/meta-raspberrypi/files/3536034/local__conf.txt)

thesillywhat, 2019-08-23:
Just an update. I Just added IMAGE_INSTALLS_append = "u-boot"
Now the screen turns on I am able to SSH into it. But somehow the colors on my splash image are completely wrong and terminal screen instead of showing a black color it is showing yellow color
Attached Image

[image]

guillaumekh, 2020-01-20:
I'm having similar issues. Did you find a solution @thesillywhat ?

texierp, 2020-01-20:
@guillaumekh,

> I'm having similar issues. Did you find a solution @thesillywhat ?

Not sure, but It seems you have the solution here: https://github.com/agherzan/meta-raspberrypi/issues/552#issuecomment-574379178

guillaumekh, 2020-01-21:
thanks @texierp :)
I noticed @ikkysleepy's [solution](https://github.com/agherzan/meta-raspberrypi/issues/552#issuecomment-574379178) too a few minutes later. I can confirm patching U-Boot fixes the yellow banding.

@agherzan this issue can probably be closed

agherzan, 2020-01-22:
Cheers @guillaumekh

closed 2020-01-22 by agherzan