agherzan/meta-raspberrypi #492: UART1 not available in 4.19.66 build in master, but works fine in 4.14.112

stevep-korg, 2019-09-10:
This is a different issue than #400, although it also happens on CM3 and CM3+.

Please follow, I'll provide extensive documentation:

Description:

1. Build linux-raspberrypi 4.14.112 in master branch with the proper settings for uart1 in config.txt, and making sure 8250 NR UARTS is 1.
2. Install and boot the system
Result: uart1 is functional on ttyS0.  Observed:

Linux Korg **4.14.112** #1 SMP PREEMPT Wed Sep 4 14:36:53 UTC 2019 aarch64 GNU/Linux
[    0.000000] Kernel command line: cma=256M **8250.nr_uarts=1** bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:55:00:EC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fsm_mask=0x0 dwc_otg.lpm_enable=0  root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.147861] bcm2835-aux-uart 3f215040.serial: irq not found - -517
**[    0.171640] 3f215040.serial: ttyS0 at MMIO 0x0 (irq = 151, base_baud = 61440000) is a 16550**

3. Build linux-raspberrypi **4.19.66** in master branch with the proper settings for uart1 in config.txt (same as for 4.14.112 above), and making sure 8250 NR UARTS is 1.
4. Install and boot the system
Result: uart1 fails to register.  Observed:

Linux Korg 4.19.66 #1 SMP PREEMPT Fri Sep 6 04:46:37 UTC 2019 aarch64 GNU/Linux
[    0.000000] Kernel command line: coherent_pool=1M cma=256M **8250.nr_uarts=1**  smsc95xx.macaddr=B8:27:EB:55:00:EC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fsm_mask=0x0 dwc_otg.lpm_enable=0  root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
**[    0.184922] bcm2835-aux-uart 3f215040.serial: unable to register 8250 port - -22
[    0.184970] bcm2835-aux-uart: probe of 3f215040.serial failed with error -22**

I expected that uart1 would work in 4.19.66 as well as it did in 4.14.112 with the same proper settings and overlays as speified in config.txt and cmdline.txt

There is no reason that uart1 should not work in 4.19.66 when the proper settings and proper overlay is being specified just as it was in 4.14.112, in which uart1 works fine.

stevep-korg, 2019-09-11:
config.txt and cmdline.txt identical for both 4.14.112 and 4.19.66 builds:

[config.txt](https://github.com/agherzan/meta-raspberrypi/files/3598172/config.txt)

[cmdline.txt](https://github.com/agherzan/meta-raspberrypi/files/3598175/cmdline.txt)

stevep-korg, 2019-09-11:
Here are the pertinent lines from config.txt:

# CONFIGURE UART0 FOR KEYSCANNER USE
#  Tell Linux to not drive serial Console on GPIO pins 14 & 15
enable_uart=1
#  Disable Bluetooth (frees up UART0 and connects UART0 TXD0 and RXD0 pins to GPIO pins 14 & 15, respectively)
dtoverlay=pi3-disable-bt
# CONFIGURE UART1 FOR LEGACY MIDI USE, connecting UART1 TXD1 and RXD1 pins to GPIO pins 32 & 33, respectively
dtoverlay=uart1,txd1_pin=32,rxd1_pin=33
# midi-uart1: fake out tty driver so that baudrate 38400 yields 31250 for UART1
dtoverlay=midi-uart1

stevep-korg, 2019-09-11:
These settings along with 8250 NR UARTS 1 should be sufficient for uart1 to be functional.  I would like it if someone involved in the meta-raspberrypi project would please build/run 4.19.66 or very close version in master branch when time permits, and see that the uart1 can not be made available using the reasonable documented settings as shown.

stevep-korg, 2019-09-11:
Some additional filesystems comparisons which may be helpful.  I also have the /var/log/messages from both systems if anyone wishes to see them.

[4_14_112-vs-4_19_66.txt](https://github.com/agherzan/meta-raspberrypi/files/3598205/4_14_112-vs-4_19_66.txt)

stevep-korg, 2019-09-11:
The kernel configs:
[config-4.14.112.zip](https://github.com/agherzan/meta-raspberrypi/files/3598622/config-4.14.112.zip)

[config-4.19.66.zip](https://github.com/agherzan/meta-raspberrypi/files/3598629/config-4.19.66.zip)

agherzan, 2019-09-11:
@stevep-korg Just a heads up that you can actually edit the description so you shouldn't have to follow up with comments when you report an issue.

Have you tried this in Raspbian?

stevep-korg, 2019-09-11:
@agherzan Thanks for the tip.  I am aware I can do that, and actually edited some of those last night.  I thought it would help to put the information in chunks in different entries, but now that I think more about it. maybe it generated a bunch of emails for people, and if so, sorry for the extra noise.

I haven't tried Raspbian, but I can try today to see if it registers it OK.

stevep-korg, 2019-09-11:
@agherzan I tried Raspbian Lite for starters, which does not initialize uart1 at all.  If you searh for "uart" and also "serial" in them messages file, you will notice only the PL011 uart, which is uart0, is initialized.

[messages.zip](https://github.com/agherzan/meta-raspberrypi/files/3602181/messages.zip)
[cmdline.zip](https://github.com/agherzan/meta-raspberrypi/files/3602182/cmdline.zip)
[config.zip](https://github.com/agherzan/meta-raspberrypi/files/3602183/config.zip)

stevep-korg, 2019-09-11:
@agherzan I also trie Raspbian Buster Full (4.19), which it appears by default also does not initialize uart1 at all.  If you search for "uart" and also "serial" in the messages file, you will notice only the PL011 uart, which is uart0, is initialized.  I did not try tweaking the system.

[messages.zip](https://github.com/agherzan/meta-raspberrypi/files/3602761/messages.zip)
[cmdline.zip](https://github.com/agherzan/meta-raspberrypi/files/3602762/cmdline.zip)
[config.zip](https://github.com/agherzan/meta-raspberrypi/files/3602763/config.zip)

Please let me know if you'd like me to try to tweak either of these system to try to initalize uart1, or please let me know if you have a different idea.

stevep-st, 2020-02-02:
@agherzan A few months ago, I took your advice and tested with Raspbian.  The problem occurred there too, so your hunch was correct.  I entered this issue in the kernel repo : https://github.com/raspberrypi/linux/issues/3312

One of the commenters there noted the -22 error during initialization was significant, and posited that the problem appeared to be caused by the use of the uart1-midi overlay.  He asked me to retest without loading that overlay, and if UART1 did not have an initialization error, that I close the bug and write it up differently, in relation to it be a uart1-midi overlay-related  problem.

I am no longer at my previous organization, so I don't have a CM3 on which to continue this testing.  I will try to see if I can recreate it on a Raspberry Pi 3 when time permits.  I'd like to see this issue resolved for the whole Raspberry Pi community, but I don't have the right test platform anymore.  Being able to MIDI on uart1 seems like an essential feature, and the capability to do it in 4.19.66+ should be ensured.

Thanks for your input so far.

stevep-st, 2020-02-07:
Please see https://github.com/raspberrypi/linux/issues/3312

closed 2020-02-07 by stevep-st

agherzan, 2020-02-08:
Thanks @stevep-st for the feedback.