agherzan/meta-raspberrypi #498: omxplayer overlapping with other shared lib providers

kraj, 2019-09-27:
I am seeing this error regularly

```
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libswresample.so.3: omxplayer, libswresample (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/bin/omxplayer.bin)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: omxplayer, libavutil (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavdevice.so.58.3.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: omxplayer, libavutil (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavcodec.so.58.18.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavcodec.so.58: omxplayer, libavcodec (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/bin/omxplayer.bin)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavformat.so.58: omxplayer, libavformat (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/bin/omxplayer.bin)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libswscale.so.5: omxplayer, libswscale (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/bin/omxplayer.bin)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: omxplayer, libavutil (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libswresample.so.3.1.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavcodec.so.58: omxplayer, libavcodec (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavformat.so.58.12.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavformat.so.58: omxplayer, libavformat (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavdevice.so.58.3.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: omxplayer, libavutil (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavfilter.so.7.16.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libswresample.so.3: omxplayer, libswresample (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavcodec.so.58.18.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: omxplayer, libavutil (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libswscale.so.5.1.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: omxplayer, libavutil (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavformat.so.58.12.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: omxplayer, libavutil (used by files: /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/bin/omxplayer.bin)
```

It seems omxplayer is bundling stuff, so we need to be careful and fix the providers to address this error.

adwaitpatil, 2020-01-06:
I am also facing a similar issue:

`ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavformat.so.58: libavformat, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/bin/omxplayer.bin)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: libavutil, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavcodec.so.58.18.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: libavutil, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libswresample.so.3.1.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: libavutil, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavformat.so.58.12.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavcodec.so.58: libavcodec, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavformat.so.58.12.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavcodec.so.58: libavcodec, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/bin/omxplayer.bin)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: libavutil, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libswscale.so.5.1.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libswresample.so.3: libswresample, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavcodec.so.58.18.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: libavutil, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/bin/omxplayer.bin)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavformat.so.58: libavformat, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavdevice.so.58.3.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libswscale.so.5: libswscale, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/bin/omxplayer.bin)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: libavutil, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavdevice.so.58.3.100)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libswresample.so.3: libswresample, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/bin/omxplayer.bin)
ERROR: omxplayer-git-r5 do_package: omxplayer: Multiple shlib providers for libavutil.so.56: libavutil, omxplayer (used by files: /home/adwait/yocto/rpi/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/omxplayer/git-r5/packages-split/omxplayer/usr/lib/omxplayer/libavfilter.so.7.16.100)
`
It is preventing the build from completing.
Any known workaround?

trowhite, 2020-01-14:
I made a change to meta-rpi/recipes-multimedia/omxplayer/omxplayer_git.bbappend (copied from https://54.194.93.196/c/yocto/meta-boot2qt/+/280226) that seems to resolve this issue.

```
PRIVATE_LIBS += "\
    libavcodec.so.58 \
    libavformat.so.58 \
    libavutil.so.56 \
    libswresample.so.3 \
    libswscale.so.5 \
"
```
Hope this helps.

kraj, 2020-01-14:
perhaps turn this into a patch ?

agherzan, 2021-10-30:
@leon-anavi @trowhite are you guys up for a patch if this is still an issue in master? I know it's an old issue hence asking.

deets, 2022-01-08:
@agherzan I built today with hardknott and had to apply this, so IMHO still worth applying.

agherzan, 2022-01-10:
@deets thanks for confirming. Could you check it on master as well?

deets, 2022-01-14:
@agherzan I just tried but got not too surprisingly caught in the whole syntax-change issue, so it's at least not easy - I'd have to update a few of my own recipes of which this is part of. Would expect the behaviour to change?

agherzan, 2022-07-29:
@deets Any chance you are still up for pushing a patch for this?

deets, 2022-08-05:
I'll try to find time in september, I'm on vacation now for a while. But I got some more exposure to the newer yocto releases, so I feel more confident about working against master. I made myself a reminder so I can get back to you.