agherzan/meta-raspberrypi #499: kernel fails to compile (warrior branch)

BlackZork, 2019-10-01:
```
| make[3]: *** No rule to make target 'arch/arm/boot/dts/overlays/hifiberry-dacplusdsp.dtbo'.  Stop.
| make[2]: *** [arch/arm/Makefile:345: overlays/hifiberry-dacplusdsp.dtbo] Error 2 
```
There no hifiberry-dacplusdsp.dtbo in SRCREV=13ce09db830e0c2fe524460b0673afb974359bc2

set in recipes-kernel/linux/linux-raspberrypi_4.19.bb

My workaround is to bump it to 4d486c17f636bb339c10cd73a07292220a973e01

agherzan, 2019-10-02:
Where do we include this dtb target?

BlackZork, 2019-10-02:
Ok, I found that I included it in my own bbappend file. I am sorry for this false report.

closed 2019-10-02 by BlackZork

agherzan, 2019-10-02:
That's what I suspected. I'm also fine in including it and updating the kernel. Feel free to push a PR for master for that.