agherzan/meta-raspberrypi #500: RPI 3b+ with official touchscreen: problems after upgrade from thud to warrior

BlackZork, 2019-10-02:
7" PI touchscreen shows boot phase, but as soon as psplash is shown screen freezes.

I tried so far:
- default kernel on warrior branch
- 4.14.112 kernel on warrior branch
- [meta-rpi layer](https://github.com/jumpnow/meta-rpi) - it uses kernel 4.19.73

Raspbian (Release 2019-09-26, 4.19.75 kernel) it boots and works ok.

Anyone has a working configuration for warrior branch?

agherzan, 2019-10-04:
@BlackZork Do you have a stable power supply? Also, can you attach a serial console to see if the entire board is frozen?

BlackZork, 2019-10-04:
It is a problem with vc4 video driver. When I blacklisted it in /etc/modprobe.d touchscreen works OK.

I tried to use VC4DTBO = "vc4-fkms-v3d" in layer.conf, but effect is the same.

I ended with MACHINE_FEATURES_remove = "vc4graphics", which I assume is a return to old working configuration. Will see when I add QT5 to my new layer.

BTW, hdmi output works OK with vc4 kernel module loaded.

agherzan, 2019-10-05:
Interesting. I sadly don't have a display at my disposal. @floion @zubair do you happen to know anything about this issue? @BlackZork this is on DSI, right?

BlackZork, 2019-10-07:
Yes, this is Official Raspberry PI 7" Touchscreen bought about a year ago.

BlackZork, 2020-05-26:
I upgraded to Dunfell and vc4graphics works ok with PI 7" Touchscreen.
Not sure if I should close this issue.

agherzan, 2020-06-09:
I'll close it as warrior is not an actively supported branch. If anyone wants to investigate it as part of that release, I'll be accepting patches.

closed 2020-06-09 by agherzan