agherzan/meta-raspberrypi #502: Should change /dev/vchiq etc ownership / permission by a udev rule?

sprhawk, 2019-10-09:
raspberry pi driver will create several devices:
/dev/vchiq
/dev/vcsm
/dev/vcio

which are with root:root 0600 ownership.

Due to this, if user will use raspistill or raspivid application, it will require root permission.

There is solution to include a udev rules with contents:
```
SUBSYSTEM=="vchiq",GROUP="video",MODE="0660"
SUBSYSTEM=="vc-sm",GROUP="video",MODE="0660"
SUBSYSTEM=="bcm2708_vcio",GROUP="video",MODE="0660"
```

and add command to extra_user_params with:  `usermod -a -G video pi`

then user can use those camera related commands without root permission.

I added them with my own layer for my own project, but I think it is better to be put inside meta-raspberrypi ( raspbian has that rules I think)

agherzan, 2019-10-09:
@sprhawk I'm tempted to say that this sounds like a distro configuration. It very much depends on how you want to name your groups, users or if at all. From the perspective of a BSP layer, I'd say that it provides all the primitives for you to build on top. This is a user management "feature" at the core. What do you think?

agherzan, 2019-10-09:
I'm up for a doc PR for this @sprhawk

agherzan, 2021-10-30:
Closing due to inactivity.

closed 2021-10-30 by agherzan