agherzan/meta-raspberrypi #513: rpi0wifi wlan0 interface missing warrior

nandra, 2019-10-23:
Hi,

I'm building image using warrior branch for rpi0w and wlan0 interface doesn't appear. Is anybody experiencing same issues? I have some other project using rocko and it works perfectly fine. Thanks.

nandra, 2019-10-23:
Same issue like #184 but I have files installed in my final rootfs:
```
root@raspberrypi0-wifi:~# ls -la /lib/firmware/brcm/
drwxr-xr-x    2 root     root          1024 Oct 23 08:07 .
drwxr-xr-x    3 root     root          1024 Oct 23 08:07 ..
-rw-r--r--    1 root     root         36264 Sep 26 13:18 BCM43430A1.hcd
-rw-r--r--    1 root     root        388739 Sep 26 13:18 brcmfmac43430-sdio.bin
-rw-r--r--    1 root     root          1121 Sep 26 13:18 brcmfmac43430-sdio.txt

```

agherzan, 2019-10-23:
@nandra can you attach the dmesg log?

nandra, 2019-10-24:
@agherzan here you go:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.71 (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 Fri Oct 4 12:41:57 UTC 2019
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1d400000
[    0.000000] On node 0 totalpages: 122880
[    0.000000]   Normal zone: 1080 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x90/0x4a4 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121800
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bct
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 466940K/491520K available (6948K kernel code, 635K rwdata, 2080K rodata, 452K init, 796K bss, 1)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xde800000 - 0xff800000   ( 528 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xde000000   ( 480 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (6950 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   ( 452 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 636 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 797 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 25141 entries in 74 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000024] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000068] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000163] bcm2835: system timer (irq = 27)
[    0.000475] Console: colour dummy device 80x30
[    0.000899] console [tty0] enabled
[    0.000959] Calibrating delay loop... 996.14 BogoMIPS (lpj=4980736)
[    0.040287] pid_max: default: 32768 minimum: 301
[    0.040690] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.040730] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.041739] CPU: Testing write buffer coherency: ok
[    0.042722] Setting up static identity map for 0x8200 - 0x8238
[    0.043775] devtmpfs: initialized
[    0.051745] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.052089] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.052144] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.053146] pinctrl core: initialized pinctrl subsystem
[    0.054341] NET: Registered protocol family 16
[    0.056729] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.061936] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.061991] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.062111] Serial: AMBA PL011 UART driver
[    0.064580] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.094883] bcm2835-dma 20007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[    0.096728] SCSI subsystem initialized
[    0.096965] usbcore: registered new interface driver usbfs
[    0.097068] usbcore: registered new interface driver hub
[    0.097234] usbcore: registered new device driver usb
[    0.110580] raspberrypi-firmware soc:firmware: Attached to firmware from 2019-08-22 13:58, variant start
[    0.120599] raspberrypi-firmware soc:firmware: Firmware hash is 0e6daa5106dd4164474616408e0dc24f997ffcf3
[    0.132229] clocksource: Switched to clocksource timer
[    0.212647] VFS: Disk quotas dquot_6.6.0
[    0.212780] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.213041] FS-Cache: Loaded
[    0.213318] CacheFiles: Loaded
[    0.226920] NET: Registered protocol family 2
[    0.227910] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[... 157 lines cut ...]
[    6.034880] Bluetooth: Core ver 2.22
[    6.038104] NET: Registered protocol family 31
[    6.040306] Bluetooth: HCI device and connection manager initialized
[    6.045820] Bluetooth: HCI socket layer initialized
[    6.048029] Bluetooth: L2CAP socket layer initialized
[    6.050607] Bluetooth: SCO socket layer initialized
[    6.067128] urandom_read: 1 callbacks suppressed
[    6.067141] random: avahi-daemon: uninitialized urandom read (4 bytes read)
[    6.103128] random: avahi-daemon: uninitialized urandom read (4 bytes read)
[    6.138748] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.140623] Bluetooth: BNEP filters: protocol multicast
[    6.151534] Bluetooth: BNEP socket layer initialized
root@raspberrypi0-wifi:~# ifconfig -a
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@raspberrypi0-wifi:~# 

```

agherzan, 2019-10-24:
It doesn't seem to have detected any wifi device. Could it be a dts issue? Can you check it out against the known (rocko) setup? Also, I'd try to compare it against a raspbian distro trying to find what we miss.

nandra, 2019-10-29:
Looks like wifi died on this board. Verified with latest raspbain and wlan0 is not showing up ;)

agherzan, 2019-10-29:
Easy one :) Glad you got to the bottom of it.

closed 2019-10-29 by agherzan