agherzan/meta-raspberrypi #515: Update RPi4 firmware

saeugetier, 2019-10-29:
**Description**

The RaspberryPi Foundation releases a tool "rpi-eeprom-update" for updating the RPi4 firmware. Is there a way to update the firmware with Yocto? Any plans?

agherzan, 2019-10-31:
@saeugetier That is something I have on my plate for the next weeks. There is some external integration done by Balena so I hope we will migrate some of that functionality to the BSP in the time to come. @floion do you think that is something you guys can push in the near future? Otherwise, I'll take a look at it myself.

saeugetier, 2019-10-31:
Anything I can help with or look at?

floion, 2019-10-31:
@agherzan not sure how we'd go about adding firmware update support in the BSP. Maybe just add the package as it is and people would call the update binary themselves?

agherzan, 2019-10-31:
Sounds like a plan.

macpijan, 2020-09-08:
@saeugetier @agherzan Any work was done in this area? Can we simply get the rpi-eeprom recipe and integrate it into the meta-respberrypi to enable all Yocto users with the ability to upgrade the EEPROM firmware?

agherzan, 2020-09-08:
No progress happened sadly in the meanwhile. I'm wondering if there is something @floion is also interested in as part of the Balena platform.

jmagnuson, 2025-01-11:
Can this be closed with #1131 in tree?

agherzan, 2025-01-31:
Nice one, @jmagnuson. Closed.

closed 2025-01-31 by agherzan