agherzan/meta-raspberrypi #519: U-Boot activating gpio 34-39 alt sdio mode on cm3

nzottmann, 2019-11-06:
**Description**

After enabling U-Boot by adding `RPI_USE_U_BOOT = "1"` to `local.conf`, raspi-gpio shows that gpios 34-39 are set to SD1 function:
```
...
GPIO 31: level=0 fsel=0 func=INPUT
GPIO 32: level=1 fsel=7 alt=3 func=TXD0
GPIO 33: level=0 fsel=7 alt=3 func=RXD0
GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK
GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
GPIO 37: level=0 fsel=7 alt=3 func=SD1_DAT1
GPIO 38: level=0 fsel=7 alt=3 func=SD1_DAT2
GPIO 39: level=0 fsel=7 alt=3 func=SD1_DAT3
GPIO 40: level=0 fsel=0 func=INPUT
...
Without U-Boot (remove `RPI_USE_U_BOOT = "1"`), gpios 34-39 are set to INPUT.
```

**Steps to reproduce the issue:**
- Add meta-raspberry and deps
- In `local.conf`:
  - `MACHINE ??= "raspberrypi-cm3"`
  - `IMAGE_INSTALL_append = "raspi-gpio"`
- Check pins 34-39 with `raspi-gpio get`
- Add `RPI_USE_U_BOOT = "1"`
- Check pins 34-39 with `raspi-gpio get` again.

**Describe the results you received:**
GPIO 34-39 are configured as `fsel=7 alt=3 func=SD1_*` after activating U-Boot, see output above.

**Describe the results you expected:**
GPIO 34-39 should be configured as INPUT (which is the default). U-Boot should not touch the mode of these pins. The board is a compute module, so there is nothing connected to these pins and U-Boot does not need anything to be connected there.
U-Boot should not configure any pins it does not need as this causes conflicts with other software relying on these pins.

**Additional information you deem important (e.g. issue happens only occasionally):**
Why is this a problem?
I have connected a sd card to pins 22-27. After activating them with the sdio overlay, both 22-27 and 34-39 are configured to SD1 alt modes, which leads to none of them working.

U-Boot also configures gpio 32 and 33 for uart use, so this bug is kind of related: https://github.com/agherzan/meta-raspberrypi/issues/374
The uart is used by U-Boot for a console, but I think the sd port should not be enabled by default. I don't know how to disable it.

**Additional details (revisions used, host distro, etc.):**
Used warrior branch, tested on cm3 and cm3+.

nzottmann, 2019-11-07:
I don't know how to prevent U-Boot from configuring there pins, but I found the position in the source code where the pins are selected: https://gitlab.denx.de/u-boot/u-boot/blob/v2019.01/arch%2Farm%2Fdts%2Fbcm2837-rpi-3-b.dts#L39.

With the following patch the pins can be changed from 34-39 to 22-27:
```diff
---
 arch/arm/dts/bcm2837-rpi-3-b.dts | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/arch/arm/dts/bcm2837-rpi-3-b.dts b/arch/arm/dts/bcm2837-rpi-3-b.dts
index 20725ca487..3d8eb933d9 100644
--- a/arch/arm/dts/bcm2837-rpi-3-b.dts
+++ b/arch/arm/dts/bcm2837-rpi-3-b.dts
@@ -36,7 +36,7 @@
 /* SDHCI is used to control the SDIO for wireless */
 &sdhci {
        pinctrl-names = "default";
-       pinctrl-0 = <&emmc_gpio34>;
+       pinctrl-0 = <&emmc_gpio22>;
        status = "okay";
        bus-width = <4>;
        non-removable;
```
This workaround is ok for me now. I still think it's a bug, look at the comment above the linked code which says "*SDHCI is used to control the SDIO for wireless*".
This is the case for a pi 3B(+), but the compute module has no wifi, these pins are not used on a compute module. Seems like a separate configuration for compute modules is missing.

agherzan, 2019-11-07:
@nzottmann good catch. uboot does its own mux configuration by providing a dts. I know there was some work that we used on RPi4 to reuse the dt blob provided by the firmware and pass it as is to the kernel. I've seen that RPi3 still has it's own dts so it might be worth raising this question in uboot. I the meanwhile see `CONFIG_OF_BOARD` (https://github.com/u-boot/u-boot/commit/193279d784e2ca2225815803b9a63f1ce3c7389d#diff-07d1b1871be7f5b207b684f300782042).

nzottmann, 2019-11-07:
I would be happy to push a PR, but I think my workaround is no solution.
My workaround patch changed the pins from 34-39 to 22-29 because my application has a sd card on 22-29, but I would like to prevent U-Boot from configuring the pins and do the config with a device tree overlay activated by config.txt.

I don't know how to prevent U-Boot from configuring these pins. I tried changing the `*defconfig` files you mentioned, but even deleting all `/configs/rpi_*deconfig` files did not have any effect on the gpio configuration in U-Boot (checked with `gpio status -a` in U-Boot console).

agherzan, 2019-11-07:
Try to use CONFIG_OF_BOARD instead of CONFIG_EMBED (that's what I meant above).

nzottmann, 2019-11-07:
I tried changing CONFIG_EMBED to CONFIG_OF_BOARD in configs/rpi_3_defconfig and configs/rpi_3_32b_defconfig, but this showed no effect. As deleting these files also has no impact, I am confused about their purpose. Which code is responsible for using the defconfigs?

nzottmann, 2021-02-15:
After switching my build from warrior (using u-boot 2019.01) to dunfell (u-boot 2020.01), the patch mentioned above did not work anymore, so I did some research and to see what changed and how to get sdio on 22-27 working again.

### Config file "chain"
In `/build/conf/local.conf` the machine is defined as `MACHINE ?= "raspberrypi-cm3"`, which refers to `raspberrypi-cm3.conf` where a `UBOOT_MACHINE` `rpi_3_32b_config` is defined:
https://github.com/agherzan/meta-raspberrypi/blob/77190af02d48adc2b28216775e6318e9eeda571c/conf/machine/raspberrypi-cm3.conf#L10

This refers to the config file rpi_3_32b_defconfig, which sets the device tree `bcm2837-rpi-3-b`:
```
CONFIG_DEFAULT_DEVICE_TREE="bcm2837-rpi-3-b"
```
https://github.com/u-boot/u-boot/blob/0b0c6af38738f2c132cfd41a240889acaa031c8f/configs/rpi_3_32b_defconfig#L22

This refers to the device tree file `bcm2837-rpi-3-b.dts`: https://github.com/u-boot/u-boot/blob/v2020.01/arch/arm/dts/bcm2837-rpi-3-b.dts

### Change
Something changed in `bcm2837-rpi-3-b.dts` from u-boot 2019.01 to 2020.01. I did not track this down.

### Initial problem
`UBOOT_MACHINE = "rpi_3_32b_config" ` uses the defconfig and device tree of the rpi 3b because there is no defconfig for the cm3. I suggested adding a cm3 defconfig in #812

### New workaround
Meanwhile, to get sdio on 22-27 working again, sdio on 34-39 has to be deactivated, otherwise there are two pin sets configured with alt mode `SD1_*`. I could achieve this either by

- Removing
  ```
  CONFIG_MMC_SDHCI=y
  CONFIG_MMC_SDHCI_BCM2835=y
  ```
  from https://github.com/u-boot/u-boot/blob/0b0c6af38738f2c132cfd41a240889acaa031c8f/configs/rpi_3_defconfig#L29-L30
- Changing
  ```
  CONFIG_DEFAULT_DEVICE_TREE="bcm2837-rpi-3-b"
  ```
  to
  ```
  CONFIG_DEFAULT_DEVICE_TREE="bcm2837-rpi-cm3-io3"
  ```
  in https://github.com/u-boot/u-boot/blob/0b0c6af38738f2c132cfd41a240889acaa031c8f/configs/rpi_3_defconfig#L22

with a patch for u-boot (https://wiki.yoctoproject.org/wiki/TipsAndTricks/Patching_the_source_for_a_recipe).

### Conclusion

This issue should be closed, as it is just one possible symptom of #812.

closed 2021-02-15 by nzottmann