agherzan/meta-raspberrypi #524: Pi 3 B+ does not boot without HDMI on a "fast SD card"

brocaar, 2019-11-13:
I have an issue booting an image based on the meta-raspberrypi BSP layer using the thud branch (latest version). Full source of my setup can be found at: https://github.com/brocaar/chirpstack-gateway-os.

What I have noticed is that when flashing the same image using Balena Etcher:

* This boot on a slow / cheap SD card
* It does boot on the more expensive and faster SD card when a HDMI monitor is plugged in
* It does not boot on a more expensive faster SD card without a HDMI monitor

After changing the `config.txt` `hdmi_force_hotplug=1`, I am able to see the output when connecting the HDMI cable when the boot "hangs". Please see below the output:

[image]

Do you have any idea what could be the cause? The closest thing I could find was http://u-boot.10912.n7.nabble.com/PATCH-0-1-dwc2-USB-controller-hangs-with-lan78xx-td332554.html.

Others reported that even a HDMI dummy solves the issue: https://github.com/brocaar/chirpstack-gateway-os/issues/9.

agherzan, 2019-11-15:
I haven't seen this before. is this the case with another OS? For example Raspbian.

brocaar, 2019-11-18:
@agherzan no, I have only seen this issue when using meta-raspberrypi (using u-boot).

agherzan, 2019-11-18:
@brocaar But have you tried to reproduce with a Raspbian OS?

brocaar, 2019-11-19:
@agherzan yes, but then all works fine. What I'm going to try in the next days is to create a meta-raspberrypi based image without u-boot to see if that makes any difference.

brocaar, 2019-11-24:
I'm closing this issue. After some more testing I'm not sure if this is an issue with meta-raspberrypi.

closed 2019-11-24 by brocaar

brocaar, 2019-11-26:
@agherzan to give you a summary of the issue (it was my own mistake):

My image uses an initramfs for mounting an overlayfs on top of a read-only rootfs. When the initramfs script starts **most of the times** the MMC block-device is initialized, however this is not guaranteed (I found out). Depending on the sdcard speed and if HDMI is connected or not, results might vary which is what happened.

Since `hdmi_force_hotplug=1` results in a lower resolution than when booting with HDMI connected (for my monitor), I missed that earlier in the boot process the mount command in my initramfs script failed.

This fixed the issue:
https://github.com/brocaar/chirpstack-gateway-os/commit/48b425160ed7151281e4669c99fe04c821144eb2

agherzan, 2019-12-01:
Thanks @brocaar for the explaination.