agherzan/meta-raspberrypi #526: require conf/machine/include/tune-cortexa72.inc

kgbook, 2019-11-22:
**Description**
I cloned from *warrior* branch commit *5cb49f4d76a311201bd36cf6538e23a59e3d8030*.
However, any bitbake operation will fail.

**Steps to reproduce the issue:**
1. `git clone git://git.yoctoproject.org/meta-raspberrypi meta-rpi`
2. configure environment using `source oe-init-build-env`.
3. `vim build/conf/local.conf`, assign to `"raspberrypi4-64"` to *MACHINE*.
4. `vim build/conf/bblayer.conf`, append `/home/user/poky/meta-rpi` to *BBLAYERS*.
5. `bitbake gstreamer`

**Describe the results you received:**
ParseError at */home/user/poky/meta-rpi/conf/machine/raspberrypi4-64.conf*:12: Could not include required file *conf/machine/include/tune-cortexa72.inc*
```shell
ERROR: Traceback (most recent call last):
  File "/home/user/poky/bitbake/lib/bb/cookerdata.py", line 175, in wrapped
    return func(fn, *args)
  File "/home/user/poky/bitbake/lib/bb/cookerdata.py", line 185, in parse_config_file
    return bb.parse.handle(fn, data, include)
  File "/home/user/poky/bitbake/lib/bb/parse/__init__.py", line 107, in handle
    return h['handle'](fn, data, include)
  File "/home/user/poky/bitbake/lib/bb/parse/parse_py/ConfHandler.py", line 149, in handle
    statements.eval(data)
  File "/home/user/poky/bitbake/lib/bb/parse/ast.py", line 39, in eval
    statement.eval(data)
  File "/home/user/poky/bitbake/lib/bb/parse/ast.py", line 63, in eval
    bb.parse.ConfHandler.include(self.filename, s, self.lineno, data, False)
  File "/home/user/poky/bitbake/lib/bb/parse/parse_py/ConfHandler.py", line 95, in include
    ret = bb.parse.handle(fn, data, True)
  File "/home/user/poky/bitbake/lib/bb/parse/__init__.py", line 107, in handle
    return h['handle'](fn, data, include)
  File "/home/user/poky/bitbake/lib/bb/parse/parse_py/ConfHandler.py", line 149, in handle
    statements.eval(data)
  File "/home/user/poky/bitbake/lib/bb/parse/ast.py", line 39, in eval
    statement.eval(data)
  File "/home/user/poky/bitbake/lib/bb/parse/ast.py", line 61, in eval
    bb.parse.ConfHandler.include(self.filename, s, self.lineno, data, "include required")
  File "/home/user/poky/bitbake/lib/bb/parse/parse_py/ConfHandler.py", line 98, in include
    raise ParseError("Could not %(error_out)s file %(fn)s" % vars(), parentfn, lineno)
ParseError: ParseError at /home/user/poky/meta-rpi/conf/machine/raspberrypi4-64.conf:12: Could not include required file conf/machine/include/tune-cortexa72.inc

ERROR: Unable to parse conf/bitbake.conf: ParseError at /home/user/poky/meta-rpi/conf/machine/raspberrypi4-64.conf:12: Could not include required file conf/machine/include/tune-cortexa72.inc
```

**Describe the results you expected:**
well, bitbake successfully.

**Additional information you deem important (e.g. issue happens only occasionally):**
*tune-cortexa72.inc* not found in */home/user/poky/meta-rpi/conf/machine/include*.

There is a file named **tune-arm1176jzf-s.inc**.

Perhaps filename is wrong, it should be **tune-arm1176jzf-s.inc**, or someone forgot to commit *tune-cortexa72.inc*.

The error disappear after I modify *raspberrypi4-64.conf* like below:
```
#require conf/machine/include/tune-cortexa72.inc
require conf/machine/include/tune-arm1176jzf-s.inc
```

I'm not sure whether my operation is wrong or not.

**Additional details (revisions used, host distro, etc.):**

pbrkr, 2019-11-22:
The file does exist in openembedded-core on the warrior branch: https://git.openembedded.org/openembedded-core/tree/meta/conf/machine/include/tune-cortexa72.inc?h=warrior

Could you confirm which branches/commits of other layers you're using?

kgbook, 2019-11-22:
> The file does exist in openembedded-core on the warrior branch: https://git.openembedded.org/openembedded-core/tree/meta/conf/machine/include/tune-cortexa72.inc?h=warrior
> 
> Could you confirm which branches/commits of other layers you're using?

As I decribe above, warrior branch, Commit 5cb49f4d76a311201bd36cf6538e23a59e3d8030 
I tried to `git pull` to fetch the latest commit, then prompt **Current branch warrior is up to date**.

pbrkr, 2019-11-22:
Which branches/commits of poky or openembedded-core and any other layers are you using though?

kgbook, 2019-11-22:
commit 87631919819b6f85f23f57689cd1065c64d7fb03 of master branch, poky.

pbrkr, 2019-11-22:
That looks to be a very old commit!

You need to match the branches you're using. So if you use the warrior branch of meta-raspberrypi you should also be using the warrior branch of poky and of any other layers you add (if any).

Please checkout the warrior branch of poky and try again, let us know if you still see problems after that.

kgbook, 2019-11-22:
It's ok now.
Another issue occurred, I'm trying  to work it out.
Thanks.

closed 2019-11-22 by kgbook