agherzan/meta-raspberrypi #527: Broken uart input comminucation on warrior branch

taschenuhr, 2019-11-26:
**Description**

Upgrading from sumo to warrior broke the uart communication on the raspberry pi 3b+.
Communication on the OUT pin (Tx) still works, but the IN pin (Rx) doesn't read anything.

This worked fine on the sumo branch, including an update via mender (using uboot as well).
But flashing a new SD card with the same image, results in the issue.

**Steps to reproduce the issue:**
1. in `local.conf` add following lines
```
ENABLE_UART = "1"
RPI_EXTRA_CONFIG = "dtoverlay=pi3-disable-bt"
```
2. used wiringPi to set RTS and CTS pins (might be optional):
```
gpio -g mode 17 alt3
gpio -g mode 16 alt3
```

**Describe the results you received:**
Using `minicom -b 1000000 -D /dev/ttyAMA0` there is no reponse at all to any input, that is send.

**Describe the results you expected:**
Expected is at least some kind of reponse, due to faulty input.

**Additional information you deem important (e.g. issue happens only occasionally):**
* A custom HAT was used to send/receive data via uart, which successfully receives data from the raspberry pi on warrior and sumo branch.

```
$ wiringPi readall:
+-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   IN | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 |   IN | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT0 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT0 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 | ALT3 | 0 | 11 || 12 | 1 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | ALT3 | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 1 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
```

**Additional details (revisions used, host distro, etc.):**
used warrior branch, tested on raspberry pi 3b+

agherzan, 2019-12-01:
Are you able to reproduce this on zeus? Have you also tried to reproduce this on raspbian?

taschenuhr, 2019-12-02:
I couldn't test it on zeus, due to the projects dependency to mender, which isn't available for zeus, yet.

But after some more testing, the patch described in
https://github.com/agherzan/meta-raspberrypi/issues/374#issuecomment-457524421 solved my issue, too.
Therefore i'll close the ticket.

closed 2019-12-02 by taschenuhr