agherzan/meta-raspberrypi #528: uBoot does not build for rpi 4 64 bit

ThomasD13, 2019-11-28:
**Description**
Trying to build uBoot for booting a raspberry pi 4 64 bit image fails during configure task:

| scripts/kconfig/conf  --defconfig=arch/../configs/rpi_4_defconfig Kconfig
| ***
| *** Can't find default configuration "arch/../configs/rpi_4_defconfig"!
| ***
| /home/ewdt/oe/yocto/build-rpi/tmp/work/raspberrypi4_64-poky-linux/u-boot/1_2019.07-r0/git/scripts/kconfig/Makefile:128: recipe for target 'rpi_4_defconfig' failed

**Steps to reproduce the issue:**
1. Checkout master of meta-raspberrypi (current ed6b75b).
2. Enable usage of uBoot via RPI_USE_U_BOOT = "1" (local.conf)
3. Set MACHINE to "raspberrypi4-64" (local.conf) to build image for raspberry pi 4 64 bit
4. Build the "core-image-base" with bitbake

ERROR: Task (/home/ewdt/oe/yocto/poky/meta/recipes-bsp/u-boot/u-boot_2019.07.bb:do_configure) failed with exit code '1'

| scripts/kconfig/conf  --defconfig=arch/../configs/rpi_4_defconfig Kconfig
| ***
| *** Can't find default configuration "arch/../configs/rpi_4_defconfig"!
| ***
| /home/ewdt/oe/yocto/build-rpi/tmp/work/raspberrypi4_64-poky-linux/u-boot/1_2019.07-r0/git/scripts/kconfig/Makefile:128: recipe for target 'rpi_4_defconfig' failed

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
It seems that there was a patch provided which maybe addresses that issue:
https://github.com/agherzan/meta-raspberrypi/commit/81f79f15c39ccdb058b0e91aa7e4fac7fc94ed3f

And it was some days later removed:
https://github.com/agherzan/meta-raspberrypi/commit/4776ecc5325978c09d3c9c92379693797db9de78

I am not sure if this is maybe related to this?

ThomasD13, 2019-11-28:
Actually I found the issue:
I used to be on the zeus branch of poky. There is still uBoot version "2019.07" used, which has not the patches of 81f79f1.

I cherry picked fdc1ccbb969b2fe80f0b32dce818ee237451c816 from poky (which uses uBoot 2019.10) and now it works like charm.

closed 2019-11-28 by ThomasD13

Craggles17, 2020-03-30:
Having this issue too