agherzan/meta-raspberrypi #529: krogoth cannot support raspberrypi3-64

churmenton, 2019-11-29:
**Description**

**Steps to reproduce the issue:**
1.donwload poky and meta-raspberrypi 
2.both git checkout krogoth branch
3.source oe-init-build-env
4.set machine to raspberrypi3-64 in local.conf and add meta-raspberrypi layer to bblayer.conf

**Describe the results you received:**
Error

**Describe the results you expected:**
Successful

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

pbrkr, 2019-11-29:
Support for raspberrypi3-64 was added after the krogoth branch was released and there's no plans to backport this.

Krogoth was released in April 2016 so it's long been out of support. Please upgrade to a newer Yocto Project version if you can!

closed 2019-11-29 by pbrkr

churmenton, 2019-11-29:
Hi Paul,

Thanks for your explanation.
Our project is based on this Yocto Project and I need to krogoth to support raspberrypi3-64.
I would like to maintain this branch if you give me some advice on how to make it.

Best regards
Churmenton Sue

pbrkr, 2019-11-29:
You'll have to backport the machine configuration, probably update the kernel, u-boot and firmware versions and maybe update other recipes as well. You may be able to look at the morty branch for reference as that's the oldest one that looks to have raspberrypi3-64 support. I can't give any guarantees you'll be successful though - for new features like 64-bit support you really should upgrade.

agherzan, 2019-12-01:
@churmenton I agree with Paul. You should really try to upgrade. If that is not an option, start to backport machine, kernel and firmware and that *should* boot. From there you can test the other bits (for example WiFi firmware).

churmenton, 2019-12-10:
@agherzan I have backported machine, kernel, and firmware successfully according to your suggestions. A pull request has been created if you think my modification is OK, please accept it.