agherzan/meta-raspberrypi #531: RPi0w - wlan0: Failed to initialize driver interface

bernardoaraujor, 2019-12-04:
**Description**

This is similar to what has been reported at https://github.com/agherzan/meta-raspberrypi/issues/31 and https://www.yoctoproject.org/pipermail/yocto/2017-May/036093.html

I have a Raspberry Pi 0W.
I'm building with poky, meta-openembedded and meta-raspberrypi all on warrior.

This is the bottom of `conf/local.conf`:
```
...
ENABLE_UART = "1"

DISTRO_FEATURES_append = " wifi systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"

IMAGE_INSTALL_append = " wpa-supplicant"
```

As suggested in https://github.com/agherzan/meta-raspberrypi/issues/31 I tried adding `IMAGE_INSTALL_append = " linux-firmware-brcm43430 "` (doesn't build) and `IMAGE_INSTALL_append = " linux-firmware-bcm43430 "`, without success.

**Describe the results you received:**
```
root@raspberrypi0-wifi:~# ifconfig
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:80 errors:0 dropped:0 overruns:0 frame:0
          TX packets:80 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:6080 (5.9 KiB)  TX bytes:6080 (5.9 KiB)

root@raspberrypi0-wifi:~# ifup wlan0
Successfully initialized wpa_supplicant
rfkill: Cannot get wiphy information
Could not read interface wlan0 flags: No such device
WEXT: Could not set interface 'wlan0' UP
wlan0: Failed to initialize driver interface
ip: SIOCGIFFLAGS: No such device
```

**Describe the results you expected:**
I expect `wlan0` to be available.

**Additional details:**

Here's the dmesg:
```
root@raspberrypi0-wifi:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.71 (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 Tue Dec 3 20:25:51 UTC 9
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 256 MiB at 0x0bc00000
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x94/0x4a4 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=256M  smsc95xx.macaddr=B8:27:EB:85t
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 179488K/458752K available (7256K kernel code, 637K rwdata, 2260K rodata, 460K init, )
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[... 203 lines cut ...]
[   24.679894] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   24.770098] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   24.925166] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   24.932212] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   25.574816] vc4_hdmi 20902000.hdmi: ASoC: Failed to create component debugfs directory
[   25.705633] vc4_hdmi 20902000.hdmi: vc4-hdmi-hifi <-> 20902000.hdmi mapping ok
[   25.842975] vc4-drm soc:gpu: bound 20902000.hdmi (ops vc4_hdmi_ops [vc4])
[   25.934092] vc4-drm soc:gpu: bound 20806000.vec (ops vc4_vec_ops [vc4])
[   25.941229] vc4-drm soc:gpu: bound 20004000.txp (ops vc4_txp_ops [vc4])
[   26.130522] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops [vc4])
[   26.224653] vc4-drm soc:gpu: bound 20206000.pixelvalve (ops vc4_crtc_ops [vc4])
[   26.232866] vc4-drm soc:gpu: bound 20207000.pixelvalve (ops vc4_crtc_ops [vc4])
[   26.454633] vc4-drm soc:gpu: bound 20807000.pixelvalve (ops vc4_crtc_ops [vc4])
[   26.774164] vc4-drm soc:gpu: bound 20c00000.v3d (ops vc4_v3d_ops [vc4])
[   26.780983] checking generic (1eb75000 85400) vs hw (0 ffffffff)
[   26.780994] fb: switching to vc4drmfb from simple
[   26.874041] Console: switching to colour dummy device 80x30
[   26.990474] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   27.103736] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   27.110480] [drm] Driver supports precise vblank timestamp query.
[   27.516832] Console: switching to colour frame buffer device 90x30
[   27.724666] vc4-drm soc:gpu: fb0: DRM emulated frame buffer device
[   28.479889] uart-pl011 20201000.serial: no DMA platform data
root@raspberrypi0-wifi:~# 
```

agherzan, 2019-12-05:
What image are you building?

bernardoaraujor, 2019-12-05:
Tried both core-image-minimal and core-image-base.

agherzan, 2019-12-05:
I'll check to see if I have a w0 around to test. Can you test the board with a Raspbian?

bernardoaraujor, 2019-12-05:
Weird! Do I have a hardware issue? I used the WiFi on this board last month!

Doing `raspi-config`:

```
          ��┌��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─┐
          ��│                                                          ��│
          ��│ No wireless interface found                              ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                                                          ��│
          ��│                          <Ok>                            ��│
          ��│                                                          ��│
          ��└��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─��─
```

bernardoaraujor, 2019-12-05:
here's the raspbian's dmesg:
```
pi@raspberrypi:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.75+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.229
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.t
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 434424K/458752K available (6973K kernel code, 635K rwdata, 2080K rodata, 464K in)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
                   lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (6975 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   ( 464 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 636 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 25197 entries in 74 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000031] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000093] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 s
[    0.000222] bcm2835: system timer (irq = 27)
[    0.000641] Console: colour dummy device 80x30
[    0.001280] console [tty1] enabled
[    0.001368] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060398] pid_max: default: 32768 minimum: 301
[    0.060970] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061030] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062465] CPU: Testing write buffer coherency: ok
[    0.063852] Setting up static identity map for 0x8200 - 0x8238
[    0.065353] devtmpfs: initialized
[    0.075972] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.076442] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191126044627s
[    0.076521] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.077908] pinctrl core: initialized pinctrl subsystem
[    0.079526] NET: Registered protocol family 16
[    0.082774] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.089788] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.089865] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.090030] Serial: AMBA PL011 UART driver
[    0.093344] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.137314] bcm2835-dma 20007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[    0.139909] SCSI subsystem initialized
[    0.140228] usbcore: registered new interface driver usbfs
[    0.140385] usbcore: registered new interface driver hub
[    0.140645] usbcore: registered new device driver usb
[    0.161107] raspberrypi-firmware soc:firmware: Attached to firmware from 2019-09-24 17:37, variant st
[    0.171136] raspberrypi-firmware soc:firmware: Firmware hash is 6820edeee4ef3891b95fc01cf02a7abd7ca57
[    0.183510] clocksource: Switched to clocksource timer
[    0.287944] VFS: Disk quotas dquot_6.6.0
[    0.288127] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.288494] FS-Cache: Loaded
[    0.288880] CacheFiles: Loaded
[    0.308257] NET: Registered protocol family 2
[... 143 lines cut ...]
[   13.175628] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[   13.175655] [vc_sm_connected_init]: start
[   13.186435] media: Linux media interface: v0.10
[   13.192201] [vc_sm_connected_init]: installed successfully
[   13.396823] videodev: Linux video capture interface: v2.00
[   13.509624] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you ha.
[   13.550506] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have bee.
[   13.614323] bcm2835_codec: module is from the staging directory, the quality is unknown, you have be.
[   13.695785] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   13.695857] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   13.706680] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   13.706756] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[... cut ...]
[   14.305399] bcm2835_audio soc:audio: card created with 8 channels
[   22.488342] random: crng init done
[   22.488366] random: 7 urandom warning(s) missed due to ratelimiting
[   23.462260] uart-pl011 20201000.serial: no DMA platform data
[   24.566786] 8021q: 802.1Q VLAN Support v1.8
[   25.061598] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   25.201518] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   25.476134] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'

```

agherzan, 2019-12-05:
Looks like it. Sorry.

bernardoaraujor, 2019-12-05:
thanks for the attention and sorry for the noise.
closing.

closed 2019-12-05 by bernardoaraujor