agherzan/meta-raspberrypi #532: U-boot, Rpi3, Zeus, BT overlays not booting

MadsAndreasenTechPeople, 2019-12-05:
Building the Zeus branch and Yocto 3.0, with U-boot and having "dtoverlay=pi3-miniuart-bt" the boot only shows the multi-color boot screen.
This also happens when using the pi3-disable-bt overlay. The w1-gpio overlay boots fine.
I have also tried to use both Yocto 3.0.1 and master. Master of meta-raspberrypi is also afftected.

Using the same configuration with Yocto 2.7.x and the Warrior branch of meta-raspberrypi works.

Are there perhaps some configuration in the newer branches that I need to sort out?

agherzan, 2019-12-05:
Hi @MadsAndreasenTechPeople . I personally haven't checked those dt overlays in ages. It should work so this really sounds like a kernel issue. Have you checked it on Raspbian? Maybe compare the kernel versions in case we miss something.

MadsAndreasenTechPeople, 2019-12-09:
I have checked with the latest raspbian and that works.
My suspicion is more along the lines of u-boot and the rpi bootloader.

I have tried using the initial commit on the rpi zeus branch, and that fails as well.

Right now I am trying to build a version with rpi-Zeus and Poky Warrior to gain some more knowledge.

agherzan, 2019-12-09:
It would need some digging but if Raspbian is working I'm a little confused. Can you try to temporary reset the rpi firmware revision to the one used in the Raspbian you have tested?

MadsAndreasenTechPeople, 2019-12-10:
I made a build with Poky Warrior and meta-rpi Master and that boots fine. 
So my suspicion is narrowing to u-boot. I'll try out some different versions.

agherzan, 2019-12-10:
Uh, wait. I remember seeing that older versions of rpi (including 3) were not reusing the dtb blob from firmware. Could it be that the firmware dtb is ignored? Can you test without uboot? It should be easy to figure out by checking the uboot defconfig.

MadsAndreasenTechPeople, 2019-12-13:
Did some more digging.
Compared the .config from 2019.01 and .07 - There was nothing really jumping out at me. @agherzan Do you have an input to what I should be looking for?

I tried using the poky 3.0 and meta-raspberrypi Zeus branch, but using u-boot 2019.01 - That combination boots and the overlay is applied.

So the plot thickens on u-boot. Any good ideas on looking for the root cause?

MadsAndreasenTechPeople, 2019-12-17:
I have narrowed it down to this commit in u-boot: 143256b353c4d2ad1cf5e5e63b38410188b35e49 fdt: update bcm283x device tree sources to Linux 5.1-rc6 state

I can see that a number of updates has been made to the rpi 3b device tree, but I am not sure whats causing the problem.
I was under the impression that the rpi bootloader supplied the device tree, but maybe the bootloader and u-boot DT's are being merged?

Is there a way to configure u-boot to ignore its own device tree?

agherzan, 2019-12-17:
@MadsAndreasenTechPeople So, uboot embeds its own dtb for rpi3. Check the defconfig for that machine: https://github.com/u-boot/u-boot/blob/ef7c2af65966a57c98d3c47e6c2fe1ce2103b7f6/configs/rpi_3_32b_defconfig#L21 As a comparison, you can look at the rpi4 which reuses the dtb from the firmware: https://github.com/u-boot/u-boot/blob/master/configs/rpi_4_32b_defconfig

In any way, I would bring this up to the uboot community as I don't see why we would not reuse the dtb.

MadsAndreasenTechPeople, 2019-12-17:
Thanks for the input @agherzan. I changed the defconfig and that works :)

I agree and will see if u-boot will accept a patch.

closed 2019-12-17 by MadsAndreasenTechPeople

agherzan, 2019-12-17:
Sounds good. Glad it worked.

zpfvo, 2020-02-18:
Hi, shouldnt there be maybe a patch in the zeus branch, that changes the defconfig till it is changed upstream? i took me way to long to finally stumble across this closed issue.

agherzan, 2020-02-20:
@zpfvo Sadly we don't actively maintain zeus but if someone is willing to push tested patches, I'm up for merging.

agherzan, 2020-02-20:
@zpfvo We are up for contributions. Or backports if it landed in uboot already.

abondoe, 2020-05-28:
Anyone knows the status on this? Has there been submitted a patch to the u-boot repo?

agherzan, 2020-06-09:
Nothing I am aware of @abondoe