agherzan/meta-raspberrypi #533: Why set `CONFIG_LOCALVERSION` to ""?

michaelbaisch, 2019-12-07:
Hello again,

I would really like to set `CONFIG_LOCALVERSION` to something meaningful. But the meta-raspberrypi layer is making it really hard, because it's set inconveniently to ""  here:

https://github.com/agherzan/meta-raspberrypi/blob/3ae288aa38e3a1609c6e861918a874416d3f0f13/recipes-kernel/linux/linux-raspberrypi.inc#L102

Can someone tell me if there is a really good reason to do this, or if we can maybe get rid of this line? It also voids the use of `LINUX_VERSION_EXTENSION`.

Greetings,
Michael

agherzan, 2019-12-09:
I can't remember why we did that but I think we should remove it now. I don't see any use of it anymore. And I agree, it renders `LINUX_VERSION_EXTENSION` useless. Do you mind pushing a PR?

agherzan, 2019-12-09:
@kraj @pbrkr Do you remmeber why we reset that config?

kraj, 2019-12-09:
We did it after switching to linux-yocto include files which sets it to `"-yocto-${LINUX_KERNEL_TYPE}"` in `do_kernel_configme`
but ideally we should not be emptying out `CONFIG_LOCALVERSION` we should have a weak assignment to set `LINUX_VERSION_EXTENSION`

so something like below(untested) might be the way to go.

```diff
diff --git a/recipes-kernel/linux/linux-raspberrypi.inc b/recipes-kernel/linux/linux-raspberrypi.inc
index ea7f778..6a7df4c 100644
--- a/recipes-kernel/linux/linux-raspberrypi.inc
+++ b/recipes-kernel/linux/linux-raspberrypi.inc
@@ -99,7 +99,7 @@ do_configure_prepend() {
CONF_SED_SCRIPT=""

# Localversion
-    kernel_configure_variable LOCALVERSION "\"\""
+    kernel_configure_variable LOCALVERSION "\"${LINUX_VERSION_EXTENSION}\""

if [ "${INITRAMFS_IMAGE_BUNDLE}" = "1" ]; then
kernel_configure_variable OVERLAY_FS y
```

michaelbaisch, 2019-12-09:
You can see in `meta/recipes-kernel/linux/linux-yocto.inc:58`:

```
LINUX_VERSION_EXTENSION ??= "-yocto-${LINUX_KERNEL_TYPE}"
```

For completeness in `meta/classes/kernel-yocto.bbclass:339`, in `do_kernel_configme()`, it's where the value is acutally set:
```
echo "# Global settings from linux recipe" >> ${B}/.config
echo "CONFIG_LOCALVERSION="\"${LINUX_VERSION_EXTENSION}\" >> ${B}/.config
``` 

So I still think removing is the best solution. Nothing won by setting it manually.

kraj, 2019-12-10:
@michaelbaisch does it append yocto-xxx string if its not set here ? if so then its wrong.

michaelbaisch, 2019-12-10:
Not sure why this would be a problem? But I guess also setting

```
LINUX_VERSION_EXTENSION ?= ""
```

would help. Should I also add this in the pull request?

kraj, 2019-12-10:
because we are not building linux-yocto, so it will not be correct to append default string from linux-yocto. but if thats not happening then i am fine to punt it.

agherzan, 2019-12-17:
@michaelbaisch I think this looks alright so I've kicked the builders.

commit ef6555c0 (2019-12-23): linux-raspberrypi.inc: no longer set CONFIG_LOCALVERSION to empty string

Fixes: #533
closed 2019-12-23 by agherzan